Klipper3d/klipper #4871: Klipper 0.10 does not support Chinese file naming. What can you do to solve it!

pangxie0, 2021-11-04:
[image]

vladimir-poleh, 2021-11-07:
It seems to be a problem caused by migration to the Python3. You can try to use release version v0.10.0

To switch to this version execute these commands (assuming that Klipper install dir is `~/klipper`

```
cd ~/klipper
git checkout v0.10.0
```

Please note that this will break update mechanism. Use following command to return to the more recent version

```
git checkout master
```

vladimir-poleh, 2021-11-24:
It seems that running Klipper on Python 3 resolves this issue.

wlhlm, 2021-11-25:
Same issue: #4794

Sineos, 2021-12-12:
As pointed out by @wlhlm

closed 2021-12-12 by Sineos