Klipper3d/klipper #4873: v0.10.0-111-gdccd5554 - Connect to "mcu" failed after update SKR V1.3

AndyWo83, 2021-11-04:
After update Klipper to the newest, no connection to mcu...

[image]

cd ~/klipper
make menuconfig --> for LCP
make clean
make

_Change the klipper.bin to firmware.bin and copied to the sd card_
sudo service klipper stop
_Put the SD Card in the Board, Reset the SKR V1.3 and wait a minute_
sudo service klipper start

**klippy.log:**
mcu 'mcu': Starting serial connect
webhooks client 1967986368: New connection
webhooks client 1967986368: Client info {'program': 'Moonraker', 'version': 'v0.7.1-111-g61fab42'}
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/pi/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed

firmware.bin changes to FIRMWARE.CUR

[klippy.log](https://github.com/Klipper3d/klipper/files/7477199/klippy.log)

AndyWo83, 2021-11-05:
Nobody an idea?
Here is the excerpt of the created klipper.bin ...

[image]

AndyWo83, 2021-11-05:
I have set the right one...

[image]

nionio6915, 2021-11-06:
I have the same issue :(
SKR 1.1

AndyWo83, 2021-11-06:
Just relaunched klipper and it doesn't work either... Same Problem... It's frustrating!

Arksine, 2021-11-06:
I notice that you have the Communication interface set as "USB" but according to your log you appear to connecting via the Pi's UART.  You will either need to set the correct uart in the menuconfig or you need to change the `serial` option in `printer.cfg` to match that of your MCU.

AndyWo83, 2021-11-06:
@Arksine thank you for this information...
After changing to serial in menuconfig it worked now. But I was 99.9% sure, that I had set up "USB" at the first time...

closed 2021-11-06 by AndyWo83

nionio6915, 2021-11-06:
I think this is related to the STM update that needed successive updates. 
the thing i did to 'fix' it was roll back 4 updates, then recompile and reflash- 
sudo systemctl stop klipper
cd ~/klipper
git reset --hard HEAD~4
sudo systemctl start klipper
...
make menuconfig to verify settings
make clean
make
and then reflash the mcu