Klipper3d/klipper #4874: BME680 gas sensor i2c_read_response error handling

falo2k, 2021-11-04:
Just had a print crash out at 90% with a complaint about reading i2c data.  Log file attached, with snippet below of unhandled exception - seems to be related to my BME680 gas sensor.  I would imagine this could be handled gracefully without impacting an ongoing print.  It looks like there was a previous issue on i2c exceptions that fizzled #4708.

```
Unhandled exception during run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 201, in run
    self.reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/extras/bme280.py", line 281, in _sample_bme680
    res_heat_0 = self._calculate_gas_heater_resistance(self.gas_heat_temp)
  File "/home/pi/klipper/klippy/extras/bme280.py", line 421, in _calculate_gas_heater_resistance
    heater_data = self.read_register('RES_HEAT_VAL', 3)
  File "/home/pi/klipper/klippy/extras/bme280.py", line 455, in read_register
    params = self.i2c.i2c_read(regs, read_len)
  File "/home/pi/klipper/klippy/extras/bus.py", line 182, in i2c_read
    return self.i2c_read_cmd.send([self.oid, write, read_len])
  File "/home/pi/klipper/klippy/mcu.py", line 501, in send
    return self._do_send([self._cmd.encode(data)], minclock, reqclock)
  File "/home/pi/klipper/klippy/mcu.py", line 499, in _do_send
    raise self._error(str(e))
CommandError: Unable to obtain 'i2c_read_response' response

```

[klippy.log.i2c_read.zip](https://github.com/Klipper3d/klipper/files/7478189/klippy.log.i2c_read.zip)

simonkuehling, 2021-11-05:
In most cases, communication faults on the i2c bus are a hardware related issue - like noise on the lines, too long cables or otherwise poor signal quality.
Can you tell a little about your electronics setup and wiring of the sensor?

falo2k, 2021-11-05:
Sure, happy to share.  Details of the setup are below.

I think the issue here, though, is less about the fact that an i2c communication failed and more about the handling of it by Klipper.  I don't think killing the print with an exception because of a sampling failure is an appropriate response - either a retry system or a warning seems more appropriate.

======= Setup ========
Sensor is installed in a Voron 2.4 300x300, Klipper running on a Pi 4B (8GB).  Layout of my electronics enclosure illustrated below (front at top).  The sensor is attached to the bottom frame extrusion a little away from the Z drag chain.  Cabling is done by 20AWG wire, dupont terminated at each end to connect to the sensor & pi.  This runs approximately 45cm into the electronics compartment and through cable management in the middle to the pi.  All cable runs are sleeved in braided sleeving to prevent tangle.

[image]

simonkuehling, 2021-11-05:
Well, i2c isn't designed for noisy environments - its mostly intended as an on-board-bus to simplify PCB schematics and reduce I/O requirements. That's why there is no error checking and fault correction available (and needed) here.
Half a meter of unshielded i2c cable run in a very noisy environment of a 3D printer (many stepper lines, heaters...) sounds like a recipe for communication faults to me.

Technically, a good answer would be to use another communication bus (which has all the checking etc.) for this. USB for example - you could use an additional mcu dedicated to the sensor (a cheap arduino nano (clone?) should do for example) and connect the sensor with short wires or solder both to a small perfboard.

falo2k, 2021-11-05:
I'm honestly happy to rip the sensor out of Klipper entirely on my side.  I'd still argue the case that if one takes the view that it's likely to be a noise risk in a 3D printer environment and is provided as a module for software for the 3D printer environment, then the sensible course here would be to gracefully ignore the failed read and move on with printing.  I wouldn't consider that fault correction, more acknowledgement of the risk of the fault and the effect that a fault should have on normal operations.  If you're making that choice to not do active error checking / handling for the module, you can either choose to let that cause an unhandled exception that terminates all operations or pass on with a warning.  Given the devices involved and the priority of operations (printing vs reading gas sensor data), the latter seems the sensible route.

nforceroh, 2021-11-06:
Just wanted to share. I've been having issues having a BME280 or a HTU21D using i2c connected to the RPi. about a week ago I connected them to a stm32f103(blue pill) running klipper instead of the RPi and it's been rock stable since. When connected to the RPi 3, for me anyways, it  crashes pretty often thru klipper. I tried  the same setup but just the RPi loaded with debian and connecting to the sensor using python, didn't experience any timeout or error. at first glance I would suspect the i2c code in klipper but I have no idea how to debug further and honestly I don't really need to anymore since it's solid on the blue pill. so just spend the 2-5$ on a blue pill and use that. I've got mine controlling my neopixel led strips, adxl and bme280.  I suspect it's something with the timings on the RPi vs a dedicated MCU which makes sense because the RPi is doing many different things

simonkuehling, 2021-11-08:
> This is where I've got to disagree with you. The module already has exception handling in the sampling process for other read failures, which will stop the polling for data and 0 the values. I'm not seeing this consensus to not handle exceptions.

Generally a fair point... I guess we should wait to see if @KevinOConnor wants to add his view on that...

samwiseg0, 2021-11-08:
I would agree that there should be a way to gracefully warn of a timeout and not have the print fail. I too have experienced where my HTU21D times-out and causes the print to fail. 

> This is where I've got to disagree with you. The module already has exception handling in the sampling process for other read failures, which will stop the polling for data and 0 the values. I'm not seeing this consensus to not handle exceptions.

I agree with this. I would prefer the values are 0 out and the print continues. In my case the sensor is used to monitor the humidity values of my filament drybox and engage the heater if a certain humidity level is present. I would much prefer to add some logic to my delayed gcode loop to account for 0 values than to have a print fail.

samwiseg0, 2021-11-08:
As a quick and dirty workaround I added this around the instances of `params = self.i2c.i2c_read([],3)`. I have not done extensive testing but I think this can serve as a temporary workaround. From my past experience if a read fails if you attempt it again it should succeed.

```py
from time import sleep

            for attempt in range(4):
                try:
                    params = self.i2c.i2c_read([],3)
                except Exception as e:
                    logging.error("htu21d: Could not read from htu21d sensor retrying... Error: {}".format(e))
                    sleep(1)
                else:
                    break
            else:
                logging.error("htu21d: Failed to get htu21d sensor data after 4 attempts!")
```

nforceroh, 2021-11-10:
What about adding an option in the temperature_sensor definition that you could specify this is only an informational sensor and we can ignore read failure and not trigger a failure/shutdown

samwiseg0, 2021-11-10:
Well... I am starting to think that there is indeed a bug in klipper. Even after implementing a check to just read the data again from the sensor it still failed. Even in a noisy environment requesting the data again should return data from the sensor. 

```
htu21d: Could not read from htu21d sensor retrying... Error: Unable to obtain 'i2c_read_response' response
htu21d: Could not read from htu21d sensor retrying... Error: Unable to obtain 'i2c_read_response' response
htu21d: Could not read from htu21d sensor retrying... Error: Unable to obtain 'i2c_read_response' response
htu21d: Could not read from htu21d sensor retrying... Error: Unable to obtain 'i2c_read_response' response
htu21d: Failed to get htu21d sensor data after 4 attempts!
htu21d: Error reading data
```

rtbon, 2021-11-24:
> Well, i2c isn't designed for noisy environments - its mostly intended as an on-board-bus to simplify PCB schematics and reduce I/O requirements. That's why there is no error checking and fault [...]

Hi, thanks for your solution, I'm trying to implement a HTU21D sensor but I still get error `MCU 'nanomcu' shutdown: i2c timeout`
I'm trying to define ` i2c bus` but I can't find it even if I know its analogs pins 4 & 5

Can you help me ?
Below my printer.cfg 

```
[mcu nanomcu]
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

[temperature_sensor chamber]
sensor_type: HTU21D
i2c_mcu: nanomcu
i2c_bus: **?**
```

simonkuehling, 2021-11-24:
Looking into the microcontroller code for the AVRs I see that the bus seems to be named `twi` (from „two wire interface“) there…

So I guess it is `i2c_bus: twi` for the nano…

rtbon, 2021-11-24:
Exactly thanks ! But unfortunately I still get the `MCU 'nanomcu' shutdown: i2c timeout` error

simonkuehling, 2021-11-24:
That error is thrown by the `i2c_wait()` function in the microcontroller i2c code when waiting for an operation (starting a transmission, sending data, receiving data...) to finish. Could be a variety of causes here. I am hesitating to dig deeper inside this issue because this is getting quite off-topic, but I would double check the pinout, wiring (SCL/SCA mixed up?) and supply voltage to the sensor.
If you are mildly familiar with programming, you could sprinkle a few `logging.info("I am here...")` into `klippy/extras/htu21d.py`-->`_init_htu21d()` to see where exactly it is running into the timeout.
On the discourse forum or the discord chat server you will probably get a helping hand with further debugging: https://www.klipper3d.org/Contact.html

rtbon, 2021-11-24:
Thank you very much for your help ! I will make my investigation

MatixYo, 2021-12-24:
Are there any plans of fixing the original issue? On my setup (also Voron) I have sensor mounted on top right and use >50 cm long wire for it. It crashes on average every 50-100h of printing. Sometimes very late in print which is frustrating.

simonkuehling, 2021-12-25:
@MatixYo, the proper fix in my opinion is still to get rid of the long i2c cable and use a dedicated MCU directly at the sensor - effectively using a USB connection for the long wire run. I don‘t think it is worth developing a workaround if a robust alternative already exists…

Gilabite, 2021-12-25:
There is also things like i2c extenders too https://www.adafruit.com/product/4756

samwiseg0, 2021-12-25:
> @MatixYo, the proper fix in my opinion is still to get rid of the long i2c cable and use a dedicated MCU directly at the sensor - effectively using a USB connection for the long wire run. I don‘t think it is worth developing a workaround if a robust alternative already exists…

I still don't see a valid reason why there should be no error handling? A complete MCU shutdown seems a bit extreme for this case. Again it was already [mentioned above](https://github.com/Klipper3d/klipper/issues/4874#issuecomment-961961817) that the sensors themselves already have error handling. Why is it not possible to extend the same basic functionality into klipper? Much like there is error handling in the rest of the code base. I don't see a valid reason there should be any in this case?

samwiseg0, 2021-12-25:
> There is things like i2c extenders too [adafruit.com/product/4756](https://www.adafruit.com/product/4756) I have a bunch of these and have used them on projects with cables several meters long with no issues. Also twisting one of the signal wires with the ground wire and the other signal wire with the power wire will help alot. Doesnt matter which ones you pair just dont twist the signal wires together.

I have not tried these. I will see if it does make a difference/prevent the issue from happening.

simonkuehling, 2021-12-25:
> I still don't see a valid reason why there should be no error handling? A complete MCU shutdown seems a bit extreme for this case. Again, it was already [mentioned above](https://github.com/Klipper3d/klipper/issues/4874#issuecomment-961961817) that the sensors themselves already have error handling. Why is it not possible to extend the same basic functionality into klipper? Much like there is error handling in the rest of the code base. I don't see a valid reason there should not be any in this case?

I am not saying there should be no error handling for error information obtained through the i2c communication from the sensor itself - but due to the implementation and intended use of the i2c protocol I think that a read error (in terms of „could not obtain a response at all“) is indeed a major fault and should always be resolved in hardware.
As far as I understood, this type of fault was the initial problem that was referred to.
Sure, you can catch the exception and throw an e-stop explicitly right there. I do not see a valid point for adding an additional configuration option to just skip over i2c communication errors. You would want to define a retry-threshold then - but that would be a very bad workaround for a hardware layout issue again.
In my view, explicitly catching that exception is not a fix to the problem. I do think that a MCU shutdown is adequate to a i2c read error - which should just never happen unless something is actually broken or the PCB needs another revision due to signal noise issues.

samwiseg0, 2021-12-25:
> I am not saying there should be no error handling for error information obtained through the i2c communication from the sensor itself - but due to the implementation and intended use of the i2c [...]

As I agree that if there is a hardware problem that should be solved. This is not so cut and dry. You could go days or weeks without an issue. People are not always in control of your environment. I live in a space where I am not in full control of what noise is in my environment. Furthermore causing a failed print by issuing an MCU shutdown for something that is not necessary/critical or mandatory for the printer to function does not make too much sense to me. Handling a sensor that is not critical to printing in a graceful manner that does not cause a failed print would be beneficial for everyone.

simonkuehling, 2021-12-27:
> As I agree that if there is a hardware problem that should be solved. This is not so cut and dry. You could go days or weeks without an issue. People are not always in control of their environment. I live in a space where I am not in full control of what noise is in my environment or what appliances maybe in the vicinity of my printer. Furthermore causing a failed print by issuing an MCU shutdown for something that is not necessary/critical or mandatory for the printer to function does not make too much sense to me. Handling a sensor/communication error that is non critical to printing in a graceful manner that does not cause a failed print would be beneficial for everyone.

I don't think external EMI noise should be much of a concern if we are talking about a wire run inside a 3D printer that has multiple high-current/high frequency stepper cables and high-current heater wires inside in the first place. Electromagnetic interference is highly dependant on distance from the radiator source (inverse square law). In this environment a 50cm (unshielded?) i2c bus cable just isn't a very robust setup to choose.

Besides using a shielded cable (the shield should be tied to GND at one side only), you could also set a lower i2c clock speed for the sensor in your config. Default speed is 100 kHz I think, so you could try something lower to make it a bit more robust against noise.

andematt, 2022-03-08:
Running into this issue as well. HTU21D connected to RPI4 i2c headers, mounted in top, back corner of voron 2.4. MCU shutdown just a few minutes after starting heat soak. 
I don't think the error handling would have to be robust around these sensor faults (retries, timeouts, etc) but a flag to choose whether it causes mcu shutdown would be helpful. 
I don't think adding another microcontroller to the mix should be considered a fix (as much as I like adding another microcontroller to things...). 
Is one of the common sensors in the docs one that could just be recommended/top-pick in the docs? These things are all <$5, I would have just chosen the recommendation. It seemed from docs and voron discord that i2c and the htu21d were preferred, so I set aside onhand DHT11's and ordered up. Just adding this to the discourse here to say - I'm putting in a vote for whatever path can deliver temperature sensor reliability out on ~1m of wire - no preference for sensor/protocol or bugfix here over additional details in the docs.

andematt, 2022-03-09:
I found a ds18b20 laying around from an old project. It seemed much more stable, fault wise, during heat soak. I started a couple hour print. Later, I found the print finished but also noticed my exhaust fan wasn't running but I could hear the toolhead fan spinning - Rpi mcu fault and shutdown with the ds18b20 reported as the culprit.

So, off topic for the 'i2c' nature of this issue I know but I tried to "run away" from the 1m i2c cable as suggested and still had a shutdown fault on an "unimportant" sensor.

`Receive: 98 13681.302919 13681.261099 57: seq: 1d, #output Error: 'Unable to find temperature value in DS18B20 report'
Receive: 99 13681.302938 13681.261099 12: seq: 1d, shutdown clock=4059597114 static_string_id=Error reading DS18B20 sensor
Transition to shutdown state: MCU 'rpi' shutdown: Error reading DS18B20 sensor`

Karlostavitch1, 2022-03-13:
For what it's worth, I've been having the same issues with a HTU21D. Forced MCU shutdown every 50 or so hours of printing. It doesn't seem logical to throw more money at and bake in additional infrastructure complexity for a common issue that could be resolved via error handling. From an end-user perspective, Klipper shouldn't need to define which devices it considers essential for mcu operation, it could be something that is defined in the printer.cfg for the corresponding sensor.

mozzel, 2022-05-06:
I find i2c devices have high MCU usage, when using a offboard MCU (blue pill) the printer will fail 1 out of 4 prints with the i2c timeout error. In the screenshot MCU TEMP is running a BME280 on a 30mm long cable for testing, I've also tried ssd1306 OLED but is has the same issues.

Can others confirm the high MCU usage with i2c devices?

[image]

gargamel778, 2022-05-08:
I feel it is a pity that Klipper is so aggressive about those temperature sensors (or miscommunications with ones).
On my printers, I have recurring issues with BME280/BME680 (I2C) and DS18B20 (1-Wire). 
Wire run length or noisy lines may not be the culprit - those errors sometimes occur even on my Voron v0.1 while the printer is idle - imagine <30cm of wiring length between the board to the sensor.

As for the BMEs - I've tried the I2C bus of the Pi, the printer's MCU (SKR E3 Mini 1.4), Arduino nano as an additional MCU, and even Klipper Expander board's MCU - all had forced shutdowns due to I2C errors.

I wish there was a way to flag these sensors as non-critical and thus not force a shutdown when they miss a response on measurement. Disabling these sensors is probably the safest for me (due to uncalled for and spontaneous shutdowns).

Sineos, 2022-08-25:
Closed as apparently no one is contributing and / or working on this topic.

closed 2022-08-25 by Sineos

samwiseg0, 2022-08-25:
> Closed as apparently no one is contributing and / or working on this topic.

Why not leave it open as maybe someone will address it eventually?

I have bumped it several times for that reason.

ChrisRonaasen, 2022-09-24:
I'm with samwise on this one. Can't even connect my MPU6050 and do a simple QUERY_ACCELEROMETER without this error popping up immediately, despite checking that the pins I'm using (PA8, PC9 on BTT E3 RRF) are correctly set to i2c3 SCL, SDA respectively during boot, seen using debug console. Address for the device is correct as default, pins are set correctly and leads directly to their corresponding pins in the MCU, and both SCL and SDA on the device have pullup resistors onboard. My wires currently are about 1cm each, with 0.05DB loss in signal (basically fiber at this point).
Essentially, I'm down to my last option, which is this error causing the issue. No idea what else to do now except fix the error handling myself, if noone else is going to.