Klipper3d/klipper #4875: SHAPER_CALIBRATE throws errors after todays stepper control changes

miklschmidt, 2021-11-04:
After todays 4 commits (Nov 4), SHAPER_CALIBRATE now fails early and the klipper transitions into shutdown state. SHAPER_CALIBRATE works again when reverting back to Klipper3d/klipper@913d099261399ba737b35752f3c3be6da618bb6a.

I have 2 users who reported the issue. Both are using an ADXL345 (confirmed working, noise within 30hz) connected to an SKR Pro 1.2.

Let me know if you need more info, i will reproduce on my own printers tonight.

[klippy.log](https://github.com/Klipper3d/klipper/files/7478460/klippy_1.2.log)

LUK48, 2021-11-05:
Please find log file 
[klippy (10).log](https://github.com/Klipper3d/klipper/files/7486240/klippy.10.log)

RuiRocha1, 2021-11-06:
I have the same problem..
[klippy (1).log](https://github.com/Klipper3d/klipper/files/7489798/klippy.1.log)
.

miklschmidt, 2021-11-06:
Fixed with @9ecddd1ba4fe6d481f01fa2b1d9175b927907156 thank you Kevin!

closed 2021-11-06 by miklschmidt