Klipper3d/klipper #4877: config ssd1306 SPI

daniellom30, 2021-11-04:
Hello,  I have a SKR 2 board, and I am trying to connect a ssd1306 SPI oled display through EXP1, but I can't get it to work, it always asks me to specify the miso pin, which my oled does not have, I leave it a photo of the module and my configuration.

[display]
lcd_type: ssd1306
cs_pin: PE9
dc_pin: PB1
reset_pin: PE10
spi_software_sclk_pin: PE5
spi_software_mosi_pin: PB0
#spi_software_miso_pin: 

[image]
[image]

Thanks for the help, and I apologize if it has already been solved, but I searched all day and I did not find anything related to this display (7pin), most of the information is related to the 4-pin I2C display

I apologize for my english

simonkuehling, 2021-11-05:
Do you have a spare pin on your mcu that you can just define as `spi_software_miso_pin` in order to make the software serial initialization happy? As long as the pin is not connected to anything else, that should work...

daniellom30, 2021-11-05:
effectively klipper starts, but the screen remains black

daniellom30, 2021-11-06:
Hello, I was able to solve the problem, and the screen already works perfectly, I leave here all the information so that if someone needs it, they can access it.

Kevin, if you can add it to the configuration examples section, I think it would be very helpful to all those who have a BTT SKR 2 board, in the end I managed to make both 4-pin I2C oled screens and 7-pin spi oled screens work, I leave here all the printer.cfg configuration for both screens will work for both ssd1306 and sh1106 screens, however, for ssd1306 it doesn't work X_offset

These settings are only for screens without encoder, they only offer information

########################################
# sh1106 or ssd1306 4 pin - I2C
# uses the I2C pins of the MCU BTT SKR 2 - see pinout image to locate them
########################################
#MCU connections#
#VDD - 5V
#GND - GND
#SCK - PB8
#SDA - PB9

[display]
lcd_type: sh1106 # or ssd1306 It will depend of the  screen controller you have, although they work practically the same
i2c_bus: i2c1a
i2c_mcu = mcu
x_offset: 2

########################################
# sh1106 7 pin spi
# uses the EXP 1 pins of the MCU BTT SKR 2 - see pinout image to locate them
########################################
#MCU connections#
#VDD - 5V
#GND - GND
#SCK - PC5
#SDA - PB0
#CS: PE9
#DC: PB1

[display]
lcd_type: sh1106 # or ssd1306 It will depend of the  screen controller you have, although they work practically the same
cs_pin: PE9
dc_pin: PB1
spi_software_sclk_pin: PC5
spi_software_mosi_pin: PB0
spi_software_miso_pin: PE11 # this pin is not connected to the screen or the mcu, only so that mcu error is not generated
contrast: 200
x_offset: 2
vcomh: 30 # if your screen shows some light bands in the background, use this parameter to eliminate them

one more time, sorry for my english

closed 2021-11-06 by daniellom30