Klipper3d/klipper #4879: "mcu 'mcu': Unable to extract params from: config_stepper" after update

BarrMo, 2021-11-05:
[klippy.log](https://github.com/Klipper3d/klipper/files/7479711/klippy.log)

I just updated the firmware and get the following:
_

> mcu 'mcu': Unable to extract params from: config_stepper
> 
> This type of error is frequently caused by running an older
> version of the firmware on the micro-controller (fix by
> recompiling and flashing the firmware).
> 
> Known versions: host=v0.10.0-111-gdccd5554, mcu=v0.9.1-779-g59bd51d2, rpi=v0.9.1-786-gc2d3a168
> 
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Protocol error connecting to printer_

_

Tried restarting firmware, restarting host, updating RPi - no change. 
Any suggestions are most welcome!

Crillas24, 2021-11-05:
You updated klipper only on host, you need to update mcu's too.
Everything is said in error message 

This type of error is frequently caused by running an older
version of the firmware **on the micro-controller** (fix by
recompiling and flashing the firmware).

https://www.klipper3d.org/Installation.html#building-and-flashing-the-micro-controller

Your host is updated to v0.10.0-111
but now you need to flash mcu and rpi mcu to the same version

BarrMo, 2021-11-05:
Thanks Crillas24. 
I followed the instructions and everything appeared to go well but essentially got the same error again when I started Klipper. rpi has updated but the mcu is still the same.

Known versions: host=v0.10.0-111-gdccd5554, mcu=v0.9.1-779-g59bd51d2, rpi=v0.10.0-111-gdccd5554

Crillas24, 2021-11-05:
You've updated klipper linux process mcu, now you need to update printer board, dont know exactly what printer do you have and board in it, some boards you need to transfer firmware.bin to sd card some can be updated via usb cable. 
Do you remember first installation? when you first flashed klipper to the board?

mmmCaribou, 2021-11-05:
I'm in a similar situation after update. Same error. Tried all the guides including putting the firmware.bin directly on the sd card. No luck.  BTT SKR E3 Turbo and Raspberry Pi 4 hardware. 

[klippy-2.log](https://github.com/Klipper3d/klipper/files/7480407/klippy-2.log)

BarrMo, 2021-11-05:
I'm running very similar hardware to mmmCaribou - BTT SKR 1.4T & RPi 4B

Crillas24, 2021-11-05:
@mmmCaribou 
You have exact same thing to do, you updated klipper on raspberry pi i guess both of you did adxl measurements.

Now you need to flash board in printer. 
> cd ~/klipper/
> make menuconfig

in make menuconfig go to micro-controler architekture and select lpc 176x
next in procesor model 1769 as you both have turbo version
then you press q to exit and y to save changes 
after that you need to flash it to board i cant say if you can do it via usb or you need to transfer it to sd card but voron docs says you need to transfer it to sd card so do it, transfer created firmware.bin to your pc and then to sd card of your printer power off / insert card / power on 

https://docs.vorondesign.com/build/software/skr13_klipper.html

BarrMo, 2021-11-05:
Excellent! got it.
Thank you very much @Crillas24 I'll leave this thread open a little longer until we hear back from @mmmCaribou to confirm it worked there.

Crillas24, 2021-11-05:
Great! 'Im glad i could help 😃

ian-watson, 2021-11-05:
Im getting the same problem and have done the same process to flash the mcu that i have done since the initial setup. 

`Known versions: host=v0.10.0-111-gdccd5554, mcu=v0.10.0-99-g5d9d91d3`

looking at the commit logs can someone explain this:

```
commit dccd5554091561ec3099de32505530b4295b67d9 (HEAD -> master, origin/master, origin/HEAD)
Author: Kevin O'Connor <kevin@koconnor.net>
Date:   Sat Oct 30 12:24:42 2021 -0400

    stepper: Move min_next_time check to stepper_event_full()
    
commit 5d9d91d3a6b2e728334a9a58574d1bdf6cbabe63
Author: Kevin O'Connor <kevin@koconnor.net>
Date:   Mon Nov 1 11:34:29 2021 -0400

    docs: Should be "LM75" instead of "lm75" in Config_Reference.md
```

commit hash for oct 30 starts with dccd5554 which appears in my host version v0.10.0-111-gdccd5554
commit hash for nov 1 starts with 5d9d91d3 which appears in my mcu version v0.10.0-99-g5d9d91d3

how can i roll back with git so my host is running v0.10.0-111-gdccd5554?

Crillas24, 2021-11-05:
If you did flash mcu again then you did something wrong.
Can you specify mcu? and tell what you did? if you have board with lpc did firmware.bin changed to firmware.cur?

ian-watson, 2021-11-05:
ran this on raspberry pi 3b+
```
make clean 
make
.....
Version: v0.10.0-111-gdccd5554
  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
  Creating hex file out/klipper.bin
```
then transferred klipper.bin to firmware.bin on 3dcard to octopus v1.1 (stm32f446) and file was renamed to firmware.cur

Crillas24, 2021-11-05:
That's strange everything seems right. Make is outputing right version and mcu is flashing it. And klipper still reports 0.10.0-99? Sadly i cant help as i dont know what is hapening :/
One question i have is, cant you flash firmware via usb? i have same chip on fysetc spider and it can be flashed using usb, one command and done.  Easier than sd card and i have confirmation if it is flashed.

ian-watson, 2021-11-05:
either trying the usb way or trying to roll back is my next mission. Thanks for trying to help. I wont be so trigger happy to keep everything up to date

ian-watson, 2021-11-05:
had no luck connecting via usb via both my pi3&4 with either lsusb or dfu-util --list so compiled firmware on newer rp4 and now mcu & host have same version. still scratching head

teuphomthichack, 2021-11-05:
make sure you rename the new .bin that you rebuild to something else different than you originally had. For example, if you had the .bin file named "klipper_firmware.bin", you need to name the new build file to something like "klipper_firmware_UPDATED.bin"

teuphomthichack, 2021-11-05:
If you have access to the current pi board. SSH into it then follow these steps:

cd klipper 
make clean
make

and then copy the output file generated by the commands into the SD card and put it back on the printer.

bill-orange, 2021-11-05:
I found `make clean` to be essential.  Moreover on one of three machines `serial:` changed, requiring printer.cfg to be updated.

mmmCaribou, 2021-11-05:
Followed the Voron instructions and I'm back in production mode.  Rather than just SSH. I used Cyberduck to transfer the file to the SD card. Maybe I missed the renaming the .bin file. It was odd that this update caused the issue. My September firmware update was uneventful and it was a USB transfer. Thanks to Crillas24 for assistance. In my book, the Voron instructions solved the problem, and I'm fine closing this ticket.

BarrMo, 2021-11-06:
I was able to run all the updates via ssh however I couldn't copy the firmware .bin file over that way. I had to pull the SD card from the RPi and copy it. Then I had to pull the SD card from the BTT 1.4T board, delete firmware.cur and then drag the new firmware.bin over to it. booted fine after that.

teuphomthichack, 2021-11-06:
> I was able to run all the updates via ssh however I couldn't copy the firmware .bin file over that way. I had to pull the SD card from the RPi and copy it. Then I had to pull the SD card from the [...]

the output file after you rebuild the firmware should be at /klipper/out/ . What I did was I use WinSCP to copy the file and transfer to my local PC and drag it into the SD card.

banec, 2021-11-06:
Hi, I got same error on BTT SKR 1.3 cards. Tried _make clean && make_, after that copied and renamed file from /out to a SD Card. 
Pi doesn't even register cards anymore (lsusb doesn't show them and there is no serial)

VitoCassisi, 2021-11-07:
I'm having the same issue. Flashed using the instructions [here](https://www.klipper3d.org/Installation.html#building-and-flashing-the-micro-controller), including `make clean` like I usually do, but no success. Also manually flashed via SD card and same result.

drgonzo31, 2021-11-07:
Hello all,
I'm having the same issue. I've done the MCU firmware upgrade on the printer before. It resolved the issue. I've done the make clean, copied over the klipper.bin, renamed to firmware.bin as previous. Tried it three times now. Still get the error below. I just upgraded to latest version via Kiauh.

mcu 'mcu': Command format mismatch: config is_config=%c crc=%u is_shutdown=%c move_count=%hu vs config is_config=%c crc=%u move_count=%hu is_shutdown=%c

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

Known versions: host=v0.10.0-113-gd4aee4f5, mcu=v0.10.0-111-gdccd5554

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer

VitoCassisi, 2021-11-07:
I had this issue today and found that I had to update the secondary RPi MCU again as detailed [here](https://www.klipper3d.org/RPi_microcontroller.html). Probably not obvious if you've only ever used that MCU to interface with an ADXL345 and have since disconnected it.

banec, 2021-11-07:
Freshly updated to v0.10.0-113-gd4aee4f5 now and it appears to be working?! No ADXL345 here (yet).
Something weird was happening here.

BarrMo, 2021-11-07:
> the output file after you rebuild the firmware should be at /klipper/out/ . What I did was I use WinSCP to copy the file and transfer to my local PC and drag it into the SD card.

I'm running an Ubuntu machine and I was trying to ssh and run the "scp" command. tbh I'm not very good in the terminal. In the end I figured instead of struggling to move the file around my LAN it'd be much easier and more reliable to physically pull the SD cards and do it manually.

closed 2021-12-20 by github-actions[bot]