Klipper3d/klipper #4889: Version: v0.10.0-113-gd4aee4f5 not flashing MCU

drgonzo31, 2021-11-07:
Hello all,
I've done the MCU firmware upgrade on the printer before. It resolved the MCU and host miss match each time.  However, this time it isn't working, I just upgraded to the latest version via Kiahah
After doing make I get this, confirming the correct firmware version. I then copy it over to USB insert into my ender 3v2, reboot it and wait sometime to flash. I restart but it still saying older version of MCU. I've tried this four time now.
  Building out/compile_time_request.o
Version: v0.10.0-113-gd4aee4f5
  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
  Creating hex file out/klipper.bin

mcu 'mcu': Command format mismatch: config is_config=%c crc=%u is_shutdown=%c move_count=%hu vs config is_config=%c crc=%u move_count=%hu is_shutdown=%c

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

Known versions: host=v0.10.0-113-gd4aee4f5, mcu=v0.10.0-111-gdccd5554

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer

AndyWo83, 2021-11-07:
Have you rename klipper.bin to firmware.bin?

drgonzo31, 2021-11-07:
Yes I have.

AndyWo83, 2021-11-07:
Hmm, I updated the MCU, but although the host and MCU have the same version, I get the message that an old version is on it.

[image]

drgonzo31, 2021-11-07:
Okay formatting the SD card didn't fix the problem. Still won't upgrade with firmware.bin.

Angelol125, 2021-11-07:
u have problems with ur rpi mcu, not ur printer's motherboard mcu. i'm guessing u have tried input shaper and thus use ur pi as an mcu. u need to update that as well for the whole system to work.

AndyWo83, 2021-11-07:
@Angelol125 that fix the problem... Thank you

ykri, 2021-11-07:
> u have problems with ur rpi mcu, not ur printer's motherboard mcu. i'm guessing u have tried input shaper and thus use ur pi as an mcu. u need to update that as well for the whole system to work.

How can I do this update?

Angelol125, 2021-11-07:
https://www.klipper3d.org/RPi_microcontroller.html#building-the-micro-controller-code

snympi, 2021-11-07:
I have exactly the same problem after updating Klipper just now. Recompiling and flashing the MCU firmware does not work. The MCU in the error message reports the more recent version. My rpi version is far behind all of a sudden:

mcu 'rpi': Command format mismatch: config is_config=%c crc=%u is_shutdown=%c move_count=%hu vs config is_config=%c crc=%u move_count=%hu is_shutdown=%c

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

Known versions: host=v0.10.0-113-gd4aee4f5, mcu=v0.10.0-113-gd4aee4f5, rpi=v0.10.0-62-gec2116b4

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer

Same problem as @AndyWo83, but my rpi version is even older. This was working fine after having done the same update (klipper update & mcu firmware recompile) 3 days ago.

How can I force the rpi klipper update - perhaps that updates the rpi to match the MCU?

snympi, 2021-11-07:
Update: @Angelol125 suggestions above works perfectly. The rpi MCU is for input shaping and was out of date.

jtrmal, 2021-11-07:
I have the same problem:
```
$ ./scripts/flash-sdcard.sh /dev/serial/by-id/usb-Klipper_stm32f103xe_35FFD4054246303604761757-if00 btt-skr-mini-e3-v1.2
Flashing /home/pi/klipper/out/klipper.bin to /dev/serial/by-id/usb-Klipper_stm32f103xe_35FFD4054246303604761757-if00
Checking FatFS CFFI Build...
Connecting to MCU...Connected
Checking Current MCU Configuration...
SD Card Flash Error: Command format mismatch: config is_config=%c crc=%u move_count=%hu is_shutdown=%c vs config is_config=%c crc=%u is_shutdown=%c move_count=%hu
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1160, in main
    spiflash.run()
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1109, in run
    self.run_reactor_task(self.run_reset)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1096, in run_reactor_task
    k_reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
```

Only in my case I'm pretty sure everyhting has the same version -- I've just finished flashing and all was on the same version.
This patch help -- I noticed the template is almost identical, only two fields swapped.

$ git diff scripts/spi_flash/spi_flash.py
diff --git a/scripts/spi_flash/spi_flash.py b/scripts/spi_flash/spi_flash.py
index 617c2d4f..fbcaad6f 100644
--- a/scripts/spi_flash/spi_flash.py
+++ b/scripts/spi_flash/spi_flash.py
@@ -99,7 +99,8 @@ SD_SPI_SPEED = 4000000
 # MCU Command Constants
 RESET_CMD = "reset"
 GET_CFG_CMD = "get_config"
-GET_CFG_RESPONSE = "config is_config=%c crc=%u move_count=%hu is_shutdown=%c"
+GET_CFG_RESPONSE = "config is_config=%c crc=%u is_shutdown=%c move_count=%hu"
 ALLOC_OIDS_CMD = "allocate_oids count=%d"
 SPI_CFG_CMD = "config_spi oid=%d pin=%s"
 SPI_BUS_CMD = "spi_set_bus oid=%d spi_bus=%s mode=%d rate=%d"
 ```
 
 After this path, everything worked flawlessly:
 ```
 $ ./scripts/flash-sdcard.sh /dev/serial/by-id/usb-Klipper_stm32f103xe_35FFD4054246303604761757-if00 btt-skr-mini-e3-v1.2
Flashing /home/pi/klipper/out/klipper.bin to /dev/serial/by-id/usb-Klipper_stm32f103xe_35FFD4054246303604761757-if00
Checking FatFS CFFI Build...
Connecting to MCU...Connected
Checking Current MCU Configuration...Done
MCU needs restart: is_config=1, is_shutdown=0
Attempting MCU Reset...Done
Waiting for device to reconnect...Done
Connecting to MCU...Connected
Initializing SD Card and Mounting file system...

SD Card Information:
Version: 2.0
SDHC/SDXC: False
Write Protected: False
Sectors: 247808
manufacturer_id: 0
oem_id: 0
product_name: APPSD
product_revision: 0.0
serial_number: AB0401B5
manufacturing_date: 6/2020
capacity: 121.0 MiB
fs_type: FAT16
volume_label: b''
volume_serial: 2826674516
Uploading Klipper Firmware to SD Card...Done
Validating Upload...Done
Firmware Upload Complete: firmware.bin, Size: 23268, Checksum (SHA1): D621783D871BD297C2DF6DBF97C5353310EB852F
Attempting MCU Reset...Done
Waiting for device to reconnect....Done
Connecting to MCU...Connected
Verifying Flash...Version matched...Done
Firmware Flash Successful
Current Firmware: v0.10.0-113-gd4aee4f5
Attempting MCU Reset...Done
SD Card Flash Complete```

drgonzo31, 2021-11-07:
I'm glad suggestions in this thread have helped people, but the error message I posted doesn't include anything about the RPI.

 I only get errors about host and MCU. Following the procedure outlined here (https://www.klipper3d.org/SDCard_Updates.html) which I've used successfully in previous upgrades doesn't work this time. my MCU stays the older version. I'm not using the RPI as a secondary MCU.

drgonzo31, 2021-11-07:
Okay I have fixed it. I removed the usb cable from my rpi and restarted the ender 3v2 with the firmware.bin file in the SD card slot. It successfully flashed now. It's weird because I pretty sure i've upgraded the MCU previously leaving the USB cable in, using the SD flash method.

closed 2021-11-07 by drgonzo31