Klipper3d/klipper #4890: Update to v0.10.0-113-gd4aee4f5(Octopus1.1) uart2 not working anymore

wolfgangmauer, 2021-11-07:
After update my working system with octupus 1.1 to v0.10.0-113-gd4aee4f5 my UART2(PD6/PD5) it was not working anymore.
I recompiled MCU and Pi... nothing helped
For Pi i used the "Linux process" and for mcu "STM32F446" display stayed dark...

mcu(stm32f446xx)
Version: v0.10.0-113-gd4aee4f5
 
Host(armv7l)
Version: v0.10.0-113-gd4aee4f5
OS: Raspbian GNU/Linux 10 (buster)

I have to switch to usb to get it back to life!
:-(

Turge08, 2021-11-07:
I don't have any issues with my Octopus 1.1 or Octopus Pro connected via UART.

What settings are you using to configure the firmware?

I'm using these: 
[image]

jrhelbert, 2021-11-08:
> After update my working system with octupus 1.1 to v0.10.0-113-gd4aee4f5 my UART2(PD6/PD5) it was not working anymore. I recompiled MCU and Pi... nothing helped For Pi i used the "Linux process" and [...]

Just updated as well, reflashed the FW via SD card and I am seeing the exact same behavior.

config settings are identical to what you posted aside from a bit of GPIO bit initialization
[image]

jrhelbert, 2021-11-08:
Attaching the relevant portion of my klippy.log fil
[klipper_mcu_error.log](https://github.com/Klipper3d/klipper/files/7494669/klipper_mcu_error.log)
e

jrhelbert, 2021-11-08:
Just noticed the "error: mcu 'rpi':" in the log I posted. Looks like this has nothing to do with the Octopus itself and it's actually having problems talking the the host mcu process.

closed 2021-12-20 by github-actions[bot]