Klipper3d/klipper #4898: MCU 'rpi': Command format mismatch

dvemerus, 2021-11-08:
After the update, it issues: 

"mcu 'rpi': Command format mismatch: config is_config=%c crc=%u is_shutdown=%c move_count=%hu vs config is_config=%c crc=%u move_count=%hu is_shutdown=%c

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

Known versions: host=v0.10.0-113-gd4aee4f5, mcu=v0.10.0-113-gd4aee4f5, rpi=v0.9.1-763-g3a497d04

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer."

Recompiling and flashing the firmware 
didn't solve this problem. 

Printer: FBG5. MKS Robin v1.1. Raspberry PI4
[klippy (2).log](https://github.com/Klipper3d/klipper/files/7499527/klippy.2.log)

Majawat, 2021-11-08:
I also updated today and got something similar:

```
mcu 'mcu': Command format mismatch: config is_config=%c crc=%u is_shutdown=%c move_count=%hu vs config is_config=%c crc=%u move_count=%hu is_shutdown=%c

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

Known versions: host=v0.10.0-113-gd4aee4f5, mcu=release-210618-2-g6e103c12

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer
```

And also re-compiled/flashed the firmware didn't change anything either.

Hardware: Creality Ender 5 Plus. Raspberry Pi 3B.

[moonraker.log](https://github.com/Klipper3d/klipper/files/7499797/moonraker.log) 
[klippy.log](https://github.com/Klipper3d/klipper/files/7499798/klippy.log)

ALeXXBody, 2021-11-08:
Same here:

`mcu 'rpi': Command format mismatch: config is_config=%c crc=%u is_shutdown=%c move_count=%hu vs config is_config=%c crc=%u move_count=%hu is_shutdown=%c

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

Known versions: host=v0.10.0-113-gd4aee4f5, mcu=v0.10.0-113-gd4aee4f5, rpi=v0.10.0-112-g9ecddd1b

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer`

Hardware: Voron 2.4 - BBT Octopus v1.1 & Raspberry Pi 3 B+

[klippy.log](https://github.com/Klipper3d/klipper/files/7499969/klippy.log)
[moonraker.log](https://github.com/Klipper3d/klipper/files/7499970/moonraker.log)

Majawat, 2021-11-08:
Ah! Mine was a dumb mistake. It had been a while since I flashed the firmware.

After making the klipper.bin on the RPi, I had forgotten to rename it to be firmware.bin on the flash drive I used to re-flash. Doing so correctly made it work again.

Not sure if that'll help others, but definitely making sure you're re-flashing correctly.

jakep82, 2021-11-08:
> This type of error is frequently caused by running an older version of the firmware on the micro-controller (fix by recompiling and flashing the firmware).
> 
> Known versions: host=v0.10.0-113-gd4aee4f5, mcu=v0.10.0-113-gd4aee4f5, rpi=v0.9.1-763-g3a497d04

The error message explains the problem.  

`Known versions:`
`host=v0.10.0-113`
`rpi=v0.9.1-763`

You have your Pi configured as a secondary MCU.  You need to update that firmware.

<https://www.klipper3d.org/RPi_microcontroller.html#building-the-micro-controller-code>

ALeXXBody, 2021-11-08:
Thanks @jakep82. Printer is up and running after updating rpi.

bassco, 2021-11-09:
@KevinOConnor 

flash_sd is unable to flash due to invalid mcu return parameters from  https://github.com/Klipper3d/klipper/commit/d4aee4f55e0203aa7cf2877227a574a1b1458a2e

The current mcu's are returning the old message format and the protocol test is failing.  Chicken and egg scenario. Can't ask for the new version if the new version has not been flashed to the devices yet.

`./scripts/flash-sdcard.sh /dev/ttyACM0 "btt-skr-v1.3"`

```text
...
Version: v0.10.0-116-ga0615e5e
  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
  Creating bin file out/klipper.bin
Flashing /home/pi/klipper/out/klipper.bin to /dev/ttyACM0
Checking FatFS CFFI Build...
Connecting to MCU...Connected
Checking Current MCU Configuration...
SD Card Flash Error: Command format mismatch: config is_config=%c crc=%u is_shutdown=%c move_count=%hu vs config is_config=%c crc=%u move_count=%hu is_shutdown=%c
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1160, in main
    spiflash.run()
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1109, in run
    self.run_reactor_task(self.run_reset)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1096, in run_reactor_task
    k_reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/reactor.py", line 48, in invoke
    res = self.callback(eventtime)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1063, in run_reset
    if self.mcu_conn.check_need_restart():
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 860, in check_need_restart
    self._serial, GET_CFG_CMD, GET_CFG_RESPONSE)
  File "/home/pi/klipper/klippy/mcu.py", line 483, in __init__
    serial.get_msgparser().lookup_command(respformat)
  File "/home/pi/klipper/klippy/msgproto.py", line 319, in lookup_command
    msgformat, mp.msgformat)
  File "/home/pi/klipper/klippy/msgproto.py", line 244, in _error
    raise error(self.warn_prefix + (msg % params))
error: Command format mismatch: config is_config=%c crc=%u is_shutdown=%c move_count=%hu vs config is_config=%c crc=%u move_count=%hu is_shutdown=%c
```

The msgproto is expecting the mcu to return the new format but the old mcu code is still presenting the old format - :chicken: and :egg: problem.

I hacked this by catching the exception and forcing an update:

```diff
diff --git a/scripts/spi_flash/spi_flash.py b/scripts/spi_flash/spi_flash.py
index 0c893ed2..21110719 100644
--- a/scripts/spi_flash/spi_flash.py
+++ b/scripts/spi_flash/spi_flash.py
@@ -856,13 +856,17 @@ class MCUConnection:

     def check_need_restart(self):
         output("Checking Current MCU Configuration...")
-        get_cfg_cmd = mcu.CommandQueryWrapper(
+        try:
+          get_cfg_cmd = mcu.CommandQueryWrapper(
             self._serial, GET_CFG_CMD, GET_CFG_RESPONSE)
-        params = get_cfg_cmd.send()
-        output_line("Done")
-        if params['is_config'] or params['is_shutdown']:
-            output_line("MCU needs restart: is_config=%d, is_shutdown=%d"
+
+          params = get_cfg_cmd.send()
+          output_line("Done")
+          if params['is_config'] or params['is_shutdown']:
+              output_line("MCU needs restart: is_config=%d, is_shutdown=%d"
                         % (params['is_config'], params['is_shutdown']))
+              return True
+        except:
             return True
         return False
```

I have a dirty version of the firmware and will now revert my changes and flash with a clean codebase.

The second flash attempt was successful.

```text
Version: v0.10.0-116-ga0615e5e
  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
  Creating bin file out/klipper.bin
Flashing /home/pi/klipper/out/klipper.bin to /dev/ttyACM0
Checking FatFS CFFI Build...
Connecting to MCU...Connected
Checking Current MCU Configuration...Done
MCU needs restart: is_config=1, is_shutdown=0
Attempting MCU Reset...Done
Waiting for device to reconnect....Done
Connecting to MCU...Connected
Initializing SD Card and Mounting file system...

SD Card Information:
Version: 2.0
SDHC/SDXC: False
Write Protected: False
Sectors: 245760
manufacturer_id: 0
oem_id: 0
product_name: APPSD
product_revision: 0.0
serial_number: 0000091E
manufacturing_date: 10/2020
capacity: 120.0 MiB
fs_type: FAT32
volume_label: SKR13
volume_serial: 251227765
Uploading Klipper Firmware to SD Card...Done
Validating Upload...Done
Firmware Upload Complete: firmware.bin, Size: 23284, Checksum (SHA1): 81ACF6C67710139DF6A82EB1822CC730F27BF074
Attempting MCU Reset...Done
Waiting for device to reconnect...Done
Connecting to MCU...Connected
Verifying Flash...Version matched...Done
Firmware Flash Successful
Current Firmware: v0.10.0-116-ga0615e5e
Attempting MCU Reset...Done
SD Card Flash Complete
```

dvemerus, 2021-11-11:
I solved the problem. When installed the input shaping sensor installed the secondary mcu. But I realized this when I reinstalled the firmware on raspberries and disabled the secondary mcu in the config. But, I think, you could just turn off secondary MCU in the config.

closed 2021-11-11 by dvemerus

meszoltan, 2021-12-31:
> I solved the problem. When installed the input shaping sensor installed the secondary mcu. But I realized this when I reinstalled the firmware on raspberries and disabled the secondary mcu in the config. But, I think, you could just turn off secondary MCU in the config.

Please ignore my incompetence, but your comment may help me. If I have a 0.1, a pi and one sku in it, shall I just turn off something in the config? How?