Klipper3d/klipper #4899: rotation_distance still dependent of microsteps?

ehtnevets, 2021-11-09:
[klippy.log](https://github.com/Klipper3d/klipper/files/7501368/klippy.log)

rotation_distance seems to be calculated based on microsteps: 16. My stepper_x axis only moves half the distance at microsteps: 8.

jakep82, 2021-11-09:
This is the expected behavior. Your x axis is running in standalone mode, so you need to set the microsteps setting to match how you've configured the driver with jumpers.

ehtnevets, 2021-11-09:
@jakep82 , thanks.

[image]

From the schematic for Minirambo, the microsteps is software adjustable (in case of X-axis it's PG0 & PG1).

How do I make Klipper adjust the 2 bits to match the microsteps config for Minirambo like the tmc driver.

jakep82, 2021-11-09:
This isn't the correct place for support. Please ask on Discord or Discourse.

ehtnevets, 2021-11-09:
Ok, thanks for the suggestion. 

I thought klipper set the MS1 & MS2 bits based on microsteps value for Minirambo integrated stepper drivers like it does for TMC driver.

closed 2021-12-22 by github-actions[bot]