Klipper3d/klipper #4902: Logo Run 2

charlespick, 2021-11-09:
Apparently I am not the first to try this #822, but I was thinking it's worth considering a new logo for Klipper again since it's been a while. 

I'm not sure who designed the current logo, but I think Klipper has reached a point in the past year where it really has it's own identity as a forward-thinking and progressive project that represents a community that's dedicated to bleeding edge hobbiest 3D printing and as such, I think it could use a new logo that communicates that attitude. 

I have about 20 more iterations but I think these are my best candidates: 

[image]

I'd like to know if anyone else has some other ideas as well

KevinOConnor, 2021-12-15:
Interesting.  FWIW, I'll leave it up to the Klipper community to decide if a logo change should be made.  (If lots of people indicate they prefer a new logo then I'm happy to make a change.)

Thanks,
-Kevin

charlespick, 2021-12-15:
Thanks Kevin! 

Where do you think would be the most visible place to hold a vote?

Sineos, 2022-01-10:
Maybe bringing it up on https://klipper.discourse.group will attract a greater audience. Also allows for polls. See https://meta.discourse.org/t/how-to-create-polls/77548

closed 2022-02-22 by github-actions[bot]