Klipper3d/klipper #4906: Klipper MCU Error while using active usb cable

bcrowie, 2021-11-10:
I run Klipper, Mainsail and Fluidd in a Debian virtual machine on my home server and have been doing so for the last 12 or so months. I am now looking to relocate the server across to room and purchased an active usb cable to compensate for the USB length limits.

After connecting the cable I am now receiving the following message in Mainsail, Fluidd and Klipper:

```
mcu 'mcu': Command format mismatch: config is_config=%c crc=%u is_shutdown=%c move_count=%hu vs config is_config=%c crc=%u move_count=%hu is_shutdown=%c

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

Known versions: host=v0.10.0-113-gd4aee4f5, mcu=v0.9.1-621-gdafb74e3

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer
```

I've never seen this before. After switching back to a basic cable I have no issues. When listing /dev/serial/by-id/* in Debian the same device as always has continues to show. 

I have attached my klippy.log as instructed. Thank you for any assistance.

[klippy.log](https://github.com/Klipper3d/klipper/files/7508838/klippy.log)

zellneralex, 2021-11-10:
Read this again:

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

You did update your klipper on the rpi but these time you need also to update your mcu board

charnwooduk, 2021-11-10:
i am getting a very similar version mismatch. It does not make sense as it is a completely new installation on the PI and a brand new printer (ender3 v2)  . I've already got 4x of the same setup installed a few weeks ago with no issues

Known versions: host=v0.10.0-118-g59314d99, mcu=v0.9.1-773-gc84956be

Using kiauh for the install on Pi Zero 2 W

zellneralex, 2021-11-10:
It’s not that hard … look at your versions 

host=v0.10.0-118-g59314d99, 
mcu=v0.9.1-773-gc84956be

so please update your mcu than it will match and work again

bcrowie, 2021-11-10:
I see. Sorry for the bother. It was printing fine the day before. I had just assumed it was fine until I used that cable. I greatly appreciate your help clearing it up!!

bcrowie, 2021-11-10:
> It’s not that hard … look at your versions 
> 
> host=v0.10.0-118-g59314d99, 
> mcu=v0.9.1-773-gc84956be
> 
> so please update your mcu than it will match and work again 

This resolved the issue. Thank you so much, once again.

For anyone with an Aquila, use the Ender 3 V2 method but put the new firmware into a folder on the SD titled 'firmware' then boot the printer with the SD card.

jottr, 2021-11-21:
@bcrowie since your issue has been solved, please close it.

closed 2021-11-21 by bcrowie