Klipper3d/klipper #4907: Option 'pin_map' is not valid in section 'mcu'

CiarlilloM, 2021-11-10:
Okay, so first off let me say I'm a complete noob when it comes to Klipper, Fluidd, etc. 

Printer:
CR-10S
Micro Swiss Direct Drive Extruder
Ender 5 Plus Silent Board

Initially, I configured Klipper & Fluidd correctly using a Teaching Tech YouTube as a guide along with Creality & Marlin RAMPS / Pinouts along with Configuration Guide, tests & checks, etc . EVERYTHING worked PERFECTLY after switching/reversing the direction of some pins (+/- !) using the Creality CR-10S template from the Official Github. The problem came about after I tried installing/adding a Genuine BLTouch V3.1.  I tried to copy + paste the BLTouch settings in my printer.cfg and that gave me the above-mentioned error. 

So I decided to use the Creality Ender 5 Plus template off the Official Github, and modified everything by changing the pin values to match what I had originally found to work from m
[moonraker.log](https://github.com/Klipper3d/klipper/files/7508879/moonraker.log)
y modified CR-10S template. This also gave me the "Option 'pin_map' is not valid in section 'mcu'" error.

Frustrated, and needing to print something I decided to manually level the bed, and copy over the backup 'printer.cfg" file -that worked.... yeah that even gave me the error after trying to "Restart Klipper" & "Firmware Restart" a million times. 
[CR10S_Template_printer.txt](https://github.com/Klipper3d/klipper/files/7508946/CR10S_Template_printer.txt)
[Ender5Plus_Template_printer.txt](https://github.com/Klipper3d/klipper/files/7508947/Ender5Plus_Template_printer.txt)

I've attached my printer.cfg file that used the CR-10S template, the printer.cfg file that uses the Ender 5 Plus template, the klippy.log & moonraker.log. 

Thank you in advance for the help, and please feel free to help change any of the BLTouch configurations since I'm a complete NOOB.

Sineos, 2021-11-10:
Replace the `pin_map` setting by the appropriate alias definition here https://github.com/Klipper3d/klipper/blob/master/config/sample-aliases.cfg

wizhippo, 2021-11-21:
Use only one of the sections mentioned in https://github.com/Klipper3d/klipper/blob/master/config/sample-aliases.cfg other wise you will get error messages because you are trying to map the same alias multiple times.

Sineos, 2021-12-30:
If this is still an issue please open a new post on https://klipper.discourse.group and we will be happy to assist

closed 2021-12-30 by Sineos