Klipper3d/klipper #4910: mcu 'mcu': Unable to connect after power cycle

that-should-work, 2021-11-10:
[klippy.log](https://github.com/Klipper3d/klipper/files/7514034/klippy.log)

Klipper/Fluidd has been running fine until the printer got power cycled

mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

Tried using 
ls /dev/serial/by-id/*
but no such file or directory

This is not the first time it's been power cycled, so I'm not sure what direction to take.

MrSkrifle, 2021-11-11:
Try running:
cd ~/klipper/

Before:
ls /dev/serial/by-id/*

It should show then, but probably it hasn't changed. 

Try disconnecting power to pi and printer, and then try to power on. Repeat a few times. I get the same issue too often if I restart pi or klipper firmware through octoprint.

that-should-work, 2021-11-11:
So I got it figured out just last night.
Changing to the klipper folder didn't work btw

I turned off power to the printer then connected the pi which powered the board. I was only able to connect then and I reflashed the board to make sure it was on the same page with all the klipper updates that the pi was on. Did a firmware restart and we're good to go.

Mostly...
pin_map is no longer used, but klipper has the Marlin pin conversion table for most boards. I used that to clean up the config file.

closed 2021-11-11 by that-should-work