Klipper3d/klipper #4916: Klippy not connected

ilpingu, 2021-11-12:
Hi, my pi is not more able to connect to the printer after a restart, before it worked all correct, i've tryed to reinstall fluidd and reflash the firmware, but nothing worked, thanks in advance for the help.

my klippy log

=============== Log rollover at Fri Nov 12 16:56:32 2021 ===============
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/klipper_config/printer.cfg', '-l', '/home/pi/klipper_logs/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'release-210618-2-g6e103c12'
CPU: 4 core ARMv7 Processor rev 3 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Building C code module c_helper.so
Start printer at Fri Nov 12 16:56:42 2021 (1636736202.4 29.4)
Unable to open config file /home/pi/klipper_config/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 109, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/klipper_config/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 157, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 134, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 226, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 115, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/klipper_config/printer.cfg
webhooks client 3046207576: New connection
webhooks client 3046207576: Client info {'program': 'Moonraker', 'version': 'v0.7.1-2-gdf43aca'}
webhooks client 3046207576: Disconnected
Restarting printer
Start printer at Fri Nov 12 17:15:53 2021 (1636737353.4 129.1)
Unable to open config file /home/pi/klipper_config/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 109, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/klipper_config/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 157, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 134, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 226, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 115, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/klipper_config/printer.cfg
webhooks client 3046242784: New connection
webhooks client 3046242784: Client info {'program': 'Moonraker', 'version': 'v0.7.1-2-gdf43aca'}
webhooks client 3046242784: Disconnected
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/klipper_config/printer.cfg', '-l', '/home/pi/klipper_logs/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'release-210618-2-g6e103c12'
CPU: 4 core ARMv7 Processor rev 3 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Building C code module c_helper.so
Start printer at Fri Nov 12 17:16:08 2021 (1636737368.8 144.6)
Unable to open config file /home/pi/klipper_config/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 109, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/klipper_config/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 157, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 134, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 226, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 115, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/klipper_config/printer.cfg
webhooks client 3045222120: New connection
webhooks client 3045222120: Client info {'program': 'Moonraker', 'version': 'v0.7.1-2-gdf43aca'}
webhooks client 3045222120: Disconnected
Restarting printer
Start printer at Fri Nov 12 17:16:14 2021 (1636737374.2 149.9)
UStarting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/klipper_config/printer.cfg', '-l', '/home/pi/klipper_logs/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'release-210618-2-g6e103c12'
CPU: 4 core ARMv7 Processor rev 3 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/klipper_config/printer.cfg', '-l', '/home/pi/klipper_logs/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'release-210618-2-g6e103c12'
CPU: 4 core ARMv7 Processor rev 3 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/klipper_config/printer.cfg', '-l', '/home/pi/klipper_logs/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'v0.10.0-122-gc179db3d'
CPU: 4 core ARMv7 Processor rev 3 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Building C code module c_helper.so
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/klipper_config/printer.cfg', '-l', '/home/pi/klipper_logs/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'v0.10.0-122-gc179db3d'
CPU: 4 core ARMv7 Processor rev 3 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/klipper_config/printer.cfg', '-l', '/home/pi/klipper_logs/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'v0.10.0-122-gc179db3d'
CPU: 4 core ARMv7 Processor rev 3 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'

theophile, 2021-11-12:
Those errors are telling you that the config files do not exist where klipper is looking for them. What is the output of `ls -l /home/pi/klipper_config/` ?

ilpingu, 2021-11-12:
> Those errors are telling you that the config files do not exist where klipper is looking for them. What is the output of `ls -l /home/pi/klipper_config/` ?

hi, thank you for the reply, 
this is the output:
total 20
-rw-r--r-- 1 pi pi 2125 Jul  9 19:13  fluidd.cfg
-rw-r--r-- 1 pi pi  695 Jul  9 19:13  moonraker.conf
-rw-r--r-- 1 pi pi 6754 Nov 12 22:36 'printer .cfg'
-rw-r--r-- 1 pi pi 2591 Jul  9 19:13  webcam.txt

BlackStump, 2021-11-12:
you have a space after printer   .cfg it should be printer.cfg

ilpingu, 2021-11-12:
> you have a space after printer ie printer .cfg it should be printer.cfg

thank you for yuour reply, i have tryed, but it doesnt still work

ilpingu, 2021-11-12:
pi@fluiddpi:~ $ ls -l /home/pi/klipper_config/
total 20
-rw-r--r-- 1 pi pi 2125 Jul  9 19:13 fluidd.cfg
-rw-r--r-- 1 pi pi  695 Jul  9 19:13 moonraker.conf
-rw-r--r-- 1 pi pi 6754 Nov 12  2021 printer.cfg
-rw-r--r-- 1 pi pi 2591 Jul  9 19:13 webcam.txt

BlackStump, 2021-11-12:
you should seek help on discord or discourse as per [instructions](https://github.com/Klipper3d/klipper/blob/master/docs/Contact.md) before opening a ticket

ilpingu, 2021-11-12:
ok, ill try there, thank you

closed 2021-12-25 by github-actions[bot]