Klipper3d/klipper #4920: Random shutdowns

bluepinner, 2021-11-13:
I’ve been having random shutdowns for the last 2 weeks. I thought it was the idle time out but I’m still getting them after removing it from the config and restarting firmware. I tried looking at the Klippy.log but I’m understanding it. Ender 3 V2 w/4.2.2 and rpi 4
[klippy.log](https://github.com/Klipper3d/klipper/files/7532861/klippy.log)

Jaans, 2021-11-20:
Had some similar random shutdowns of my PI 4. Are you running the latest version of Raspberry PI OS `Bullseye`? (Or still the previous one `Buster`)

bluepinner, 2021-11-20:
I was running the current mainsailOS. I have since switched to fluiddpi. Things are running smooth for now.

Jaans, 2021-11-22:
Update to my previous post. Most recently, the UI (both Fluidd / Mainsail) stopped working due to not being able to connect to Moonraker API anymore. The printer was busy printing at the time and much to my surprise didn't stop printing, and finished it's print without issue later on.

Eventually, the UI (both Fluidd & Mainsail) came good and started working again, as if nothing crashed / stopped.

Will the Klipper auto updater update the components (e.g. Moonraker) during an active print?

closed 2021-12-23 by bluepinner