Klipper3d/klipper #4921: MAXxxxxx temperature sensors section in Config reference rtd_reference_r: 400 instead of 430

sprior, 2021-11-14:
It seems that while rtd_reference_r: 430 might have been correct for SKR boards, the correct value seems to be closer to 400 for the Spyder and Octopus boards people are using today.

I set up my Voron 2.4 with a PT100 stick with a Spyder board and ended up troubleshooting extruder clicking with the documented settings. Only when I decided the issue was lower than reported extruder temp did I find out the rtd_reference_r config should be 400 instead of 430.  Maybe the comments should reflect this.

closed 2021-12-27 by github-actions[bot]

viesturz, 2022-03-14:
+1 this is also what Duet expansion board uses (MAX31865)

sprior, 2022-03-14:
That's why I suggested enhancing the comments to reflect the options that seem to be out in the wild these days.