Klipper3d/klipper #4930: Evaluation of variables leads to lowercase conversion

Roycinger, 2021-11-16:
Today I noticed that evaluation of macro variables leads to an unwanted lowercase conversion.
To be specific:
I created a variable within a VARIABLES macro (sort of a container):
`variable_ABS_temp_hotend: 245`
Notice the "ABS" in uppercase.
Then I wanted to use that variable within another macro:
`{% set hotend_temp = printer["gcode_macro VARIABLES"].ABS_temp_hotend|float %}`
Which lead to the following error message:
`Error evaluating 'gcode_macro LOAD_FILAMENT:gcode': UndefinedError: 'dict object' has no attribute 'ABS_temp_hotend'`

After some headscratching and talking to some people, I looked at the output of my VARIABLES macro:
[image]
I noticed that "ABS" is stated in lowercase here!

When using lowercase in my macros section, all works out fine.

Conclusion:
Issue is already solved, just making aware of a potential issue with an unwanted upper- to lowercase conversion where it's not specifically asked for.

For complying with the issue rules, here's the klippy log:
[klippy_20211116.log](https://github.com/Klipper3d/klipper/files/7548909/klippy_20211116.log)

Sineos, 2021-11-17:
See https://www.klipper3d.org/Command_Templates.html#variables
Upper case is variable names is not supported

Roycinger, 2021-11-17:
Must have read over that. Thanks for clarification!

closed 2021-11-17 by Roycinger