Klipper3d/klipper #4933: Lose Connection with Duet 2 WiFi after flashing

armysolo, 2021-11-17:
I was having issues with Moonraker so I decided to create a new Klipper USB(I like to use USB drives because their faster and more stable than SD cards). I have a RPI 4 8gb, 32gb USB 3.1 flash drive and Duet 2 WiFi. I used the kiauh.sh script to install Klipper, Fluid and Moonraker.

`ls /dev/serial/by-id/* 
/dev/serial/by-id/usb-03eb_6124-if00
`

 I'm able to flash the Duet but after flashing it seems like the board becomes locked. When I re-run `ls /dev/serial/by-id/` I get an error for "ls: cannot access '/dev/serial/by-id/*': No such file or directory". If I erase the flash on the Duet the RPI can communicate with it again. I've tried reflashing the Duet and the RPI many times but I'm not sure what else I can do to fix this.

[klippy (1).log](https://github.com/Klipper3d/klipper/files/7550757/klippy.1.log)

armysolo, 2021-11-17:
After swapping USB cables and re-flashing this is what the output is.

`/dev/serial/by-id/usb-03eb_6124-if00
ls /dev/serial/by-id/* 
/dev/serial/by-id/usb-03eb_6124-if00
make flash FLASH_DEVICE=/dev/serial/by-id/usb-03eb_6124-if00

pi@raspberrypi:~ $ ls /dev/serial/by-id/*                                                                          
/dev/serial/by-id/usb-03eb_6124-if00                                                                               
pi@raspberrypi:~ $ ls /dev                                                                                         
autofs         hwrng         null   ram9            stdin   tty24  tty42  tty60      vcs2      vga_arbiter         
block          initctl       port   random          stdout  tty25  tty43  tty61      vcs3      vhci                
bsg            input         ppp    raw             tty     tty26  tty44  tty62      vcs4      video10             
btrfs-control  kmsg          ptmx   rfkill          tty0    tty27  tty45  tty63      vcs5      video11             
bus            log           pts    rpivid-h264mem  tty1    tty28  tty46  tty7       vcs6      video12             
cachefiles     loop0         ram0   rpivid-hevcmem  tty10   tty29  tty47  tty8       vcsa      video13             
cec0           loop1         ram1   rpivid-intcmem  tty11   tty3   tty48  tty9       vcsa1     video14             
cec1           loop2         ram10  rpivid-vp9mem   tty12   tty30  tty49  ttyACM0    vcsa2     video15             
char           loop3         ram11  sda             tty13   tty31  tty5   ttyAMA0    vcsa3     video16             
console        loop4         ram12  sda1            tty14   tty32  tty50  ttyprintk  vcsa4     video18             
cuse           loop5         ram13  sda2            tty15   tty33  tty51  ttyS0      vcsa5     watchdog            
disk           loop6         ram14  serial          tty16   tty34  tty52  uhid       vcsa6     watchdog0           
dma_heap       loop7         ram15  serial0         tty17   tty35  tty53  uinput     vcsm-cma  zero                
dri            loop-control  ram2   serial1         tty18   tty36  tty54  urandom    vcsu                          
fd             mapper        ram3   sg0             tty19   tty37  tty55  v4l        vcsu1                         
full           media0        ram4   shm             tty2    tty38  tty56  vchiq      vcsu2                         
fuse           media1        ram5   snd             tty20   tty39  tty57  vcio       vcsu3                         
gpiochip0      mem           ram6   spidev0.0       tty21   tty4   tty58  vc-mem     vcsu4                         
gpiochip1      mqueue        ram7   spidev0.1       tty22   tty40  tty59  vcs        vcsu5                         
gpiomem        net           ram8   stderr          tty23   tty41  tty6   vcs1       vcsu6                         
pi@raspberrypi:~ $ ls /dev/serial                                                                                  
by-id  by-path                                                                                                     
pi@raspberrypi:~ $ ls /dev/serial/by-id                                                                            
usb-03eb_6124-if00                                                                                                 
pi@raspberrypi:~ $ lsusb                                                                                           
Bus 002 Device 002: ID 0781:5583 SanDisk Corp. Ultra Fit                                                           
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub                                                     
Bus 001 Device 007: ID 03eb:6124 Atmel Corp. at91sam SAMBA bootloader                                              
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub                                                                
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub                                                     
pi@raspberrypi:~ $ cd klipper                                                                                      
pi@raspberrypi:~/klipper $ make clean                                                                              
pi@raspberrypi:~/klipper $ make menuconfig                                                                         
  Creating symbolic link out/board                                                                                 
Loaded configuration '/home/pi/klipper/.config'                                                                    
No changes to save (for '/home/pi/klipper/.config')                                                                
pi@raspberrypi:~/klipper $ sudo service klipper stop
pi@raspberrypi:~/klipper $ make                                                                                    
  Building out/autoconf.h                                                                                          
  Compiling out/src/sched.o                                                                                        
  Compiling out/src/command.o                                                                                      
  Compiling out/src/basecmd.o                                                                                      
[... cut ...]
Bus 002 Device 002: ID 0781:5583 SanDisk Corp. Ultra Fit                                                           
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub                                                     
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub                                                                
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub                                                     
`

It seems the USB isn't recognized after flashing the Duet and RPI MCU and reboot.

armysolo, 2021-11-18:
It seems it's an issue with Klipper. I've tried cutting the 5v wire on the USB to ensure the Duet isn't being powered from the RPI.

I've tried with 3 different Duet boards and 2 different RPIs along with trying to manually flash with my PC using Bossa. Every way I've tried I get the same result. The board cannot communicate once it's flashed. But if I erase the firmware it can communicate.

Sineos, 2021-11-18:
Try this approach: https://duet3d.dozuki.com/Wiki/Installing_and_Updating_Firmware#Section_All_other_Duet_boards

armysolo, 2021-11-18:
I've already tried that

zueski, 2021-11-22:
I am having the same issue on a fresh install on a new pi4 with the latest on debian.  I had a working install on a different pi that will not work now even after reflashing without rebuilding.  The interesting part is the serial ID changed from:
/dev/serial/by-id/usb-Klipper_sam4e8e_00313753595252533136303532303436-if00
to a non-working:
/dev/serial/by-id/usb-03eb_6124-if00
I have not been able to get it back up after the first flash on the latest commit from Nov 21

p5y, 2021-11-28:
I am in the same boat as above.
Was running 0.9.1-something before, that worked as intended. 
Created a fresh raspbian-bullseye SD and used kiauh to set things up. I was able to see the Duet when the 0.9.1-something firmware was on the board, but after flashing the new firmware the board wouldn't boot properly.

I tried erasing by jumper, and also flashing the firmware with bossa from windows. This did not help.
Going back to the old SD and flashing would fix things.
Switching to a SKR mini e3 and things ran in no time.

Unfortunately I don't have the time nor tools to debug the issue right now.
Hopefully this info will help someone along.

p5y, 2021-12-08:
I'm not sure how to approach debugging this board. How do I easily debug the code without JTAG? I can't find anything in the klipper developer docs or Duet wiki that can help me along either. @KevinOConnor can you maybe give me some pointers?

Stepping through klipper revisions to see where it breaks is a fallback, but that's a tedious job.

p5y, 2021-12-13:
I played with this issue a bit this weekend.
- Rolled back to master branch nov19 on rpi4/raspbian buster, flashing works and can connect to Duet2 with klipper. 
- Setup a dev environment on Ubuntu 20.04 with latest klipper master, flashing works and can connect to Duet2 with klipper. 
I'm not exactly sure what is going wrong here. It seems raspbian bullseye is somehow involved. Ubuntu 20.04, however, is also technically bullseye.

danielschramm, 2022-01-08:
Hello,

I also like to confirm this issue in a similar setup. 
I had to move to RPI with USB Boot this week and I also have Issues.

Board reset to bootloader mode via Erase jumper is OK:

```
usb 1-1.4: new full-speed USB device number 43 using xhci_hcd
usb 1-1.4: New USB device found, idVendor=03eb, idProduct=6124, bcdDevice= 1.10
usb 1-1.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device

```

Flashing via make flash works without issues.

And than I just get an error:

```
usb 1-1.4: new full-speed USB device number 59 using xhci_hcd
usb 1-1.4: unable to read config index 0 descriptor/all
usb 1-1.4: can't read configurations, error -110
usb 1-1-port4: unable to enumerate USB device
```

Lastest revision from git:

```
daniel@raspberrypi:~/klipper $ git show
commit 4c8d24ae03eadf3fc5a28efb1209ce810251d02d (HEAD -> master, origin/master, origin/HEAD)
Author: Kevin O'Connor <kevin@koconnor.net>
Date:   Fri Jan 7 22:17:26 2022 -0500

    stm32: Update Kconfig as CANBUS isn't available on stm32f401

    Signed-off-by: Kevin O'Connor <kevin@koconnor.net>

```

danielschramm, 2022-01-08:
I just repeated everything with Ubuntu 20.10 on the same setup and everything is working as expected .
So it really seems that the build process of the duet2 firmware is just failing on the current raspbian.

Zodiak1993, 2022-02-16:
I have the same problem with a Duet Wifi. Really dont know what to do now...

Sineos, 2022-02-16:
> I just repeated everything with Ubuntu 20.10 on the same setup and everything is working as expected .
> So it really seems that the build process of the duet2 firmware is just failing on the current raspbian.

@Zodiak1993 see the reply just above by @danielschramm

Zodiak1993, 2022-02-16:
Sorry I am new to Linux and Raspberries - Ubuntu is an alternative to PI OS Lite?
Why is this happening, that the build process fails ?

Sineos, 2022-02-16:
Yes, Ubuntu is another Linux OS and can replace PI OS.
Reason why this is happening is not clear. As Klipper is depending on certain packages that come with the Linux OS, there might be some unwanted interaction.

Zodiak1993, 2022-02-16:
So for everybody - I could solve it with the "Mainsail OS"
https://docs.mainsail.xyz/setup/mainsail-os

Thank you for your support.

Sineos, 2022-02-18:
Topics seems concluded. Apparently some OS issue and not depending on Klipper.

ccollinscj, 2022-04-25:
The problem still exists.

mak0t0san, 2022-07-08:
> So for everybody - I could solve it with the "Mainsail OS" https://docs.mainsail.xyz/setup/mainsail-os
> 
> Thank you for your support.

Had this same issue. Could not get a working setup with the latest Bullseye version of Raspbian.  Using MainsailOS , which uses Buster, works fine.

DefektSan, 2022-08-13:
Actual problem. 
When compile fw for SAM4E8E mcu in Debian 11 Bullseye, mcu does not stars and still place in DFU mode. 
Fw compiled  in Debian 10 buster works normal.

gjsmo, 2022-08-30:
@Sineos Can this issue please be re-opened? I can reliably reproduce this on a brand new installation of Debian 11, with only Klipper installed. This is the latest version of Raspberry Pi OS, so it's not a particularly unique setup. Yes, it does change based on OS, bu that isn't evidence that the OS itself is the issue.

AlexTOOT, 2022-11-08:
I got same issue on Debian11 using bigtreetech CB1 and Duet2 wifi. 
before flashing, it shows up as  usb-03eb_6124-if00,
after flashing, duet goes offline, can not find anything in /dev/serial/by-id
switching os version do solve the problem but still wanna fix the issue

viesturz, 2023-04-01:
Still a problem.

underpk, 2023-06-17:
Anyone know how to solve this problem? I just have the same issue.

Sineos, 2023-06-17:
See https://klipper.discourse.group/t/duet-2-firmware-compilation-broken-on-debian-11/4762/

Only known fix is to use a different Linux version for compiling and flashing

lesaux, 2023-09-11:
Sep 11 2023, I needed to use the ubuntu-server 20.04 image to be able to flash my duet2 wifi + duext5. Tried raspian buster-lite, mainsailos and neither would work.

DefektSan, 2023-09-12:
> Tried raspian buster-lite, mainsailos and neither would work.

You need use raspberry Pi OS Lite (Legacy) on Debian 10 (buster), on this version firmware compile correctly and work fine.