Klipper3d/klipper #4936: MCU Fails to Connect to Klipper

abdurrehmanzea, 2021-11-17:
After an update to Klipper the MCU fails to connect. I had been using Klipper for a couple of months, it had been working excellent.

Hardware: Ender 3 v2, Raspberry Pi 4 
After connection with printer following error appears

> mcu 'mcu': Command format mismatch: config is_config=%c crc=%u is_shutdown=%c move_count=%hu vs config is_config=%c crc=%u move_count=%hu is_shutdown=%c
> 
> This type of error is frequently caused by running an older
> version of the firmware on the micro-controller (fix by
> recompiling and flashing the firmware).
> 
> Known versions: host=v0.10.0-130-g68c92991, mcu=v0.10.0-112-g9ecddd1b
> 
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Protocol error connecting to printer

I've done following to correct the issue without any positive result:

1. Use an earlier config file 
2. Update klipper/moonraker/fluidd/OS Packages
3. Update printer firmware after creating new bin file
4. Clean install Raspi image
5. Update printer firmware after creating new bin file
6. Clean install using Octoprint image (isolating issue from Fluidd/KIAUH setup)
7. Update printer firmware after creating new bin file

The host and mcu versions are host=v0.10.0-130-g68c92991, mcu=v0.10.0-112-g9ecddd1b and are not very far off and should be working. 

Help!

[klippy.log](https://github.com/Klipper3d/klipper/files/7557067/klippy.log)

Sineos, 2021-11-18:
> Known versions: host=v0.10.0-130-g68c92991, mcu=v0.10.0-112-g9ecddd1b
> 
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Protocol error connecting to printer

> The host and mcu versions are host=v0.10.0-130-g68c92991, mcu=v0.10.0-112-g9ecddd1b and are not very far off and should be working.

Well, Klipper is even telling you, that "not very far off" is too far in this case. Update your MCU

abdurrehmanzea, 2021-11-18:
@Sineos after your suggestion, I tried to update the firmware of MCU but couldn't update it for some reason. Therefore downgraded the Klipper version to match that of mcu and viola, it works now.

Problem has to do with process of upgrade using SD card. Which I'll sort out separately. 

Klipper is working as good as ever.

closed 2021-11-18 by abdurrehmanzea

josephchh, 2022-01-10:
I have the same issue, how to downgrade Klipper?

abdurrehmanzea, 2022-01-11:
> I have the same issue, how to downgrade Klipper?

Downgrading the firmware is only a temporary solution. The problem I was facing was firmware upgrade on the printer. I was using 64GB SDCard which didn't seem to work. Swapped it with 8GB and it now works flawlessly. 

Below are the commands to downgrade to earlier release, The bold part is the release you want to downgrade to.

cd ~/klipper
git fetch
git checkout **_9ecddd1ba4fe6d481f01fa2b1d9175b927907156_**
sudo service klipper restart

josephchh, 2022-01-11:
> Downgrading the firmware is only a temporary solution. The problem I was facing was firmware upgrade on the printer. I was using 64GB SDCard which didn't seem to work. Swapped it with 8GB and it now [...]

Thank you!
Yes, I also want to fix the firmware problem when upgrading the system.  Just want to make the print work again for some urgent printing.  
For your case, I think you may need replace a new 64GB SDCard.