Klipper3d/klipper #4937: Unable to use adxl because of missing packages

definitely-not-a-t-rex, 2021-11-17:
hi, I have klipper setup on zero 2 w running bullseye and I tried installing the software requirements for using the adxl345 to calibrate resonance compensation for my printer following [https://www.klipper3d.org/Measuring_Resonances.html](url), it says to install numpy and matplotlib 

```
 sudo apt update
 sudo apt install python-numpy python-matplotlib
```

however the pi says that it's either missing or obsolete 
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-matplotlib is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package python-numpy is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-numpy' has no installation candidate
E: Package 'python-matplotlib' has no installation candidate

```

is it because of bullseye being relatively recent or because klipper hasn't yet moved from the long unsupported python 2? Can I fix this? It was working on buster (a year old install so something might have changed since) on my pi4 but I got the zero 2w to use the 4 for other tasks

phil65, 2021-11-17:
you could try "python3-numpy" and "python3-matplotlib" as package name.

majcn, 2021-11-18:
As I understood from recent commits, klipper should run on python3, but install scripts are still written for python2

Sineos, 2021-11-18:
Klipper runs in an virtualenv using its own Python. 
The command `sudo apt install python-numpy python-matplotlib` installs Python packages on OS level.
On newer distribution you will only have Python 3 as default and the solution of @phil65 is correct

definitely-not-a-t-rex, 2021-11-18:
> Klipper runs in an virtualenv using its own Python. The command `sudo apt install python-numpy python-matplotlib` installs Python packages on OS level. On newer distribution you will only have Python 3 as default and the solution of @phil65 is correct

that didn't really work, at least not to create the usual graphs as even after installing NumPy with python3 it said it didn't have it, don't really need it tho

Jaans, 2021-11-20:
I ran into the same issue, with a fresh install on Raspberry PI 4B with Bullseye and Klipper et. al. (via KIAUH).

Installed python3 versions as per @phil65 suggestion:
```
sudo apt install python3-numpy python3-matplotlib
```

That in and of itself didn't resolve the issue. I took a shot in the dark by editing the `calibrate_shaper.py` script and changing the  environment to Python 3:
[image]

And to my surprise that worked! (or at least it seems to have):
[image]

Sineos, 2021-11-20:
No need to edit the script. Just call the script via:

```
python3 ~/klipper/scripts/calibrate_shaper.py ...
```

Sineos, 2021-12-26:
https://github.com/Klipper3d/klipper/pull/5011

closed 2021-12-26 by Sineos

koshisan, 2021-12-29:
Same problem with my 2-day-old installation here. Using KIAUH on Raspbian 11 - which is as supported as it gets, I guess -> There should be at least an update to the documentation IMHO