Klipper3d/klipper #4938: Firmware no longer builds for Atmega 328p

cbc02009, 2021-11-17:
Looks like the image is too big for flash. Not sure what commit exactly caused this.

``` 
 Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/spi_software.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/avr/main.o
  Compiling out/src/avr/timer.o
  Compiling out/src/avr/gpio.o
  Compiling out/src/avr/adc.o
  Compiling out/src/avr/spi.o
  Compiling out/src/avr/i2c.o
  Compiling out/src/avr/hard_pwm.o
  Compiling out/src/avr/watchdog.o
  Compiling out/src/avr/serial.o
  Compiling out/src/generic/serial_irq.o
  Building out/compile_time_request.o
Version: v0.10.0-130-g68c92991
  Linking out/klipper.elf
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800a2e of out/klipper.elf section `.data' is not within region `data'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800b98 of out/klipper.elf section `.bss' is not within region `data'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800a2e of out/klipper.elf section `.data' is not within region `data'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800b98 of out/klipper.elf section `.bss' is not within region `data'
collect2: error: ld returned 1 exit status
make: *** [Makefile:72: out/klipper.elf] Error 1
```

Log is irrelevant but added anyway.

[klippy.log](https://github.com/Klipper3d/klipper/files/7557601/klippy.log)

cbc02009, 2021-11-17:
Sorry, forgot to upload the kconfig file.

[config.txt](https://github.com/Klipper3d/klipper/files/7558152/config.txt)

cbc02009, 2021-11-18:
Okay a bit more information. I have two separate klipper instances to compare side-by-side. One is a fresh mainsailOS install and the second is my current ubuntu 21.10 setup. Both are running identical klipper commits.

MainsailOS utility versions:
gcc-avr: 1:5.4.0+Atmel3.6.1-2

Ubuntu utility versions:
gcc-avr: 1:5.4.0+Atmel3.6.2-1

MainsailOS memory configuration:
```
Name             Origin             Length             Attributes
text             0x00000000         0x00020000         xr
data             0x00800060         0x0000ffa0         rw !x
eeprom           0x00810000         0x00010000         rw !x
fuse             0x00820000         0x00000003         rw !x
lock             0x00830000         0x00000400         rw !x
signature        0x00840000         0x00000400         rw !x
user_signatures  0x00850000         0x00000400         rw !x
*default*        0x00000000         0xffffffff
```

Ubuntu Memory Configuration:
```
Name             Origin             Length             Attributes
text             0x0000000000000000 0x0000000000008000 xr
data             0x0000000000800100 0x0000000000000800 rw !x
eeprom           0x0000000000810000 0x0000000000000400 rw !x
fuse             0x0000000000820000 0x0000000000000003 rw !x
lock             0x0000000000830000 0x0000000000000400 rw !x
signature        0x0000000000840000 0x0000000000000400 rw !x
user_signatures  0x0000000000850000 0x0000000000000400 rw !x
*default*        0x0000000000000000 0xffffffffffffffff
```

Mainsail .text size: 0x6488 (25,736)
Ubuntu .text size: 0x76e8 (30, 440)

Mainsail .data size: 0x2a (42)
Ubuntu .data size: 0x92E (2,350)

Here are the map files for comparison:

[Ubuntu.map.txt](https://github.com/Klipper3d/klipper/files/7563954/Ubuntu.map.txt)

[Mainsail.map.txt](https://github.com/Klipper3d/klipper/files/7563960/Mainsail.map.txt)

Sineos, 2021-11-18:
https://linux.debian.bugs.dist.narkive.com/xGY1AKa0/bug-985578-arduino-core-avr-flashing-bootloader-fails-due-to-invalid-size

Related? If yes this would mean it is an OS / Linux issue on Bullseye

Itay-Sharoni, 2021-11-22:
same issue for me,
I am using Clean Debian on Rpi4 (no mainsailOS), and also have Arduino Nano for ADXL, firmware install to nano (328p) not working, getting same errors.

as a workaround, I flashed the HEX file using Windows with xLoader_V1_339 (V1.0 did not work)

teookie, 2021-11-23:
I can add two data points:

My machine running klipper v0.10.0-164 on a pi 3b with raspberry pi OS light and mainsail (OS and mainsail installed seperately - did not use mainsailOS image) fails to compile for Atmega 328p.

My other machine running klipper v0.10.0-164 on a pi 3b with mainsail _installed from mainsail OS image_ **does** compile successfully for Atmega 328p.

cbc02009, 2021-11-24:
Full output of verbose build command:
```  Creating symbolic link out/board
mkdir -p out/src out/src/avr out/src/generic
echo "#"avr"" > out/board-link.temp
if ! cmp -s out/board-link.temp out/board-link; then rm -f out/*.d out/src/*.d out/src/avr/*.d out/src/generic/*.d ; mv out/board-link.temp out/board-link ; fi
rm -f out/board
ln -sf /home/chris/klipper/src/"avr" out/board
mkdir -p out/board-generic
rm -f out/board-generic/board
ln -sf /home/chris/klipper/src/generic out/board-generic/board
  Building out/autoconf.h
mkdir -p out/
KCONFIG_AUTOHEADER=out/autoconf.h python2 lib/kconfiglib/genconfig.py src/Kconfig
  Compiling out/src/sched.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/sched.c -o out/src/sched.o
  Compiling out/src/command.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/command.c -o out/src/command.o
  Compiling out/src/basecmd.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/basecmd.c -o out/src/basecmd.o
  Compiling out/src/debugcmds.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/debugcmds.c -o out/src/debugcmds.o
  Compiling out/src/initial_pins.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/initial_pins.c -o out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/gpiocmds.c -o out/src/gpiocmds.o
  Compiling out/src/stepper.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/stepper.c -o out/src/stepper.o
  Compiling out/src/endstop.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/endstop.c -o out/src/endstop.o
  Compiling out/src/trsync.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/trsync.c -o out/src/trsync.o
  Compiling out/src/adccmds.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/adccmds.c -o out/src/adccmds.o
  Compiling out/src/spicmds.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/spicmds.c -o out/src/spicmds.o
  Compiling out/src/thermocouple.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/thermocouple.c -o out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/i2ccmds.c -o out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/pwmcmds.c -o out/src/pwmcmds.o
  Compiling out/src/spi_software.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/spi_software.c -o out/src/spi_software.o
  Compiling out/src/sensor_adxl345.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/sensor_adxl345.c -o out/src/sensor_adxl345.o
  Compiling out/src/lcd_st7920.o
avr-gcc -Iout/ -Isrc -Iout/board-generic/ -std=gnu11 -O2 -MD -Wall -Wold-style-definition -Wtype-limits -ffunction-sections -fdata-sections -flto -fwhole-program -fno-use-linker-plugin -ggdb3 -mmcu="atmega328p" -c src/lcd_st7920.c -o out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
[... cut ...]
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800a2e of out/klipper.elf section `.data' is not within region `data'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800b98 of out/klipper.elf section `.bss' is not within region `data'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800a2e of out/klipper.elf section `.data' is not within region `data'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800b98 of out/klipper.elf section `.bss' is not within region `data'
collect2: error: ld returned 1 exit status
make: *** [Makefile:72: out/klipper.elf] Error 1
```

VinnyCordeiro, 2021-12-22:
Looks like it isn't compiling for AVR mcus with smaller Flash space. Tried to compiled for the ATmega32U4 with no luck, while compiling for the ATmega646 and up still works.

Compiling log for the32u4:

`~/klipper$ make
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/spi_software.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/avr/main.o
  Compiling out/src/avr/timer.o
  Compiling out/src/avr/gpio.o
  Compiling out/src/avr/adc.o
  Compiling out/src/avr/spi.o
  Compiling out/src/avr/i2c.o
  Compiling out/src/avr/hard_pwm.o
  Compiling out/src/avr/watchdog.o
  Compiling out/src/avr/usbserial.o
  Compiling out/src/generic/usb_cdc.o
  Building out/compile_time_request.o
Version: v0.10.0-185-gdebcc22f
  Linking out/klipper.elf
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: out/klipper.elf section `.data' will not fit in region `text'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800bc7 of out/klipper.elf section `.bss' is not within region `data'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800bc7 of out/klipper.elf section `.bss' is not within region `data'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: region `text' overflowed by 1598 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:72: out/klipper.elf] Error 1`

closed 2022-02-03 by github-actions[bot]

ShawnBuckley, 2022-02-11:
This is a problem with the version of gcc 5.4.0 that Ubuntu ships with. Using a newer version of gcc works. Try building with fluidd Pi OS or mainsail OS if you have either installed or can use the guide here to build a newer version on Ubuntu: https://blog.zakkemble.net/avr-gcc-builds/

orobardet, 2022-04-10:
A suspect the issue to be in avr libc rather than gcc avr. But that's a whole.

As this issue only occurs on bullseye based distributions, and it's still working on buster based, another workaround is to revert back the AVR packages to buster.

It's not ideal. Upgrading to a more recent version of AVR toolchain is better, but it's also harder.

Here is the "buster rollback" workaround for those who don't know how to compile and setup a new avr toochain, or don't wants to. It allows to compile klipper firmware on small AVR directly from your rasperry pi, the normal way.
Tested on a Raspi OS bullseye (Fuildd image), but should work on any Debian distribution, on raspberry pi or not.

1. Add the buster package source: 
   - Edit the source file: `sudo vi /etc/apt/sources.list`
   - Add the following line just below the bullseye one: 
   `deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi`
2. Tells apt to use only buster for the AVR packages (and only those):
    - Create a new apt preferences file: `sudo vi /etc/apt/preferences.d/avr-buster`
    - With the following content:
    ```
   Package: avr-libc avrdude binutils-avr gcc-avr
   Pin: release n=buster
   Pin-Priority: 1001
    ```
3. Make apt discover the buster package: `sudo apt update`
4. Install the avr package (even if they are already installed, the command is to force apt to downgrade them: `sudo apt install avr-libc avrdude binutils-avr gcc-avr`
    apt should output a warning message for each packages saying they are downgraded
5. Try again to compile Klipper for AVR 328p, it should work now (do a `make clean` before the `make`).

This setup does not prevent the rest of you OS to have regular package updates on bullseye. `apt upgrade` can still be used, it will update all the system as usual but just keep the 4 AVR packages on the buster branch. It also means that if new versions of these package are publishes on the buster branch, they will used.

_This may also works on Ubuntu, but instead of rollbacking package from bullseye to buster, update the steps to rollback from jammy (which seems to have the faulty packages inherited from debian), to focal. Untested._

jotpowers, 2022-04-25:
Confirmed that using @orobardet 's workaround above, worked.  I had a printer I hadn't updated klipper or OS packages on and could compile on that.