Klipper3d/klipper #4940: Pause doesn't work

haplm, 2021-11-19:
I'm using the PAUSE RESUME CANCEL macros from Mainsail documentation (https://docs.mainsail.xyz/necessary-configuration).

**Lately, I have seen several occasions when pausing print didn't work**. Basically the print head moved briefly away from the print, but then immediately moved back and started to push filament again (it didn't even seem to go back to the right position, as some spaghetti occurred).

The message in the console was: `Print already paused`

Only thing that worked was to hit Emergency Stop button.

[Here is the log](https://github.com/Klipper3d/klipper/files/7569049/klippy.log.zip). It shows following:
* Several attempts to pause the print
* Emergency Stop
* Klipper restart

troypouliot, 2021-11-19:
I will add that I've noticed this as well after a recent update. It used to work properly, now to cancel a print you need to `PAUSE` first, then `CANCEL`

samcannell, 2021-11-22:
I noticed the same thing after updating Klipper recently -- an attempt to pause would move the extruder away and then immediately back again and continue printing.

My pause and resume macros originally came from [this Prusa MK3s configuration](https://raw.githubusercontent.com/PrusaOwners/klipper/master/config/printer-prusa-i3-mk3-2017.cfg) (though I've changed them for the new default parameter syntax):

```
[gcode_macro PAUSE]
rename_existing: BASE_PAUSE
gcode:
    {% set X = params.X|default(230)|int %}
    {% set Y = params.Y|default(230)|int %}
    {% set Z = params.Z|default(10)|int %}
    {% set E = params.E|default(1)|int %}
    SAVE_GCODE_STATE NAME=PAUSE_state
    BASE_PAUSE
    G91
    G1 E-{E} F2100
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F6000

[gcode_macro RESUME]
rename_existing: BASE_RESUME
gcode:
    {% set E = params.E|default(1)|int %}
    G91
    G1 E{E} F2100
    G90
    RESTORE_GCODE_STATE NAME=PAUSE_state MOVE=1
    BASE_RESUME
```

I've reverted to 59314d99e0e2b12720f144afff7c127b11103556 and pause and resume are now working fine again, so the issue was introduced somewhere since that commit.

haplm, 2021-12-28:
I have found out that the problem mostly occurs while doing pressure advance calibration (printing the tower and stepping PA values). I suppose that stepping the PA values somehow interferes with pause functionality.

khoff03, 2021-12-31:
I believe I'm having the same issue, only with Octoprint and with no PA.

At the beginning of a print, I test my filament runout sensor several times.  It appears to work as intended every time.  I'll then print for 1-2 hours and let a runout occur.  At that point, I experience the same symptoms experienced as OP
[klippy.log](https://github.com/Klipper3d/klipper/files/7796535/klippy.log)
.

matthewdolman, 2022-01-06:
I have also had the same issue.  Some prints can pause and resume fine, other times the behavior above occurs.  What is concerning also is that as it tries to travel to the park location it will travel to the z-endstop, trigger it, but then continue to move along the x-axis and there is a grinding sound before moving to the incorrect print locaion.  I am using Voron 2.4, mainsail.

haplm, 2022-01-06:
While I have previously stated that it occurs mainly when doing PA calibration, I have to retract that statement. Since then, Klipper failed to pause on bunch of regular prints. I can't see any pattern, the occurrence is purely random to me.

Arksine, 2022-01-06:
I would recommend adding `CLEAR_PAUSE` to your start gcode (or start macro).  At appears that you are triggering a PAUSE without subsequently resuming or otherwise clearing the paused state.  You must do one of the following after a pause before you can call PAUSE again:
1) RESUME
2) Cancel via `CANCEL_PRINT`
3) Call `CLEAR_PAUSE`

khoff03, 2022-01-06:
I agree that it appears that PAUSE is not getting cleared.  The pattern I've observed is that the first triggered runout works correctly, but sometimes fails on subsequent runouts.  

I'm always using the RESUME command when encountering these problems.  I can try making a Resume macro, and incorporate CLEAR_PAUSE followed by a RESUME command to see if this corrects it.

haplm, 2022-01-06:
@Arksine - unless something else calls PAUSE without me knowing before the print starts, it fails to pause on the first PAUSE call.

Yes, the log is then full of messages that PAUSE was already called, as I (or the runout sensor) try to pause again.

But I will definitely try to add `CLEAR_PAUSE` into the start gcode.

Arksine, 2022-01-06:
> I agree that it appears that PAUSE is not getting cleared. The pattern I've observed is that the first triggered runout works correctly, but sometimes fails on subsequent runouts.

Calling `RESUME` will always clear the pause.  In your particular log, I can see several instances where `PAUSE` and `RESUME` are called, sending the action commands for each. There is a runout event triggered by the sensor on line 10186, and there is no subsequent resume before what appears to be a print start (I can see where the extruder begins to heat on line 10347, however its difficult to tell when the print actually starts when its done through Octoprint).  To me this looks like a classic issue of a false trigger by the sensor.  It could be a faulty switch, wiring, or just a situation where the filament isn't depressing the switch far enough and vibration/movement is causing it to trigger.

FWIW, it looks like something similar is happening with @haplm's sensor.  On line 1631 a runout event is detected, resulting a pause with no subsequent resume.  This appears to occur before the SD Print starts during some sort of initialization procedure.  I'm not familiar with the motion sensor, so I can't really make an educated guess as to what is causing a false positive here.

khoff03, 2022-01-06:
@Arksine I'm definitely having problems as you describe, as I'm frequently getting phantom pauses when filament has not runout. I switched to shielded wire, which improved things, but has not totally solved the problem.

Regardless, shouldn't a false runout trigger (even if only momentary) still trigger a PAUSE that requires a user action to resume printing?

haplm, 2022-01-06:
The problem in my case is that even the first triggered run-out doesn't really stop the print (tools moves out but the print is immediately resumed - but not by me). And since the situation (runout ) still exists, Klipper tries to pause again, so now it is correctly reported that the print was already paused (but that didn't happen in reality).

But let's put the runout sensor aside. The same thing happens if I pause manually - it will immediately start printing again. Should I send you that log, @Arksine ?

haplm, 2022-02-12:
It would be suboptimal to close this issue. It manifest literally every single time I try to pause my printer now, which is really bad.

method1020, 2022-02-22:
I am also having this issue as of the last month or so. Going to try the CLEAR_PAUSE with start gcode.

plloppii, 2022-03-19:
I just ran into this this week.

I'm going to try to do a CLEAR_PAUSE in the beginning of my pause macro
`[gcode_macro PAUSE]
rename_existing: BASE_PAUSE
gcode:
    CLEAR_PAUSE
    SET_GCODE_VARIABLE MACRO=RESUME VARIABLE=e0_temp VALUE={printer.extruder.target}
    SET_GCODE_VARIABLE MACRO=RESUME VARIABLE=e1_temp VALUE={printer.extruder1.target}
    SAVE_GCODE_STATE NAME=PAUSE_STATE
    BASE_PAUSE
    PARK_NOZZLE`

spuder, 2022-03-23:
I was also running into this same problem with [the stock pause macro](https://github.com/spuder/klipper-ender2pro/blob/8d13ecf2b12dfee2ba59c67f6b10ede3f3e3bbe9/macros/pause.cfg)

Adding `CLEAR_PAUSE` fixed the issue for me.

haplm, 2022-04-27:
OK, so update. Adding CLEAR_PAUSE helps somewhat. Basically the pattern when pausing works changed, but it doesn't fix the issue completely. 

What it actually does is that it pauses in 100% of cases, but then not in 100% of cases one can resume the print. I need to find time to play with this more

khoff03, 2022-04-27:
I've been having better luck, but never needed CLEAR_PAUSE added to my macro.  I changed to a different switch, changed the logic from normally closed to normally open, and removed the "RESTORE_GCODE_STATE" and "SAVE_GCODE_STATE" from my macro.  I don't know which of these changes to attribute the improvement to.

I have two printers running the macro I posted above.  One works great 95 out of 100 times.  Occasionally it'll still exhibit the bug as described above.   The other printer has had 0 errors (it's using latest version of klipper).

Some type of PAUSE macro standardization would likely help the troubleshooting process.

haplm, 2022-04-28:
@khoff03 - I can't see your macro you are referring to (it seems like you haven't posted it in this thread). Would you mind to post it, please? I will give it a try to see if it helps. Thank you!

khoff03, 2022-04-29:
Here it is!

`[filament_switch_sensor RUNOUT_SENSOR]
switch_pin: ^rpi:gpio6
pause_on_runout: False
#   When set to True, a PAUSE will execute immediately after a runout
#   is detected. Note that if pause_on_runout is False and the
#   runout_gcode is omitted then runout detection is disabled. Default
#   is True.
runout_gcode:
  M600 #calls a macro with the name M600 (so it can be called by slicers too)
insert_gcode:
  M117 Filament inserted #just a throwaway GCode command to have something in this space. You can add your own.
#   A list of G-Code commands to execute after a filament insert is
#   detected. See docs/Command_Templates.md for G-Code format. The
#   default is not to run any G-Code commands, which disables insert
#   detection.
event_delay: 3.0
pause_delay: 0.0001

[pause_resume]

[gcode_macro M600]
gcode:
    {% set X = params.X|default(50)|float %}
    {% set Y = params.Y|default(0)|float %}
    {% set Z = params.Z|default(50)|float %}
    #SAVE_GCODE_STATE NAME=M600_state
    SET_IDLE_TIMEOUT TIMEOUT=30000
    PAUSE
    G91
    G1 E-5 F2700
    G1 Z{Z}
    #G1 E-100 F5000
    G90
    G1 X{X} Y{Y} F3000
    #G91

`

ld3300, 2022-05-02:
Are you expecting the pause to prevent the commands `g91` through `G1 X{X} Y{Y} F3000`?
Maybe this statement needs to be moved to a better spot in the docs, but:
```Important! Macros are first evaluated in entirety and only then are the resulting commands executed. If a macro issues a command that alters the state of the printer, the results of that state change will not be visible during the evaluation of the macro. This can also result in subtle behavior when a macro generates commands that call other macros, as the called macro is evaluated when it is invoked (which is after the entire evaluation of the calling macro).```
Which suggests that using a pause inside a gcode_macro won't prevent the rest of that macro from executing.
[Check here](https://www.klipper3d.org/Command_Templates.html#the-printer-variable)

bistory, 2022-06-09:
I'm facing the same issue and adding CLEAR_PAUSE doesn't do the trick...
I have no idea where this issue can come from...

The head is parked then the next command is executed like nothing happened...

My macros are the ones documented in Mainsail : 

```
[pause_resume]

[gcode_macro PAUSE]
description: Pause the actual running print
rename_existing: PAUSE_BASE
gcode:
  PAUSE_BASE
  _TOOLHEAD_PARK_PAUSE_CANCEL

[gcode_macro RESUME]
description: Resume the actual running print
rename_existing: RESUME_BASE
gcode:
  ##### read extrude from  _TOOLHEAD_PARK_PAUSE_CANCEL  macro #####
  {% set extrude = printer['gcode_macro _TOOLHEAD_PARK_PAUSE_CANCEL'].extrude %}
  #### get VELOCITY parameter if specified ####
  {% if 'VELOCITY' in params|upper %}
    {% set get_params = ('VELOCITY=' + params.VELOCITY)  %}
  {%else %}
    {% set get_params = "" %}
  {% endif %}
  ##### end of definitions #####
  {% if printer.extruder.can_extrude|lower == 'true' %}
    M83
    G1 E{extrude} F2100
    {% if printer.gcode_move.absolute_extrude |lower == 'true' %} M82 {% endif %}
  {% else %}
    {action_respond_info("Extruder not hot enough")}
  {% endif %}  
  RESUME_BASE {get_params}

[gcode_macro CANCEL_PRINT]
description: Cancel the actual running print
rename_existing: CANCEL_PRINT_BASE
variable_park: True
gcode:
  ## Move head and retract only if not already in the pause state and park set to true
  {% if printer.pause_resume.is_paused|lower == 'false' and park|lower == 'true'%}
    _TOOLHEAD_PARK_PAUSE_CANCEL
  {% endif %}
  TURN_OFF_HEATERS
  CANCEL_PRINT_BASE

[gcode_macro _TOOLHEAD_PARK_PAUSE_CANCEL]
description: Helper: park toolhead used in PAUSE and CANCEL_PRINT
variable_extrude: 1.0
gcode:
  ##### set park positon for x and y #####
  # default is your max posion from your printer.cfg
  {% set x_park = printer.toolhead.axis_maximum.x|float - 5.0 %}
  {% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}
  {% set z_park_delta = 2.0 %}
  ##### calculate save lift position #####
  {% set max_z = printer.toolhead.axis_maximum.z|float %}
  {% set act_z = printer.toolhead.position.z|float %}
  {% if act_z < (max_z - z_park_delta) %}
    {% set z_safe = z_park_delta %}
  {% else %}
    {% set z_safe = max_z - act_z %}
  {% endif %}
  ##### end of definitions #####
  {% if printer.extruder.can_extrude|lower == 'true' %}
    M83
    G1 E-{extrude} F2100
    {% if printer.gcode_move.absolute_extrude |lower == 'true' %} M82 {% endif %}
  {% else %}
    {action_respond_info("Extruder not hot enough")}
  {% endif %}
  {% if "xyz" in printer.toolhead.homed_axes %}
    G91
    G1 Z{z_safe} F900
    G90
    G1 X{x_park} Y{y_park} F6000
    {% if printer.gcode_move.absolute_coordinates|lower == 'false' %} G91 {% endif %}
  {% else %}
    {action_respond_info("Printer not homed")}
  {% endif %}
```

vmorrisPolar, 2022-06-16:
Having this issue as well.. All I've implemented is the M600 filament change with an hour long idle timeout. When the machine gets an M600, it moves into the corner and then immediately moves back to resume printing.

`[pause_resume]

[gcode_macro M600]
gcode:
    {% set X = params.X|default(50)|float %}
    {% set Y = params.Y|default(0)|float %}
    {% set Z = params.Z|default(10)|float %}
    SAVE_GCODE_STATE NAME=M600_state
    SET_IDLE_TIMEOUT TIMEOUT=3600
    PAUSE
    G91
    G1 E-.8 F2700
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F3000
    G91
    G1 E-50 F1000
    RESTORE_GCODE_STATE NAME=M600_state`

gztproject, 2022-06-22:
I'm also facing this issue. The last configuration change was adding the [exclude_object] so I disabled it and the pause seems to work again. Re-enabled and it doesn't stop...

closed 2022-08-03 by github-actions[bot]

Kozova1, 2022-08-16:
Hello, I have the same issue. It seems to only occur when PAUSE is called inside another macro.

dsloyer74, 2022-11-07:
I have an issue with the built-in pause with latest version of klipper/moonraker.  The Pause stops the print in place and does not move the head to a safe location.  Resume works, but if I do not resume immediately, hotend cooling/heating deforms the print.

Nickperillo, 2023-01-05:
I'm also experiencing very strange behaviour much like those above. Any insight into this from the Klipper team?

geirskogul, 2023-02-06:
I've also started experiencing this same issue after updating klipper recently

anubis71x, 2023-02-22:
I agree that the issue seems related to altering the pressure advance setting. I think merely changing the setting is enough to trigger the issue. I bumped the PA up one click and the next pause was effected.

testack18, 2023-04-21:
I don't want to hijack this thread as I'm also having a difficult time with PAUSE however I'm not playing with PA at all.
I found this interruptible heatsoak macro where some writes this about the PAUSE:

> The call to PAUSE will not stop the print start script from running, it just stops the next line of GCode in the file being printed from running

[Source](https://klipper.discourse.group/t/interruptible-heat-soak/1552/13).

I believe this is true not only for Start_Print but for any macro. PAUSE does not supposed to stop running of a macro, but only processing the next GCODE line read from the file being printed. That would explain what we all experiencing. Not sure thought if that is desired or not as someone mentioned this thread that this worked differently in a previous version...

ld3300, 2023-04-21:
@testack18 you are correct, pause is used in the same functional sense of other 3d firmware systems.  It does not stop you from doing anything, like manual moves, temperature adjustments, etc.  It, as you mentioned, only pauses the execution of commands from the gcode file being printed, after the current command has completed.  While paused you can manually run gcode commands, allow gcode_macros and delayed_gcodes to run, etc.

If you want to pause the process of gcode macros you need to structure them correctly.  A single gcode macro is evaluated all at once and is executed through to the end without being interrupted unless M112 is specified.  You can, however, make it so your gcode_macros enter a "pause" state.  One way this could be accomplished would be with a delayed_gcode that monitors a variable that could be set through some other process, a UI macro command button, or a physical button, which would then trigger a resume, or whatever the next series of gcode_macros you would like.  Additionally the UI macro command or physical button could themselves trigger the next desired operations directly.  The method would be determined by the use case.

So if you have a "start_gcode" that includes, for instance, pre-warming the bed, homing, bed mesh, and heating to print temperatures then pressing the PAUSE command won't stop any of those processes until they have all completed.  It will pause before the next command call that is actually in the gcode print file.

If using pause during klipper gcodes is important then it is much better to have several shorter and specific macros that get called in sequence then doing one large gcode_macro that does everything.

ld3300, 2023-04-21:
Additionally, if you are going to issue a pause and move the hot end don't include RESTORE_GCODE_STATE in the same macro where the pause exists.  Issue the pause and the moves as you would like, but do the restore state in the process that resumes the print.  While there may be an issue here with it choosing to move the head without MOVE=1 being specified, it would be best not to restore the gcode state until the resume anyway, in case any manually entered commands or macros get executed while paused.

VoronManiac, 2023-07-12:
Should the M600 macro in  klipper/config/sample-macros.cfg be re-written?  Seems like it is broken.

@ld3300 thanks for the explanation. Is there a section in main klipper docs that covers this as a reference?

ld3300, 2023-07-16:
> Should the M600 macro in klipper/config/sample-macros.cfg be re-written? Seems like it is broken.
> 
> @ld3300 thanks for the explanation. Is there a section in main klipper docs that covers this as a reference?

Yes, it seems like the M600 macro could be updated.  Might be best to submit a pull request that includes what the resume macro may look like.

If you look under [The "printer" Variable](https://www.klipper3d.org/Command_Templates.html?h=evaluated#the-printer-variable), a couple lines down starts with 'Important!' describes how the macros are evaluated.  There is a statement at the end of the 'Variables' statement that alludes to this as well.