Klipper3d/klipper #4945: Failed to verify BLTouch probe is raised; retrying.

JJPrinsloo, 2021-11-20:
I have read all information regarding this issue and did every thing everyone suggested, but still no luck. Reloaded firmware from scratch. Only failing on auto bed leveling. not failing for homing or z-tilt actions. also fails at the same point when auto bed leveling.
[klippy.log](https://github.com/Klipper3d/klipper/files/7574000/klippy.log)

Destroyerzx1, 2021-11-22:
Same issue Bltouch (Genuine), I have a FLSUN SR with mks robin nano v3 main board. I flashed klipper to it and for some reason the probe that comes with it (clips on via magnet) doesn't load the bed mesh and it's a pain to have to clip it on and off Every Time you want a mesh. So I bought a bltouch for it seeing as the main board "supports" it. Followed the directions on hooking it up but when I power it up the probe goes down up down up then klipper gets mad and shoots out messages like “Failed to verify BLTouch probe is raised; retrying”. I followed all the steps in bltouch setup for klipper. Made changes that others have posted and tried over and over again, But no luck once when I first started testing I did get open on query_probe but now it only shows triggered. My CR10s pro v2 is running klipper and equipped with a bltouch so I swapped them to make sure the probe I bought works and it does, GREAT. I installed the stock probe on the SR and it works but doesn’t listen to probe offset so it tries to jam the nozzle through the bed if not ESD during bed leveling. So what should I do now?  either my setup is junk,I have to go back to marlin or get a different main board. 
[klippy.log](https://github.com/Klipper3d/klipper/files/7582673/klippy.log)

Sineos, 2021-11-22:
Generally speaking:
* Verify wiring
* Use different sensor pins. Usually endstop pins will work
* Try with pull-up (^) or without for the sensor pin
* Maybe obvious: Make sure then pin does not hit the bed during the BLTouch startup sequence

Destroyerzx1, 2021-11-23:
> 

OMG I could f'n hug you I'm so happy the Pull-up (^!) was my issue, I just copied the sensor pin setting from previous setup and applied it to bltouch it was set at sensor_pin: ^!PC8 the removal of ! and it works, thank you

Sineos, 2021-12-08:
Issue seems concluded. Please open a new one if new problems arise. Preferable use https://klipper.discourse.group for general support topics.

closed 2021-12-08 by Sineos