Klipper3d/klipper #4947: More heat bed sections

ice2009, 2021-11-20:
Hello :)

It is something way to more connect heat bed? Souch as board with four bed + heat sensorand connect with multi mcu?

TheRevenantStar, 2021-11-26:
Per the documentation, not at this time. Is there a specific use case so the devs may see this?
I've thought about a feature like this for "checkerboard" heating very large beds... but haven't worked out if this should be handled by a separate circuit board or the MCU(s) with multiple heaters.

Gilabite, 2021-11-26:
You can already do this. I have a 510x510mm dual zone AC heater powered by SSRs. The 200mm inner zone is heater_bed and the outer ring is heater_generic, each has its own thermistor. For DC you can use multiple boards that have a bed connections or use a mosfet breakout board that can handle the current for the heaters.

ice2009, 2021-11-26:
Please can you share config file? :) 

>

Gilabite, 2021-11-27:
Well Im on vacation and wont be by my printers for awhile but may still be able to help. Setup one section as a regular bed heater like

[heater_bed]
heater_pin:
sensor_pin: 
sensor_type: 
control = pid
#pid_kp = 
#pid_ki = 
#pid_kd = 
min_temp: 
max_temp: 

And for every other heater section you have set it up as

[heater_generic pick_unique_name_for_each]
gcode_id: A #each one needs a unique gcode ID
heater_pin: 
sensor_type: 
sensor_pin: 
control: pid
#pid_kp = 
#pid_ki = 
#pid_kd = 
min_temp: 
max_temp: 

Now for all the other heaters besides the first one, gcodes like m190 wont work directly so you will need to figure out how you want your workflow to be. When I first set mine up I was using Cura as my slicer and I wrote a post processing script that would calculate the area of the mesh and pass that value to klipper. In klipper I wrote a macro that would compare the passed value against a stored value that represented the area of my inner ring. It would then turn on the outer ring with SET_HEATER_TEMPERATURE HEATER=heater_name TARGET={printer.heater_bed.target} if the area of the mesh was bigger than the inner ring. But I have since moved to superslicer and I havent had time to dig into it to much and have been turning on the outer ring manually directly in mainsail when I need it. Now matter what you do workflow wise in your end_gcode make sure you turn off EVERY bed heater you setup with SET_HEATER_TEMPERATURE HEATER=heater_name TARGET=0. Hope that helps.

ice2009, 2021-11-27:
> Well Im on vacation and wont be by my printers for awhile but may still be able to help. Setup one section as a regular bed heater like
> 
> [heater_bed] heater_pin: sensor_pin: sensor_type: control: pid pid_kp: pid_ki: pid_kd: min_temp: max_temp:
> 
> And for every other heater section you have set it up as
> 
> [heater_generic pick_unique_name_for_each] gcode_id: A #each one needs a unique gcode ID heater_pin: sensor_type: sensor_pin: control: pid pid_kp: pid_ki: pid_kd: min_temp: max_temp:
> 
> Now for all the other heaters besides the first one, gcodes like m190 wont work directly so you will need to figure out how you want your workflow to be. When I first set mine up I was using Cura as my slicer and I wrote a post processing script that would calculate the area of the mesh and pass that value to klipper. In klipper I wrote a macro that would compare the passed value against a stored value that represented the area of my inner ring. It would then turn on the outer ring with SET_HEATER_TEMPERATURE HEATER=heater_name TARGET={printer.heater_bed.target} if the area of the mesh was bigger than the inner ring. But I have since moved to superslicer and I havent had time to dig into it to much and have been turning on the outer ring manually directly in mainsail when I need it. No matter what you do workflow wise in your end_gcode make sure you turn off EVERY bed heater you setup with SET_HEATER_TEMPERATURE HEATER=heater_name TARGET=0. Hope that helps.

Can you send me mail ? mamra2893[at]gmail.com i have any question for you :)

closed 2022-01-09 by github-actions[bot]

DougAccatino, 2022-03-15:
@Gilabite id actually love to keep this going. I recently saw the prusa xl has a modular bed system, and thought it would be fun to design my own. I have no understanding of how the prusa implements this, but im trying to come up with my own. I have an Ender 5 Plus with a Mercury One Corexy Mod. Bed size being 377x370mm.

Most of my prints do not need to utilize that full size, nor use the full power. I primarily print ABS so that means 100C bed temp. Plus I’ve been wanting to upgrade my bed to cast tooling plate. This system, which I’m dubbing “B3D”, has the potential to reduce warping from heating unevenly, however that will be tested. 

I’ve drawn up a rough sketch of my intended design, ordered the mini plates, and figured out how to wire up 9 thermistors using an additional MCU flashed with klipper (Adafruit Grand Central was my first stab, looking to move to seeeduino Xiao), and went as far as designing a voltage divider PCB (one with 9, and another with 16 inputs) to help input the thermistors signal. 

I ordered 9 x 24V 30W silicone heaters to mount to each of my 120x120x6mm MIC6 plates, and each plate will have 3 point leveling with springs. Each bed gets its own MOSFET.

Beyond constructing the actual unit, I’ll have to figure out how to implement the intended use case, which I’ve come up with, as follows: 

A macro will look at the size of the print in the gcode, and toggle the specific beds to be turned on according to which plates it sits.
To further this idea, it would be ideal to have a slicer plugin (I’m currently using superslicer) that orients the object in the center of whichever plates it needs. 

I do know, that initially I’ll be manually turning these beds on and off until I can figure out how to even develop the necessary code.

shreeramlive, 2022-06-24:
#5597 @DougAccatino 
https://klipper.discourse.group/t/automatic-control-of-rectangular-heatbed-sections-based-on-print-area/3252