Klipper3d/klipper #4955: support for PA13 and PA14 on Anycubic Vyper

FotoFieber, 2021-11-22:
The mainboard on the Anycubic Vyper has the same issue as mentioned in https://github.com/Klipper3d/klipper/issues/3077?fbclid=IwAR0m5Ln2g7KPCejYGTvoeqMWChmSCv2Oo5nDEUAWZfZY1Ip0j77SGxByamY#issuecomment-939369046

Without the patch above, the Hotend LED and Motherboard fans can not be adressed from Klipper.

I may make a pull request to solve the issue, if I get directions, how I should implement id. Maybe I should add an STM32F103GD in the processor model?

zebra777, 2021-11-24:
Hello klipper team,

in the file "home/pi/klipper/src/stm32/stm32F1.c" at line 124 you wrote that its possible to discable swd to free pin PA13 + PA14. How can we discable this?

actually its only possible with a change in this file.

`if (gpio == GPIO('A', 13) || gpio == GPIO('A', 14)) {
        // Disable SWD to free PA13, PA14
		enable_pclock(AFIO_BASE);
        stm32f1_alternative_remap(AFIO_MAPR_SWJ_CFG_Msk,
                                  AFIO_MAPR_SWJ_CFG_DISABLE);
	}`

Perhaps there is an easier method to do this.

Greetings

KevinOConnor, 2021-11-24:
I'd probably try to fix it with the patch below, and instruct affected users to select "low-level options" and "Disable SWD at startup".  I don't have any of these impacted boards though, so I can't test this.

-Kevin

```diff
diff --git a/src/stm32/Kconfig b/src/stm32/Kconfig
index 7a1a6d188..da1fc2f5f 100644
--- a/src/stm32/Kconfig
+++ b/src/stm32/Kconfig
@@ -155,6 +155,16 @@ config STACK_SIZE
     int
     default 512
 
+config STM32F103GD_DISABLE_SWD
+    bool "Disable SWD at startup (for Gigadevices stm32f103 clones)"
+    depends on MACH_STM32F103 && LOW_LEVEL_OPTIONS
+    default n
+    help
+        The Gigadevices clone of the STM32F103 may not be able to
+        reliably disable SWD at run-time. This can prevent the PA13
+        and PA14 pins from being available. Selecting this option
+        disables SWD at startup and thus makes these pins available.
+
 
 ######################################################################
 # Bootloader
diff --git a/src/stm32/stm32f1.c b/src/stm32/stm32f1.c
index 88500165e..e76390938 100644
--- a/src/stm32/stm32f1.c
+++ b/src/stm32/stm32f1.c
@@ -257,8 +257,13 @@ armcm_main(void)
 
     // Disable JTAG to free PA15, PB3, PB4
     enable_pclock(AFIO_BASE);
-    stm32f1_alternative_remap(AFIO_MAPR_SWJ_CFG_Msk,
-                              AFIO_MAPR_SWJ_CFG_JTAGDISABLE);
+    if (CONFIG_STM32F103GD_DISABLE_SWD)
+        // Gigadevices clone can't enable PA13/PA14 at runtime - enable here
+        stm32f1_alternative_remap(AFIO_MAPR_SWJ_CFG_Msk,
+                                  AFIO_MAPR_SWJ_CFG_DISABLE);
+    else
+        stm32f1_alternative_remap(AFIO_MAPR_SWJ_CFG_Msk,
+                                  AFIO_MAPR_SWJ_CFG_JTAGDISABLE);
 
     sched_main();
 }
```

zebra777, 2021-11-24:
Are the pins PA13 + PA14 free with this fix.

I read "// Disable JTAG to free PA15, PB3, PB4 enable_pclock(AFIO_BASE);", but nothing about PA 13 + 14.

Zimbo111, 2021-11-24:
It would be great if that would be included.  I think there are a lot of users who would be very grateful for it.  I included.

FotoFieber, 2021-11-24:
@KevinOConnor Thx, I will try it tomorrow.

Zimbo111, 2021-11-24:
@FotoFieber Then keep us up to date and then report on how to integrate the patch.

FotoFieber, 2021-11-25:
@KevinOConnor verified your patch. It is working great. Thx!

KevinOConnor, 2021-11-25:
Okay, thanks.  I committed this change (commit 2b7d0bba).

-Kevin

Zimbo111, 2021-11-25:
Thank you @KevinOConnor

zebra777, 2021-11-26:
Thank you. Tested and Works.

closed 2021-12-03 by github-actions[bot]