Klipper3d/klipper #4958: Klipper probe not triggert

Max2883, 2021-11-23:
Hey, i´m new to klipper and trying to configure an inductive probe.
I use an Creality 3d 1.1.4 (8bit) Board.
My probe needs 10-36v .
I get 5.3V on the signal line by lowering it via resistance. (messured by Multimeter . Probe is working!)
It' s an "closer" probe, so I invertet the z-min pin in klipper.

Klipper is not recognizing my probe, always open or always closed by switching the logic back.
Strange enough a normal endstop works fine?!

My config: 
######################################################################
[stepper_z]
step_pin: PB3
dir_pin: !PB2
enable_pin: !PA5
microsteps: 16
rotation_distance: 8 # Use 4 for Ender5 versions after late 2019
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0
position_min: -2 # The Z carriage may need to travel below the Z=0
                 # homing point if the bed has a significant tilt.
position_max: 160

[probe]
pin: ^PC4
z_offset: 2.0
################################################################

I´m absolutly running out of ideas ....

Thank you for your help!
Max

simonkuehling, 2021-11-24:
Hi Max,

> My probe needs 10-36v .

Can you provide more detail on your specific probe and how it is wired up?

> I get 5.3V on the signal line by lowering it via resistance. (messured by Multimeter . Probe is working!)

How exactly did you measure and how did you trigger the probe?

Max2883, 2021-11-24:
Hi Simon,
I use a BEM08-02-NK as probe (had one lying around).

Probe                                                                                      
12v (brown)-------------------------------------------------- PSU
0v (blue)---------------------+-------------------------------PSU
                                             |
                                          15k ohm
                                             |
Sig. (black)---10k ohm------ +-------------------------------Sig. Pin Board 

I triggert the probe by holding a screwdriver against it ( Red led lights up).

Sig (Black)-------- +Multimeter - ------------------ 0V PSU 
=5.3V

simonkuehling, 2021-11-25:
Do the nano and the 12V PSU have a shared ground connection?
You could check by taking the same measurement again, but holding the negative voltmeter probe to a GND pin on the nano...

Max2883, 2021-11-25:
Did it .... 5.26v ...

simonkuehling, 2021-11-25:
oh, totally missed this:

```
pin: ^PC4
```
you have configured the internal pull-up resistor on that pin. But since the probe is triggering HIGH, the pin will never change state... It is either pulled high by the probe signal, or from the internal pull-up.
Well, technically you have an external pull-down with the voltage divider as well, but I guess the pull-up is countering it enough for the logic level to be logic high at all times...

Try it without:
```
pin: PC4
```

Max2883, 2021-11-25:
OK, i tried PC4 and !PC4 .
Both didn't worked.
I also tried ~PC4 and get a crashed mcu config ....

I gues the combination between probe and  board ist the main Problem ....

closed 2021-12-14 by Max2883