Klipper3d/klipper #4959: "!! Unable to open file" when printing a second file from the LCD menu

YouMakeTech, 2021-11-24:
Starting a print from the LCD menu of my Ender 3 Pro always works fine the first time. However, upon completion of the first print (no abort), when I try to start a second file I get always an error message "!! Unable to open file"
I am using the default menu.cfg of Klipper.

This is the error reported in the log file (attached):
`virtual_sdcard file open
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 178, in _load_file
    fname = files_by_lower[fname.lower()]
KeyError: 'top'
Unable to open file
Script running error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/display/menu.py", line 956, in dispatch_gcode
    self.gcode.run_script(script)
  File "/home/pi/klipper/klippy/gcode.py", line 216, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 170, in cmd_M23
    self._load_file(gcmd, filename)
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 186, in _load_file
    raise gcmd.error("Unable to open file")
CommandError: Unable to open file`

No matters how many times I try, it is impossible to initiate a print from the LCD display. However, starting a print of the same file in the Fluidd web interface works right away.

fluidd v1.16.2
klipper v0.10.0-139-g7ef7bf60
moonraker v0.7.1-154-gd51d65e

Any help would be appreciated. Thank you!
Vincent

[klippy.log.zip](https://github.com/Klipper3d/klipper/files/7594529/klippy.log.zip)

YouMakeTech, 2021-11-24:
Ok, I think I found something new... The LCD menu does not seem to like long filenames with spaces... For example, I could not start printing a file named "Top Cover.stl" from the LCD menu...  I renamed the file to "Top_Cover.stl" (replacing the space with an underscore) and then I could start printing the file from the LCD menu.

techyg, 2021-12-17:
I ran into the same issue. I had a long file name with both spaces and underscores in it, removing both allowed me to print. It would be good to know if there are any more limitations found. I believe the intended behavior should be any valid file name on the disk should print.

Sineos, 2021-12-18:
Duplicate of https://github.com/Klipper3d/klipper/issues/4794

closed 2021-12-18 by Sineos