Klipper3d/klipper #4960: mcu 'mcu': Unable to connect

gnutison, 2021-11-24:
Hi,
Today I tried install Klipper on Raspberry Pi Zero 2W. After uploading printer.cfg and restart it say:

mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
Ender 3v2, 4.2.2
it say unable to open printer.cfg, dont know why. Thank for helps and tips
[klippy.log](https://github.com/Klipper3d/klipper/files/7595458/klippy.log)
[moonraker.log](https://github.com/Klipper3d/klipper/files/7595460/moonraker.log)
r

gnutison, 2021-11-24:
Fixed, was my bad. double checked menuconfig, there is one missing setting by me.

closed 2021-11-24 by gnutison

Talha909, 2021-11-25:
> Fixed, was my bad. double checked menuconfig, there is one missing setting by me.

what is the missing setting? I am also getting my pi 2 w

gnutison, 2021-11-25:
do you have same issues like me? send klippy.log and screenshot of menuconfig, may I able to help you.