Klipper3d/klipper #4963: Issue with the Documentation Search Function on https://www.klipper3d.org/

Sineos, 2021-11-25:
Browser:
* Firefox 94.0.1
* Chrome  95.0.4638.69

https://www.klipper3d.org/Rotation_Distance.html

[image]

When going to the same Chapter via the search function and highlighting:

https://www.klipper3d.org/Rotation_Distance.html?h=rotat

[image]

KevinOConnor, 2021-12-31:
I'd guess this is a defect in mkdocs.  I'm not sure how we'd be able to fix this in the Klipper build itself.

-Kevin

Sineos, 2021-12-31:
https://github.com/squidfunk/mkdocs-material/issues/2068

Seems there is a fix available. Maybe @dingyifei could cross check

KevinOConnor, 2021-12-31:
Okay.  It's possible to update the mkdocs-material version (it is specified in docs/_klipper3d/mkdocs-requirements.txt ).  However, it is unclear to me if this issue was addressed in the latest publicly released version.

-Kevin

KevinOConnor, 2021-12-31:
It does seem that an upgrade to mkdocs-material v8.1.3 fixes this issue (commit 608cd38d).

Thanks.
-Kevin

Sineos, 2021-12-31:
Fix confirmed 👍

closed 2021-12-31 by Sineos