Klipper3d/klipper #4965: [Request] support of attached host keyboard

LMS0815, 2021-11-25:
Please consider adding support for an attached host (raspberry) USB keyboard.

The idea is to assign gcode macros to a keystroke. 

**Proof of concept:**
I am currently using a Python3 script and a  wireless USB number pad to move the hotend.
The script is just receiving the keycodes and calling a gcode_macro with parameters using the klipper API.

This is my  Klipper printer.cfg
```
# this is how a KBD module config might look like.
#[kbd]
#device: /dev/input/event3
#   The device to use.
#macro: _KBD
#   The gcode_macro to use. The default is _KBD.
#key: KEY
#   The gcode_macro key to use. The default is KEY.

# This macro is meant to be used via fluid UI 
[gcode_macro keyboard]
description: Keyboard move parameter
gcode:
 {% set speed = params.SPEED|default(6000)|float %}
 {% set hight = params.HIGHT|default(10)|float %}
 {% if params.SPEED|default(0)|float > 0 %}
  SET_GCODE_VARIABLE MACRO=_KBD VARIABLE=kbmove_speed VALUE={speed}
 {% endif %}
 {% if params.HIGHT|default(0)|float > 0 %}
  SET_GCODE_VARIABLE MACRO=_KBD VARIABLE=kbmove_hight VALUE={hight}
 {% endif %}
  _KBD

# This macro is meant to do all the keystroke magic hidden from fluid UI 
[gcode_macro _KBD]
variable_kbmove_hight: 10.0
variable_kbmove_astep: [ 0.1, 1.0,10.0, 25.0, 50.0,100.0]
variable_kbmove_speed: 6000.0
description: USB Keyboard Moves
gcode:
 {% set key = params.KEY|default('') %}
 {% set hight = printer["gcode_macro _KBD"].kbmove_hight %}
 {% set speed = printer["gcode_macro _KBD"].kbmove_speed %}
 SAVE_GCODE_STATE NAME=my_move_state
 G91
 {% if key == '5' %}
  G90
  G0 X160 Y160 F{speed}
 {% elif key == '8' %}
  G0 Y{hight} F{speed}
 {% elif key == '2' %}
  G0 Y-{hight} F{speed}
 {% elif key == '4' %}
  G0 X-{hight} F{speed}
 {% elif key == '6' %}
  G0 X{hight} F{speed}
 {% elif key == '7' %}
  G0 X-{hight} Y{hight} F{speed}
 {% elif key == '9' %}
  G0 X{hight} Y{hight} F{speed}
 {% elif key == '1' %}
  G0 X-{hight} Y-{hight} F{speed}
 {% elif key == '3' %}
  G0 X{hight} Y-{hight} F{speed}
 {% elif key == 'NUMLOCK' %}
  G0 Z-{hight} F{speed}
 {% elif key == '/' %}
  G90
  G0 Z5 F{speed}
 {% elif key == 'STAR' %}
  G0 Z{hight} F{speed}
 {% elif key == '+' %}
  {% set astep = printer["gcode_macro _KBD"].kbmove_astep %}
  {% set index = 0 if hight not in astep else astep.index(hight)  %}
  {% set hight = astep[(index+1 if index < astep|length -1  else -1 )] %}
  SET_GCODE_VARIABLE MACRO=_KBD VARIABLE=kbmove_hight VALUE={hight}
  {action_respond_info("New Hight: %s" % (hight))}
 {% elif key == '-' %}
  {% set astep = printer["gcode_macro _KBD"].kbmove_astep %}
  {% set index = 0 if hight not in astep else astep.index(hight)  %}
  {% set hight = astep[(index-1 if index > 0 else 0)] %}
  SET_GCODE_VARIABLE MACRO=_KBD VARIABLE=kbmove_hight VALUE={hight}
  {action_respond_info("New Hight: %s" % (hight))}
 {% elif key == 'BACKSPACE' %}
  G90
  G0 X0 Y0 F{speed}
 {% elif key == 'ENTER' %}
  G28 F{speed}
  G90
  G0 X160 Y160 F{speed}
 {% else %}
  {action_respond_info("Key: '%s' hight: %.2f speed: %.2f" % (key,hight,speed))}
 {% endif %}
 RESTORE_GCODE_STATE NAME=my_move_state

```

And this is the associated Python 3 script (with hard coded device name `/dev/input/event3` which is using the klipper API:
```
#!/usr/bin/python
# python3 keyboard.py

import sys, os, optparse, socket, fcntl, select, json, errno, time, requests
from evdev import InputDevice, list_devices, ecodes, categorize

CODE_MAP_CHAR = {
    'KEY_ENTER': "ENTER",
    'KEY_PLUS': "+",
    'KEY_MINUS': "-",
    'KEY_SPACE': " ",    
    'KEY_BACKSLASH': "\\",
    'KEY_GRAVE': "`",
    'KEY_NUMERIC_0': "0",
    'KEY_NUMERIC_1': "1",
    'KEY_NUMERIC_2': "2",
    'KEY_NUMERIC_3': "3",
    'KEY_NUMERIC_4': "4",
    'KEY_NUMERIC_5': "5",
    'KEY_NUMERIC_6': "6",
    'KEY_NUMERIC_7': "7",
    'KEY_NUMERIC_8': "8",
    'KEY_NUMERIC_9': "9",
    'KEY_NUMERIC_STAR': "STAR",
    'KEY_NUMLOCK': "NUMLOCK",
    'KEY_BACKSPACE': "BACKSPACE",
    'KEY_A': "A",
    'KEY_B': "B",
    'KEY_C': "C",
    'KEY_D': "D",
    'KEY_E': "E",
    'KEY_F': "F",
    'KEY_G': "G",
    'KEY_H': "H",
    'KEY_I': "I",
    'KEY_J': "J",
    'KEY_K': "K",
    'KEY_L': "L",
    'KEY_M': "M",
    'KEY_N': "N",
    'KEY_O': "O",
    'KEY_P': "P",
    'KEY_Q': "Q",
    'KEY_R': "R",
    'KEY_S': "S",
    'KEY_T': "T",
    'KEY_U': "U",
    'KEY_V': "V",
    'KEY_W': "W",
    'KEY_X': "X",
    'KEY_Y': "Y",
    'KEY_Z': "Z",
    'KEY_KPPLUS': "+",
    'KEY_KPMINUS': "-",
    'KEY_KPASTERISK': "STAR",
    'KEY_KPDOT(': ".",
    'KEY_KPSLASH': "/",
    'KEY_KPENTER': "ENTER",
    'KEY_KP0': "0",
    'KEY_KP1': "1",
    'KEY_KP2': "2",
    'KEY_KP3': "3",
    'KEY_KP4': "4",
    'KEY_KP5': "5",
    'KEY_KP6': "6",
    'KEY_KP7': "7",
    'KEY_KP8': "8",
    'KEY_KP9': "9",
    'KEY_0': "0",
    'KEY_1': "1",
    'KEY_2': "2",
    'KEY_3': "3",
    'KEY_4': "4",
    'KEY_5': "5",
    'KEY_6': "6",
    'KEY_7': "7",
    'KEY_8': "8",
    'KEY_9': "9",
    'KEY_LEFTBRACE': "[",
    'KEY_RIGHTBRACE': "]",    
    'KEY_COMMA': ",",
    'KEY_EQUAL': "=",    
    'KEY_SEMICOLON': ";",
    'KEY_APOSTROPHE': "'",
    'KEY_TAB': "\t",
    'KEY_DOT': ".",
    'KEY_SLASH': "/",
}

def webhook_socket_create(uds_filename):
    sock = socket.socket(socket.AF_UNIX, socket.SOCK_STREAM)
    sock.setblocking(0)
    sys.stderr.write("Waiting for connect to %s\n" % (uds_filename,))
    while 1:
        try:
            sock.connect(uds_filename)
        except socket.error as e:
            if e.errno == errno.ECONNREFUSED:
                time.sleep(0.1)
                continue
            sys.stderr.write("Unable to connect socket %s [%d,%s]\n"
                             % (uds_filename, e.errno,
                                errno.errorcode[e.errno]))
            sys.exit(-1)
        break
    sys.stderr.write("Connection.\n")
    return sock

class SocketLoop:
    def __init__(self, uds_filename, kbd_filename ):
        self.kbd_fd = InputDevice(kbd_filename)
        #self.kbd_fd.grab()
        self.webhook_socket = webhook_socket_create(uds_filename)
        self.poll = select.poll()
        self.poll.register(self.webhook_socket, select.POLLIN | select.POLLHUP)
        self.socket_data = ""
    def process_kbd(self):
        for event in self.kbd_fd.read_loop():
            if event.type == ecodes.EV_KEY:
                e = categorize(event)
                if e.keystate == e.key_up:
                    #sys.stdout.write("{1}({0})\n".format(e.keycode, parse_key_to_char(e.keycode)))
                    try:
                        cm = json.dumps({'id': 123, 'method': 'gcode/script', 'params': {'script': "_KBD KEY={}".format(parse_key_to_char(e.keycode))}}, separators=(',', ':'))
                    except:
                        sys.stderr.write("ERROR: Unable to parse line\n")
                        continue
                    #sys.stdout.write("SEND: %s\n" % (cm,))
                    try:
                        self.webhook_socket.send("{}\x03".format(cm).encode('utf-8'),)
                    except:
                        break
    def run(self):
        self.process_kbd()

def parse_key_to_char(val):
    return CODE_MAP_CHAR[val] if val in CODE_MAP_CHAR else val[4:]

def main():
    while 1:
        ml = SocketLoop('/tmp/klippy_uds', '/dev/input/event3')
        ml.run()

if __name__ == "__main__":
    main()

```

LMS0815, 2022-01-11:
Awaiting answer...

Sineos, 2022-01-12:
The Klipper lead developer has limited time and apparently this request has unfortunately not made it to the short-list. Check the PR section and you will notice that there already is a backlog of developed features awaiting review and final approval.

Either you wait until someone with enough time and knowledge is willing to implement this feature or you hand in a Pull Request with your implementation

LMS0815, 2022-01-13:
THX. I am happy with the solution i described in this request.

closed 2022-01-13 by LMS0815