Klipper3d/klipper #4966: BL-Touch - QUERY_PROBE not working as documented

pietchaki, 2021-11-25:
Documentation for [BL-Touch initial tests](https://www.klipper3d.org/BLTouch.html#initial-tests) are not correct. 
`Run BLTOUCH_DEBUG COMMAND=pin_down, verify that the pin moves down, run BLTOUCH_DEBUG COMMAND=touch_mode, run QUERY_PROBE, and verify that command reports "probe: open". Then while gently pushing the pin up slightly with the nail of your finger run QUERY_PROBE again. Verify the command reports "probe: TRIGGERED".`

Problem is, QUERY_PROBE will only report a triggered state, if issued within the 10ms while the probe is sending that state.
[Seems that it is a known issue](https://github.com/Klipper3d/klipper/issues/1042#issuecomment-450133458), so why does that information stays in the documentation?
To be clear, probing works without problem. Just the "QUERY_PROBE" command does not work as documented.

My problem with it is that I was installing Klipper on my custom made printer, I was running Marlin+Octoprint before. Got a Fluidd image and everything went without any problems, until I reached the BL-Touch initial tests... I followed those instructions, and my probe was not reporting as triggered, so I thought it had a configuration problem. I followed a lot of debugging tutorials and nothing I did got me to read a triggered state. I opened my electronics box, changed a bunch of wires, rewired my toolhead and did a lot of testing, only to finally come into some people saying that the problem was just the official documentation was invalid. 

You can fix the documentation page, to give a proper step-by-step testing walkthrough. with the correct information there.
OR, the better way, fix the "QUERY_PROBE" command so it retain the "TRIGGERED" state until the probe pin goes down. Maybe, if you can't retain the triggered state, then add a command that will wait for the probe to be triggered, and then send that message on console. Don't forget to fix the documentation for the Bl-Touch tests too.

jakep82, 2021-11-26:
Looks correct to me.

pietchaki, 2021-11-26:
So you say that I somehow should be aware that I have to send the "QUERY_PROBE" command less then 10 milliseconds after triggering the probe?

jakep82, 2021-11-26:
The probe should remain triggered until you clear the state. I no longer have a bltouch to test, but that's the way it worked when I had one. If it doesn't then it's a problem with the code and not the documentation.

jakep82, 2021-11-26:
I should also note that many changes to the bltouch code have been made since the 3 year old post you linked.

Sineos, 2021-11-26:
```
18:26:37 Sending BLTOUCH_DEBUG COMMAND=pin_down
18:27:02 Sending BLTOUCH_DEBUG COMMAND=touch_mode
18:27:22 probe: open
18:27:27 probe: open
18:27:33 probe: open
18:27:46 probe: TRIGGERED
18:27:49 probe: TRIGGERED
18:27:51 probe: TRIGGERED
```

Works as expected

pietchaki, 2021-11-26:
> I should also note that many changes to the bltouch code have been made since the 3 year old post you linked.

Yeah, I know.
But, for what I tested here, my current, up-to-date Klipper install, is behaving exactly as reported in that 3yo post.
In that Issue, they solved by setting a "test_sensor_pin" in the BL-Touch configuration. That option was later renamed to "pin_up_reports_not_triggered". Problem is, it does not solve the "QUERY_PROBE" problem in my printer. In fact, setting it to FALSE makes G28 fail by dropping the probe pin before lowering the bed.

> Works as expected

Funny, I have a BL-Touch here and that simply does not work.
It's behaving as I told, QUERY_PROBE  only reports "open", never reports as "TRIGGERED", anyway G28 works as intended, so that is definitely not an issue in the probe itself.

[image]
[image]

Gilabite, 2021-11-26:
mine work too without issue

jakep82, 2021-11-26:
Do you have a bltouch, or a clone?  The clones often don't work the same. Also, are you pushing the pin up after putting it in touch mode?  It sounds like it's working as intended for others.

pietchaki, 2021-11-27:
> Do you have a bltouch, or a clone? The clones often don't work the same. 

Mine is a clone, but I had read people with original ones also having this issue, as G28 works, they didn't report.
Also, as both official and clones work seamlessly on Marlin, no need to adjust anything different for them, I expected that their output signals was also exactly the same, clones just having worse reliability. 

> Also, are you pushing the pin up after putting it in touch mode? It sounds like it's working as intended for others.

For that console print I sent, the first "testprobe" is without any touch, second was slightly touching it just until red led lights up, third was pushing a little more, and the fourth was after pushing the pin all the way in so it stays up, this last one is not shown in the print but also reported as open.

KevinOConnor, 2022-01-04:
See #5088 .

-Kevin

closed 2022-01-04 by pietchaki