Klipper3d/klipper #4970: Option 'z_offset' in section 'bltouch' must be specified

Talha909, 2021-11-28:
I did the z calibration and after that I click on accept and then save_config. After pressing the save config I got this error message 

Option 'z_offset' in section 'bltouch' must be specified

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
[image]
[klippy.log](https://github.com/Klipper3d/klipper/files/7613668/klippy.log)
[printer (2).txt](https://github.com/Klipper3d/klipper/files/7613670/printer.2.txt)

gnutison, 2021-11-28:
add to [bltouch] z_offset: 0

Talha909, 2021-11-29:
I have set the Z_offset:0 but when i calibrate the proble. I do testz and then accept and then save_config.  At that time klipper commented z offset and through the error message. I then open the printer.cfg file and uncommented it and save it. The new value is saved in the file. But the offset did not work.  There is always gap between the nozzle and bed.

gnutison, 2021-11-29:
I had exactly issues like you before.
Adding z_offset: 0 in bltouch was "little backdoor" to avoid an error message so I can move to next steps PROBE_CALIBRATE. After properly Probe Calibrate  z offset should be commented in bltouch section and the real one is down below when u scrolled down. Under this text
#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.

snafu1282, 2021-12-19:
I'm have the same issue, except Klipper will not accept the fact that there is a SAVE_CONFIG section. In fact, I wind up with 2 SAVE_CONFIG sections, and I still get the "Option 'z_offset' in section 'bltouch' must be specified" error.

# This file contains common pin mappings for the BIGTREETECH SKR V1.4
# board. To use this config, the firmware should be compiled for the
# LPC1768 or LPC1769(Turbo).

# See docs/Config_Reference.md for a description of parameters.

[bltouch]
sensor_pin: P0.10  # Pull-up (^ symbol) needed in open drain mode
control_pin: P2.0
# Some BLTouch V3 and many clones apparently require this, though mine didnt:
pin_up_touch_mode_reports_triggered: true
#flavor: genuine_smart_3.1
x_offset: -43
y_offset: -10
# z_offset = 1.199
pin_move_time: 1
samples: 2

[stepper_x]
step_pin: P2.2
dir_pin: !P2.6
enable_pin: !P2.1
microsteps: 16
rotation_distance: 80
#endstop_pin: !P1.29
endstop_pin: P1.29
position_endstop: 228
position_max: 228
homing_speed: 40
position_min: 0

[stepper_y]
step_pin: P0.19
dir_pin: !P0.20
enable_pin: !P2.8
microsteps: 16
rotation_distance: 80
#endstop_pin: !P1.28
endstop_pin: P1.28
position_endstop: 220
position_max: 220
homing_speed: 40
position_min: 0

[stepper_z]
step_pin: P0.22
dir_pin: !P2.11
#dir_pin: P2.11
enable_pin: !P0.21
microsteps: 16
rotation_distance: 8
#endstop_pin: !P1.27
#endstop_pin: P1.28
#position_endstop: 0.0
position_max: 300
endstop_pin: probe:z_virtual_endstop
position_min: -3

[extruder]
step_pin: P2.13
#dir_pin: !P0.11
dir_pin: P0.11
enable_pin: !P2.12
microsteps: 16
max_extrude_only_distance: 100.0
#rotation_distance: 22.8346860205898
#rotation_distance: 44
#rotation_distance: 22.8346860205898
[... 216 lines cut ...]

#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	0.010000, -0.006250, 0.048750, 0.063750, 0.041250
#*# 	-0.068750, -0.045000, -0.003750, -0.015000, 0.006250
#*# 	-0.070000, -0.073750, -0.035000, -0.015000, 0.015000
#*# 	-0.072500, -0.063750, -0.026250, -0.026250, 0.046250
#*# 	-0.032500, -0.010000, -0.048750, -0.045000, 0.037500
#*# tension = 0.2
#*# min_x = 20.0
#*# algo = lagrange
#*# y_count = 5
#*# mesh_y_pps = 2
#*# min_y = 18.0
#*# x_count = 5
#*# max_y = 206.0
#*# mesh_x_pps = 2
#*# max_x = 180.0

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bltouch]
#*# z_offset = 1.399

snafu1282, 2021-12-19:
I'm have the same issue, except Klipper will not accept the fact that there is a SAVE_CONFIG section. In fact, I wind up with 2 SAVE_CONFIG sections, and I still get the "Option 'z_offset' in section 'bltouch' must be specified" error.

Here's the relevant info. I have also attached a zip of printer.cfg.

```
[bltouch]
sensor_pin: P0.10  # Pull-up (^ symbol) needed in open drain mode
control_pin: P2.0
# Some BLTouch V3 and many clones apparently require this, though mine didnt:
pin_up_touch_mode_reports_triggered: true
#flavor: genuine_smart_3.1
x_offset: -43
y_offset: -10
# z_offset = 1.199
pin_move_time: 1
samples: 2

# Set z offset = -1.001 after testz=-4

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bltouch]

#*# z_offset = 1.199

#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	0.010000, -0.006250, 0.048750, 0.063750, 0.041250
#*# 	-0.068750, -0.045000, -0.003750, -0.015000, 0.006250
#*# 	-0.070000, -0.073750, -0.035000, -0.015000, 0.015000
#*# 	-0.072500, -0.063750, -0.026250, -0.026250, 0.046250
#*# 	-0.032500, -0.010000, -0.048750, -0.045000, 0.037500
#*# tension = 0.2
#*# min_x = 20.0
#*# algo = lagrange
#*# y_count = 5
#*# mesh_y_pps = 2
#*# min_y = 18.0
#*# x_count = 5
#*# max_y = 206.0
#*# mesh_x_pps = 2
#*# max_x = 180.0

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bltouch]
#*# z_offset = 1.399
```
[printer2021dec19 #2.zip](https://github.com/Klipper3d/klipper/files/7741676/printer2021dec19.2.zip)

Sineos, 2021-12-20:
> So this begs the question: why 2 SAVE_CONFIGs, and which 1 takes precedence?

Your excerpt looks like you have manually tinkered with the `SAVE_CONFIG` section as it does not produce blank lines. I would delete the whole block and start over.

snafu1282, 2021-12-20:
> > So this begs the question: why 2 SAVE_CONFIGs, and which 1 takes precedence?
> 
> Your excerpt looks like you have manually tinkered with the `SAVE_CONFIG` section as it does not produce blank lines. I would delete the whole block and start over.

Oh how I wish that was true. It would be so easy to fix in that case. Unfortunately, it's not the case. The 2 sections shown were both completely auto-generated. That was after doing what you suggested: i.e. completely deleting the SAVE_CONFIG section.

I wonder what would happen if I deleted the blank lines?

snafu1282, 2021-12-20:
> I wonder what would happen if I deleted the blank lines?

So I have my answer. I removed the blank lines from the top SAVE_CONFIG and deleted the bottom 2 (yes, it had gone up to 3 SAVE_CONFIGs after doing another probe calibration). I also ensured that there was no space between the # and z_offset in [bltouch]. After a restart, there was no error. At the moment, it's working. Fantastic.

Thank you for the info about blank lines: not something I'm likely to forget any time soon 👍

Sineos, 2021-12-30:
If this is still an issue please open a new post on https://klipper.discourse.group and we will be happy to assist

closed 2021-12-30 by Sineos