Klipper3d/klipper #4971: After updating Klipper (trough fluidd) and flashing the MCU, i get an mcu 'host': Command format mismatch: error

olif0x, 2021-11-28:
Hey there,
yesterday at around 3 am GMT+1 i've decided to update all components trough the Fluidd GUI.
1. Moonraker
2. Klipper (Errors started to happen)
3. afterwards OS Updates (to ensure that everything is up2date)

First errors where because in my pause, cancel makros it didn't accept the default_parameter_x commands

so i completely commented those macros out.

Afterwards i then got the error seen in the screenshot.
[image]

I did flash my skr-mini-e3-v2 with the following command:
```

pi@fluiddpi:~ $ sudo service klipper stop
[sudo] Passwort für pi:
pi@fluiddpi:~ $ cd ~/klipper
pi@fluiddpi:~/klipper $ git pull
Bereits aktuell. (already on latest version)
pi@fluiddpi:~/klipper $ make clean
pi@fluiddpi:~/klipper $ make menuconfig
  Creating symbolic link out/board
Loaded configuration '/home/pi/klipper/.config'
No changes to save (for '/home/pi/klipper/.config')
pi@fluiddpi:~/klipper $ make
```

And afterwards i did flash the mcu:
```
`pi@fluiddpi:~/klipper $ ./scripts/flash-sdcard.sh -f ~/klipper/out/klipper.bin /dev/ttyACM0 btt-skr-mini-e3-v2
Flashing /home/pi/klipper/out/klipper.bin to /dev/ttyACM0
Checking FatFS CFFI Build...
Connecting to MCU...Connected
Checking Current MCU Configuration...Done
MCU needs restart: is_config=1, is_shutdown=0
Attempting MCU Reset...Done
Waiting for device to reconnect...Done
Connecting to MCU...Connected
Initializing SD Card and Mounting file system...

SD Card Information:
Version: 2.0
SDHC/SDXC: False
Write Protected: False
Sectors: 245760
manufacturer_id: 0
oem_id: 0
product_name: APPSD
product_revision: 0.0
serial_number: 000002EE
manufacturing_date: 9/2020
capacity: 120.0 MiB
fs_type: FAT16
volume_label:
volume_serial: 2826674516
Uploading Klipper Firmware to SD Card...Done
Validating Upload...Done
Firmware Upload Complete: firmware.bin, Size: 23280, Checksum (SHA1): 2D48ACAB6D73784074331DFBD644B3D9E5A9E8BA
Attempting MCU Reset...Done
Waiting for device to reconnect....Done
Connecting to MCU...Connected
Verifying Flash...Version matched...Done
Firmware Flash Successful
Current Firmware: v0.10.0-167-g2b7d0bba
Attempting MCU Reset...Done
SD Card Flash Complete
pi@fluiddpi:~/klipper $ sudo service klipper start
```
`

And i've still got the error on the klippy GUI:
[image]

After the mcu flash i also did disconnect my 3d-printer from the power to let it completly powercycle.

 (CR10s Pro v1 but with an skr-mini-e3 as the mainboard and a microSwiss Extruder)

I've attached all logfiles and my config-files maybe you could help me make my printer run again? :-) I wanted to start a 2-day print :D 
Should had wait with that update :P

many thanks and best regards
Olif
[github_issue_klippy_log&config-files.zip](https://github.com/Klipper3d/klipper/files/7613739/github_issue_klippy_log.config-files.zip)
0x

Sineos, 2021-11-28:
Update Klipper **and** MCU again. Also see https://www.klipper3d.org/FAQ.html?h=update#how-do-i-upgrade-to-the-latest-software

For the macros, replace

```
[gcode_macro CANCEL_PRINT]
rename_existing = BASE_CANCEL_PRINT
default_parameter_x = 0
default_parameter_y = 0
default_parameter_z = 10
```

by

```
[gcode_macro CANCEL_PRINT]
rename_existing = BASE_CANCEL_PRINT
gcode:
    {% set x = params.x|default(0)|float %}
    {% set y = params.y|default(0)|float %}
    {% set z = params.z|default(10)|float %}
```

olif0x, 2021-11-28:
Thx for your reply!

Ok, i've updated the makros and also Updated Klipper and the MCU again as in the steps mentioned (altough make flash doesn't work for me).
Still no Luck :(
[image]

Why does it shop up 2 times with an host-entry? is there maybe a way to remove that fluidd entry?

 - I've updated the MCU also with the KIAUH update Script and uninstalled installed Klipper, Moonraker, Fluidd, KlipperScreen multiple times....

I guess my only chance is to start from scratch :-(
Logfiles:
[klippy (1).log](https://github.com/Klipper3d/klipper/files/7614257/klippy.1.log)

Angelol125, 2021-11-28:
I guess u also use ur rpi as an mcu for input shaper. that also needs to be updated.

Sineos, 2021-11-29:
Yep, sorry, missed this one:
[image]
Basically the message is quite clear: host and mcu have same version, fluiddpi is off

olif0x, 2021-11-30:
> Yep, sorry, missed this one: [image] Basically the message is quite clear: host and mcu have same version, fluiddpi is off
> 
> You need to build the linux mcu in `make menuconfig`

Yes that was it. 
I also had another friend looking at all those configs and just at the end he also mentioned it.

and now that i remember, when i first wanted to update the mcu i was wondering why linux processes where activated in make menuconfig :-D which i completely ignored and changed everything so that the printer board gets updated :D

well, lesson learned ;-) 
Printing works again (altough in the full end i just reinstalled fluiddpi to have a clean start and then updated the linux mcu as then came the hint in with the linux process).
Then everything was working again :-)

Many thanks for the help!

closed 2021-11-30 by olif0x