Klipper3d/klipper #4973: Clone 3DTouch won't work properly with 4.2.7

gnutison, 2021-11-28:
Hi, I am a newbie with Klipper, recently upgraded from Jyers to Klipper. And have little issues with clone 3Dtouch. I've been fighting with it the whole weekend, and I am on the edge to swap back to Jyers. I hope someone can help.

Raspberry Pi Zero 2 W
Ender 3v2 4.2.7, Dual Z
MS DD + Clone 3D Touch connected directly to ZProbe port. The wiring is correct, it works well with Jysers, everything is fine. 
I tested with BTT SKR Mini E3 V2 connected directly to the ZProbe port works fine too. 

So after installing Klipper and config the files, i start with Initial tests for BL-Touch [follow this tutorials](https://www.klipper3d.org/BLTouch.html)
When turn printer ON - pin going down and up 2x - testing if BLTouch is working.
BLTOUCH_DEBUG COMMAND=pin_down : pin go down
QUERY_PROBE : probe: open
BLTOUCH_DEBUG COMMAND=touch_mode
Pushing pin-up, by finger
QUERY_PROBE : probe: open (it should be triggered) <- so i stuck here
[follow this steps for clone](https://klipper.info/helpful-tips-1/bltouch-tips) add: pin_up_reports_not_triggered: False (wont help)

I cant not Home All because 3Dtouch will not trigger and nozzle will crash into bed
Thanks for the help and tips. I searched the forum and the whole weekend on the internet for problems. 

`[stepper_x]
step_pin: PB9
dir_pin: PC2
enable_pin: !PC3
microsteps: 16
rotation_distance: 40
endstop_pin: ^PA5
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PB7
dir_pin: PB8
enable_pin: !PC3
microsteps: 16
rotation_distance: 40
endstop_pin: ^PA6
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PB5
dir_pin: !PB6
enable_pin: !PC3
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
#endstop_pin: ^PA7
#position_endstop: -2.8
position_min: -5
position_max: 250

[extruder]
max_extrude_only_distance: 100.0
step_pin: PB3
dir_pin: PB4
enable_pin: !PC3
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PA1
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC5
#control: pid
#pid_Kp: 21.527
#pid_Ki: 1.063
#pid_Kd: 108.982
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PA2
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC4
#control: pid
#pid_Kp: 54.027
#pid_Ki: 0.770
#pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PA0

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[virtual_sdcard]
path: ~/gcode_files

[display_status]

[pause_resume]

[bltouch]
sensor_pin: PB1
control_pin: PB0
pin_up_reports_not_triggered: False
x_offset: -45
y_offset: -10
z_offset: 0
pin_move_time: 0.4
speed: 10

[display]
lcd_type: st7920
cs_pin: PB12
sclk_pin: PB13
sid_pin: PB15
encoder_pins: ^PB14, ^PB10
click_pin: ^!PB2

[safe_z_home]
home_xy_position: 155,120
speed: 150
z_hop: 10
z_hop_speed: 10

[bed_mesh]
speed: 120
horizontal_move_z: 5
mesh_min: 32,6  
mesh_max: 172, 210
probe_count: 3,3
mesh_pps: 2,2

[gcode_macro CANCEL_PRINT]
description: Cancel the actual running print
rename_existing: CANCEL_PRINT_BASE
gcode:
  TURN_OFF_HEATERS
  CANCEL_PRINT_BASE`
[klippy.log](https://github.com/Klipper3d/klipper/files/7614406/klippy.log)

Sineos, 2021-11-29:
See https://klipper.discourse.group/t/bltouch-creality-427-on-ender-5/533

gnutison, 2021-11-29:
Hi, I reach this post am yesterday and go a bit overnight, finale Z Home work without crashing into bed, but when I follow the BL-touch test after BLTOUCH_DEBUG COMMAND=touch_mode
Pushing pin-up, by finger
QUERY_PROBE : probe: open (it should be triggered) ? so idk

Sineos, 2021-11-29:
It should report triggered, yes. There are reports that it does not, but cause is unknown. See https://github.com/Klipper3d/klipper/issues/4966

gnutison, 2021-11-29:
Ok, thank you!

pietchaki, 2021-11-30:
Until the issue is fixed, I'd say to update the tutorial to include instructions on how to safely test if G28 works.

pietchaki, 2021-11-30:
@gnutison You installed Klipper directly in Octoprint? or used the Fluidd option?
Mine is using Fluidd, Raspberry 3B+, Arduino Mega and RAMPS 1.4. 
[Here is my current printer.cfg](https://gist.github.com/pietchaki/c71cc508770ef275ccf0671bde1b6e93).

gnutison, 2021-11-30:
Hello, I using Fluidd with Raspberry pi w2. I can call is "work", thanks to Sineos for the first links. I follow the official tutorials and seems won't work for me, but some guys in the link have the same issues and "removed pin_up_reports_not_triggered: False" after that, and adding samples x, it finally "work" but as I said there is still a bug, when QUERY_PROBE : probe: open <always. Anyway, I am very happy with Klipper I can push down Benchy under 1hours print with insane quality. 
[image]
[image]

KevinOConnor, 2022-01-04:
See PR #5088.

-Kevin

closed 2022-01-11 by github-actions[bot]