Klipper3d/klipper #4984: MCU 'mcu' shutdown: Not a valid ADC pin Once the underlying issue is corrected

pure100kim, 2021-12-01:
klipper installed  24 oct in my printer. 
few days ago, After update not connection.

Board : artillery ruby v1.2 
mcu : stm32f401

Extruder max_extrude_ratio=0.266081
mcu 'mcu': Starting serial connect
webhooks client 3042801864: New connection
webhooks client 3042801864: Client info {'program': 'Moonraker', 'version': 'v0.7.1-194-geda5623'}
Loaded MCU 'mcu' 94 commands (v0.10.0-169-g520273e5 / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1)
MCU 'mcu' config: BUS_PINS_i2c1=PB6,PB7 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_spi3a=PC11,PC12,PC10 BUS_PINS_spi2a=PC2,PC3,PB10 BUS_PINS_i2c1a=PB8,PB9 RESERVE_PINS_USB=PA11,PA12 CLOCK_FREQ=84000000 BUS_PINS_spi1a=PB4,PB5,PB3 STATS_SUMSQ_BASE=256 BUS_PINS_spi3=PB4,PB5,PB3 STEPPER_BOTH_EDGE=1 ADC_MAX=4095 BUS_PINS_spi4=PE13,PE14,PE12 PWM_MAX=255 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi1=PA6,PA7,PA5 RESERVE_PINS_crystal=PH0,PH1 MCU=stm32f401xc
mcu_temperature 'mcu' nominal base=-274.640000 slope=1310.400000
Sending MCU 'mcu' printer configuration...
MCU 'mcu' shutdown: Not a valid ADC pin
clocksync state: mcu_freq=84000000 last_clock=5223140602 clock_est=(1649.496 5194485661 84006666.799) min_half_rtt=0.000071 min_rtt_time=1649.632 time_avg=1649.496(0.017) clock_avg=5194485661.922(1436534.149) pred_variance=5202473096.471
Dumping serial stats: bytes_write=830 bytes_read=3856 bytes_retransmit=9 bytes_invalid=0 send_seq=109 receive_seq=109 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0
Dumping send queue 97 messages

..ommit...

Receive: 79 1649.838957 1649.838771 10: seq: 10, config is_config=0 crc=0 is_shutdown=0 move_count=0
Receive: 80 1649.846685 1649.846418 12: seq: 19, shutdown clock=928951545 static_string_id=Not a valid ADC pin
Transition to shutdown state: MCU 'mcu' shutdown: Not a valid ADC pin
Dumping 10 requests for client 3042801864
Received 1649.259458: {"id": 2815599312, "method": "info", "params": {"client_info": {"program": "Moonraker", "version": "v0.7.1-194-geda5623"}}}
Received 1649.262283: {"id": 2815601904, "method": "list_endpoints", "params": {}}
Received 1649.265368: {"id": 2805270960, "method": "objects/subscribe", "params": {"objects": {"webhooks": null}, "response_template": {"method": "process_status_update"}}}
Received 1649.268039: {"id": 2805587600, "method": "gcode/subscribe_output", "params": {"response_template": {"method": "process_gcode_response"}}}
Received 1649.522833: {"id": 2805272080, "method": "info", "params": {}}
Received 1649.525504: {"id": 2805270096, "method": "info", "params": {}}
Received 1649.528689: {"id": 2805671344, "method": "list_endpoints", "params": {}}
Received 1649.566680: {"id": 2805270096, "method": "info", "params": {}}
Received 1649.783648: {"id": 2815601904, "method": "info", "params": {}}
Received 1649.785298: {"id": 2816052976, "method": "list_endpoints", "params": {}}
Reactor garbage collection: (1649.427591942, 0.0, 0.0)
MCU error during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 162, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 730, in _connect
    config_params = self._send_get_config()
  File "/home/pi/klipper/klippy/mcu.py", line 707, in _send_get_config
    self._name, self._shutdown_msg))
error: MCU 'mcu' error during config: Not a valid ADC pin
Build file /home/pi/klipper/klippy/../.config(2329): Wed Dec  1 11:38:42 2021
========= Last MCU build config =========
CONFIG_LOW_LEVEL_OPTIONS=y
# CONFIG_MACH_AVR is not set
# CONFIG_MACH_ATSAM is not set
# CONFIG_MACH_ATSAMD is not set
# CONFIG_MACH_LPC176X is not set
CONFIG_MACH_STM32=y
# CONFIG_MACH_RP2040 is not set
# CONFIG_MACH_PRU is not set
# CONFIG_MACH_LINUX is not set
# CONFIG_MACH_SIMU is not set
CONFIG_BOARD_DIRECTORY="stm32"
CONFIG_MCU="stm32f401xc"
CONFIG_CLOCK_FREQ=84000000
CONFIG_USBSERIAL=y
CONFIG_FLASH_START=0x8000000
CONFIG_FLASH_SIZE=0x40000
CONFIG_RAM_START=0x20000000
CONFIG_RAM_SIZE=0x10000
CONFIG_STACK_SIZE=512
CONFIG_STM32_SELECT=y
# CONFIG_MACH_STM32F103 is not set
# CONFIG_MACH_STM32F207 is not set
CONFIG_MACH_STM32F401=y
# CONFIG_MACH_STM32F405 is not set
# CONFIG_MACH_STM32F407 is not set
# CONFIG_MACH_STM32F429 is not set
# CONFIG_MACH_STM32F446 is not set
# CONFIG_MACH_STM32F031 is not set
# CONFIG_MACH_STM32F042 is not set
# CONFIG_MACH_STM32F070 is not set
# CONFIG_MACH_STM32F072 is not set
# CONFIG_MACH_STM32H743 is not set
# CONFIG_MACH_STM32H750 is not set
CONFIG_MACH_STM32F4=y
CONFIG_HAVE_STM32_USBOTG=y
CONFIG_HAVE_STM32_CANBUS=y
# CONFIG_STM32_FLASH_START_4000 is not set
CONFIG_STM32_FLASH_START_0000=y
CONFIG_STM32_CLOCK_REF_8M=y
# CONFIG_STM32_CLOCK_REF_12M is not set
# CONFIG_STM32_CLOCK_REF_16M is not set
# CONFIG_STM32_CLOCK_REF_25M is not set
# CONFIG_STM32_CLOCK_REF_INTERNAL is not set
CONFIG_CLOCK_REF_FREQ=8000000
CONFIG_STM32F0_TRIM=16
CONFIG_STM32_USB_PA11_PA12=y
# CONFIG_STM32_SERIAL_USART1 is not set
# CONFIG_STM32_SERIAL_USART1_ALT_PB7_PB6 is not set
# CONFIG_STM32_SERIAL_USART2 is not set
# CONFIG_STM32_SERIAL_USART2_ALT_PD6_PD5 is not set
# CONFIG_STM32_CANBUS_PA11_PA12 is not set
# CONFIG_STM32_CANBUS_PB8_PB9 is not set
# CONFIG_STM32_CANBUS_PI9_PH13 is not set
# CONFIG_STM32_CANBUS_PB5_PB6 is not set
# CONFIG_STM32_CANBUS_PB12_PB13 is not set
# CONFIG_STM32_CANBUS_PD0_PD1 is not set
CONFIG_CANBUS_FREQUENCY=500000
CONFIG_USB_VENDOR_ID=0x1d50
CONFIG_USB_DEVICE_ID=0x614e
# CONFIG_USB_SERIAL_NUMBER_CHIPID is not set
CONFIG_USB_SERIAL_NUMBER="12345"

pure100kim, 2021-12-01:
I found it 
please close it

closed 2021-12-01 by Sineos

R2-G2, 2021-12-07:
@pure100kim how did you manage to set up the Ruby v1.2 at all? I'm struggling. ;)