Klipper3d/klipper #4985: Nozzle diameter in extruder section--Allow larger max value

lunatic1972, 2021-12-01:
Recently I had gotten some Bondtech CHT nozzles which allow you to print with a nozzle size up to 1.8mm. Is it probably crazy sure but it does work under the right circumstances. When I went to change the filament diameter under the extruder section to 2.0 to allow a little wiggle room and did a reboot it errored out. So I changed it to 1.4 however I feel that if my flow goes above a certain point the print will error out and I'm assuming there is some kind of limitation set in the software to not go above a certain point

Sineos, 2021-12-01:
See `max_extrude_cross_section`: https://www.klipper3d.org/Config_Reference.html?#extruder

lunatic1972, 2021-12-01:
I thought it defaulted to 4 * nozzle_diameter^2? So if I change it(max_extrude_cross_section) then it won't matter about nozzle diameter?

Sineos, 2021-12-01:
Klipper, as a printer firmware, does what the gcode is commanding to do. There is no second guessing or hidden optimization etc.
The only thing Klipper does is to "scrutinize" the gcode if there is anything that would cause issues or even a safety impact.

As such `nozzle_diameter`, `filament_diameter` and `max_extrude_cross_section` are used to determine a sane amount of extrusion that would be expected for typical printing. If this "sane amount" is exceeded, Klipper will error out and give you the notification that something unwanted / unexpected has happened.

Setting
`max_extrude_cross_section: 16`
is effectively the same as setting `nozzle_diameter` to 2.0 and not setting `max_extrude_cross_section` at all.

lunatic1972, 2021-12-02:
Thanks for the help. I'll try it out but I think we can close this one out since there's an alternative to using nozzle diameter.

Sineos, 2021-12-02:
Closed as concluded.

closed 2021-12-02 by Sineos