Klipper3d/klipper #4987: Cannot set input_shaper shaper_type

Burtfaceman, 2021-12-02:
When attempting to set an input shaper type in my config file, I get the following error:
[image]

Here's the input_shaper section of my config file:
> [input_shaper]
> shaper_freq_x = 45.2
> shaper_freq_y = 45.5
> shaper_type = MZV

Config file attached
[printer.cfg.txt](https://github.com/Klipper3d/klipper/files/7639405/printer.cfg.txt)

Gilabite, 2021-12-02:
Try it in lowercase mzv

AndyWo83, 2021-12-02:
this is the newer input_shaper section:
[image]

And lowercase mzv is the better one...

Burtfaceman, 2021-12-04:
Thanks, that worked. Documentation tends to use uppercase for shaper types. This behavior is unexpected.

closed 2021-12-04 by Burtfaceman