Klipper3d/klipper #4989: Issue setting up BLTouch

KC6331, 2021-12-03:
I have a ANTClabs Bltouch I have been trying to set up on my CR 10/SKR 2 rig.  When I save and restart, I get the following error:
"Option 'position_endstop' is not valid in section 'stepper_z'"

Attached is my printer.cfg and the klippy log

[gcode_macro START_PRINT]
gcode:
    # Set and wait for bed to reach temperature
    BED_MESH_CLEAR
      G29
    # Move Z up
    G1 Z10.0 F3000
    # Reset extruder
    G92 EO1
    # Move Z axis up a little to prevent scratching heat bed
    G1 Z2.0 F3000
     Move to start position
    G1 X0.1 Y20 Z5.0 F5000.0
    # Move to start for purge line
    G1 X0.1 Y20 Z0.3 F5000.0
    # Draw 1st line
    G1 X0.1 Y150.0 Z0.3 F1500.0 E15
    # Move to side a little
    G1 X0.4 Y150.0 Z0.3 F5000.0
    # Draw 2nd line
    G1 X0.4 Y20 Z0.3 F1500.0 E30
    # Reset extruder
    G92 E0
    # Move Z up a little
    G1 Z5.0 F3000

[gcode_macro END_PRINT]
gcode:
    G91
    # Retract a bit
    G1 E-2 F2700
    G1 E-2 Z0.2 F2400
    G1 X5 Y5 F3000
    # Turn off bed, extruder, and fan
    M140 S0
    M104 S0
    M106 S0
    # Raise nozzle by 10mm
    G1 Z10 F3000
    G90
    # Deliver print
    G1 X0 Y220
    # Disable steppers
    M84 X Y E
    # Clear bed mesh
    BED_MESH_CLEAR
    # Turn off gcode offset
    SET_GCODE_OFFSET Z=0

[stepper_x]
step_pin: PE2
dir_pin: !PE1
enable_pin: !PE3
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC1
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_y]
step_pin: PD5
dir_pin: !PD4
enable_pin: !PD6
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC3
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_z]
step_pin: PA15
dir_pin: PA8
enable_pin: !PD1
microsteps: 16
rotation_distance: 8.1
position_endstop: 0.0
position_max: 400
endstop_pin: probe:z_virtual_endstop

[extruder]
step_pin: PD15
dir_pin: !PD14
enable_pin: !PC7
microsteps: 16
rotation_distance: 23.129
nozzle_diameter: 0.600
filament_diameter: 1.750
heater_pin: PB3
sensor_type: NTC 100K beta 3950
sensor_pin: PA2
control: pid
pid_kp = 28.224
pid_ki = 1.742
pid_kd = 114.306
min_temp: 0
max_temp: 250
max_extrude_only_distance: 120.0
pressure_advance: 0.051
min_extrude_temp: 0

[heater_bed]
heater_pin: PD7
sensor_type: NTC 100K beta 3950
sensor_pin: PA1
control: pid
pid_Kp: 58.035
pid_Ki: 0.557
pid_Kd: 1512.549
min_temp: 0
max_temp: 130

[bltouch]
sensor_pin: ^PE4
control_pin: PE5
stow_on_each_sample: False
probe_with_touch_mode: True
x_offset: -43
y_offset: -7
z_offset =0.01
speed: 4.0
samples: 2
sample_retract_dist: 3.0

[safe_z_home]
home_xy_position: 150, 150
speed: 50
z_hop: 10
z_hop_speed: 5

[bed_mesh]
speed: 120
horizontal_move_z: 10
mesh_min: 10,30
mesh_max: 295,295
probe_count: 3,3
algorithm: bicubic
fade_start: 1
fade_end:10
fade_target: 0

[filament_switch_sensor RunoutSensor]
pause_on_runout: False
runout_gcode: PAUSE
insert_gcode: RESUME
switch_pin: PC2

[fan]
pin: PB7

[heater_fan fan1]
pin: PB6

[heater_fan fan2]
pin: PB5

[output_pin motor_power]
pin: PC13
value: 1

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f407xx_420033001047393437303337-if00

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 500
max_z_velocity: 10
max_z_accel: 500

[virtual_sdcard]
path: ~/gcode_files
[pause_resume]

[display_status]

[pause_resume]

[gcode_macro CANCEL_PRINT]
rename_existing: BASE_CANCEL_PRINT
gcode:
  TURN_OFF_HEATERS
  CLEAR_PAUSE
  SDCARD_RESET_FILE
  BASE_CANCEL_PRINT

[gcode_macro RESUME]
rename_existing: BASE_RESUME
gcode:
  {% set E = printer["gcode_macro PAUSE"].extrude|float %}
  G91
  G1 E{E} F2100
  RESTORE_GCODE_STATE NAME=PAUSE_state
  BASE_RESUME

[gcode_macro PAUSE]
rename_existing: BASE_PAUSE
variable_extrude: 1.0
gcode:
  {% set E = printer["gcode_macro PAUSE"].extrude|float %}
  {% set x_park = printer.toolhead.axis_maximum.x|float - 5.0 %}
  {% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}
  {% set max_z = printer.toolhead.axis_maximum.z|float %}
  {% set act_z = printer.toolhead.position.z|float %}
  {% if act_z < (max_z - 2.0) %}
      {% set z_safe = 2.0 %}
  {% else %}
      {% set z_safe = max_z - act_z %}
  {% endif %}
  SAVE_GCODE_STATE NAME=PAUSE_state
  BASE_PAUSE
  G91
  G1 E-{E} F2100
  G1 Z{z_safe} F900
  G90
  G1 X{x_park} Y{y_park} F6000

########################################
# EXP1 / EXP2 (display) pins
########################################

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PC5, EXP1_3=PB1, EXP1_5=PE10, EXP1_7=PE12, EXP1_9=<GND>,
    EXP1_2=PB0, EXP1_4=PE9, EXP1_6=PE11, EXP1_8=PE13, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=PA6, EXP2_3=PE7, EXP2_5=PB2, EXP2_7=PC4,   EXP2_9=<GND>,
    EXP2_2=PA5, EXP2_4=PA4, EXP2_6=PA7, EXP2_8=<RST>, EXP2_10=<NC>

# See the sample-lcd.cfg file for definitions of common LCD displays.

########################################
# TMC2209 configuration
########################################

[tmc2209 stepper_x]
uart_pin: PE0
run_current: 0.580
hold_current: 0.580
diag_pin:

[tmc2209 stepper_y]
uart_pin: PD3
run_current: 0.650
hold_current: 0.650
diag_pin:

[tmc2209 stepper_z]
uart_pin: PD0
run_current: 1.3
hold_current: 1.25
diag_pin:

[tmc2209 extruder]
uart_pin: PD12
run_current: 0.750
hold_current: 0.500
diag_pin:

[klippy (8).log](https://github.com/Klipper3d/klipper/files/7646377/klippy.8.log)

Gilabite, 2021-12-03:
you dont use position_endstop with a bltouch so comment it out

KC6331, 2021-12-03:
Did as mentioned, now I get the following:
"Option 'position_min' is not valid in section 'bltouch'"

Gilabite, 2021-12-03:
That goes in your stepper_z section not bltouch

Sineos, 2021-12-30:
If this is still an issue please open a new post on https://klipper.discourse.group and we will be happy to assist

closed 2021-12-30 by Sineos