Klipper3d/klipper #4990: PT1000 temperature problem

panik988, 2021-12-03:
Hello,
after updating to the last release I had problem with the extruder thermistor.
I'm using a PT1000 for the extruder and Generic 3950 for bed amd chamber.
The room temp is 17C and the 3950 say 17C but PT1000 say 29.7C

Now I'm running v0.10.0-172-g051133f8 of Klipper but in th
e before versions I don't have the issue.

What should I do?

Klippy log attached below

[klippy.log](https://github.com/Klipper3d/klipper/files/7648791/klippy.log)

Thanx

Sineos, 2021-12-03:
Do you have a PT100 or PT1000? The values in your cfg indicate a PT100:

```
rtd_nominal_r = 100
rtd_reference_r = 430
```

panik988, 2021-12-03:
I've got a PT1000 (thousand)
[image]

panik988, 2021-12-03:
> Do you have a PT100 or PT1000? The values in your cfg indicate a PT100:
> 
> ```
> rtd_nominal_r = 100
> rtd_reference_r = 430
> ```

Sineos, 2021-12-03:
Ah, sorry, my fault. Wrong config

Well, it should at least kind of work with a directly connected PT1000 and I would not be aware that there has been any changes that could affect this just recently

panik988, 2021-12-03:
I'd tryed 3 different PT1000 and all say 29C instead 18C
Untill yesterday all work fine but after the update I have thi issue so I think something does't work properly with the update

panik988, 2021-12-07:
Well, I change the sensor port on the mobo and now the temperature readings are good.
Other EPCOS thermistors on the PT1000 port work fine, can't undertand why this happened after an update.

Sineos, 2021-12-08:
Issue seems concluded. Please open a new one if new problems arise. Preferable use https://klipper.discourse.group for general support topics.

closed 2021-12-08 by Sineos