Klipper3d/klipper #4992: Startup issue

sthgardner, 2021-12-04:
Keep getting this. Was working with Marlin, no hardware changes.

MCU 'mcu' shutdown: ADC out of range
This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdow
[klippy.log](https://github.com/Klipper3d/klipper/files/7654547/klippy.log)
[printer.txt](https://github.com/Klipper3d/klipper/files/7654550/printer.txt)
n

Sineos, 2021-12-05:
According to the leg you have multiple issues:

```
Configured MCU 'mcu' (1024 moves)
TMC stepper_y failed to init: Unable to read tmc uart 'stepper_y' register IFCNT
TMC extruder failed to init: Unable to read tmc uart 'extruder' register IFCNT
Starting heater checks for heater_bed
Failed to verify BLTouch probe is raised; retrying.
MCU 'mcu' shutdown: ADC out of range
```

You will need to find the root cause, as certain errors could be consequential from other errors:
- https://www.klipper3d.org/TMC_Drivers.html#i-keep-getting-unable-to-read-tmc-uart-stepper_x-register-ifcnt-errors
- https://github.com/Klipper3d/klipper/issues/4945#issuecomment-975737940

Sineos, 2021-12-30:
If this is still an issue please open a new post on https://klipper.discourse.group and we will be happy to assist

closed 2021-12-30 by Sineos