Klipper3d/klipper #4995: The klippy host software is attempting to connect.

filcsix, 2021-12-05:
Hi!
I'm bulinding a Voron Legacy and wanted a Klipper firmware to it. I have a RPI-zero, an SKR mini 2, and tft35 display. I watched NERO 3D video's video about it and doing the stuff step by step. But in the very end I got this message:
Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.
I already checked the printer.cfg and tried to restart but nothing happend.
So the question is where am I screw up? 

Thanks in advance!
filcsix

[klippy.log](https://github.com/Klipper3d/klipper/files/7655989/klippy.log)

[printer.txt](https://github.com/Klipper3d/klipper/files/7655997/printer.txt)

Sineos, 2021-12-05:
Please follow https://www.klipper3d.org/Installation.html and in particular the section `Configuring Klipper` on how to obtain the correct serial address.

For the correct settings for your board refer to https://github.com/Klipper3d/klipper/blob/master/config/generic-bigtreetech-skr-mini-e3-v2.0.cfg

Sineos, 2021-12-30:
Issue seems concluded. If you need support, we will be happy to assist on https://klipper.discourse.group

closed 2021-12-30 by Sineos