Klipper3d/klipper #4999: SKR 1.4 hotend temp reading, Error

Koinian, 2021-12-06:
I'm running mainsail OS on my BIQU B1. Everything went great until a print failed. I switched off the printer and cleared the mess off the hot end. Once I started up the printer again my hot end sensor read >600°C. I have changed thermistors as well as the BTT B1 HOTMODE V1.0 LED Adapter Board with no success, I also tried the various sensor types as shown in the klipper configuration reference also with no success.

I'm using an 100K 1% NTC Thermistor.

I would really appreciate it if someone can help me or even just point me in the right direction.

Note: the klippy log was take with only the thermistor attached to the hot end assembly.
[klippy.log](https://github.com/Klipper3d/klipper/files/7661499/klippy.log)
.

Sineos, 2021-12-06:
Most likely some hardware defect. Broker thermistor, broken cable, lose connector etc.

Koinian, 2021-12-08:
Thank you. I opened up my machine and figured out the the TH0 pin was not working anymore. Swapped to TH1 and now the problem is solved.

closed 2021-12-08 by Koinian