Klipper3d/klipper #5000: MCU 'mcu' shutdown: Timer too close

Chry113, 2021-12-06:
Suddenly the print stopped giving the error "MCU 'mcu' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown ". I notice a high system load even when the print is still, the value varies between 4.6 and 5.76 (I have another identical printer always with klipper which is 0.4 in idle). it had never happened to me and in git (in some cases the same kevin replied) it seems attributable to the pa. with care slicer I had foreseen a change of flow at layer 71 (I have a layer height of 0.2 and the print stopped at 3.28mm in height). the PA value is 0.107.
in the klippy log I read :
Receive: 99 99538.575435 99538.574971 12: seq: 1e, shutdown clock=359896519 static_string_id=Command request
Virtual sdcard (2621139): 'X149.296 Y159 E3244.27766\nG1 X149.325 Y159.339 E3244.2885\nG1 X149.413 Y159.668 E3244.29935\nG1 X149.557 Y159.975 E3244.31016\nG1 X149.752 Y160.254 E3244.321\nG1 X149.993 Y160.495 E3244.33186\nG1 X150.272 Y160.69 E3244.34271\nG1 X150.579 Y160.834 E3244.35351\nG1 X150.908 Y160.922 E3244.36437\nG0 F10800 X150.833 Y161.355\nG1 F2400 E3243.36437\nG0 F10800 X177.866 Y177.928\n;TYPE:WALL-INNER\nG1 F2400 E3244.36437\nG1 F4800 X178.205 Y177.957 E3244.37521\nG1 X222.325 Y177.992 E3245.78101\nG1 X222.87 Y177.905 E3245.7986\nG1 X223.432 Y177.617 E3245.81872\nG1 X223.878 Y177.166 E3245.83893\nG1 X224.16 Y176.607 E3245.85888\nG1 X224.247 Y176.064 E3245.8764\nG1 X224.247 Y128.659 E3247.38688\nG1 X224.132 Y128.049 E3247.40666\nG1 X223.782 Y127.449 E3247.42879\nG1 X223.246 Y127.003 E3247.45101\nG1 X222.629 Y126.779 E3247.47192\nG1 X222.292 Y126.749 E3247.4827\nG1 X178.157 Y126.749 E3248.88898\nG1 X177.549 Y126.864 E3248.9087\nG1 X176.943 Y127.214 E3248.931\nG1 X176.501 Y127.75 E3248.95314\nG1 X176.276 Y128.367 E3248.97406\nG1 X176.247 Y128.706 E3248.9849\n'
Upcoming (2622163): 'G1 X176.249 Y176.045 E3250.49327\nG1 X176.361 Y176.658 E3250.51313\nG1 X176.716 Y177.26 E3250.5354\nG1 X177.25 Y177.703 E3250.55751'
Reactor garbage collection: (99527.957208231, 99452.804403624, 98366.038393934)
Requested toolhead position at shutdown time 9971.632689: (150.908, 160.922, 3.270438168776423)
Exiting SD card print (position 2622196)
Stats 99539.4: gcodein=0 mcu: mcu_awake=0.002 mcu_task_avg=0.000016 mcu_task_stddev=0.000010 bytes_write=12631017 bytes_read=4482416 bytes_retransmit=181 bytes_invalid=0 send_seq=286445 receive_seq=286445 retransmit_seq=286379 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=72003260 rpi: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000004 bytes_write=61477 bytes_read=191039 bytes_retransmit=0 bytes_invalid=0 send_seq=10223 receive_seq=10223 retransmit_seq=0 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=50002374 adj=49999941 Raspberry_pi: temp=38.6 Robin_nano35: temp=35.8 heater_bed: target=75 temp=75.0 pwm=0.835  sysload=4.74 cputime=745.978 memavail=719456 print_time=9974.133 buffer_time=1.505 print_stall=2 extruder: target=205 temp=204.8 pwm=0.691
stepcompress o=4 i=0 c=17 a=0: Invalid sequence
stepcompress o=4 i=0 c=17 a=0: Invalid sequence
stepcompress o=4 i=0 c=17 a=0: Invalid sequence
stepcompress o=4 i=0 c=17 a=0: Invalid sequence
Exception in flush_handler
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/toolhead.py", line 392, in _flush_handler
    self.flush_step_generation()
  File "/home/pi/klipper/klippy/toolhead.py", line 337, in flush_step_generation
    self.move_queue.flush()
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 333, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 286, in _update_move_time
    sg(sg_flush_time)
  File "/home/pi/klipper/klippy/stepper.py", line 209, in generate_steps
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress
Stats 99541.4: gcodein=0 mcu: mcu_awake=0.002 mcu_task_avg=0.000016 mcu_task_stddev=0.000010 bytes_write=12631160 bytes_read=4482820 bytes_retransmit=181 bytes_invalid=0 send_seq=286454 receive_seq=286453 retransmit_seq=286379 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=72003260 rpi: mcu_awake=0.000 mcu_task_avg=0.000004 mcu_task_stddev=0.000004 bytes_write=61483 bytes_read=191039 bytes_retransmit=0 bytes_invalid=0 send_seq=10224 receive_seq=10223 retransmit_seq=0 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=50002374 adj=50001133 Raspberry_pi: temp=38.1 Robin_nano35: temp=37.4 heater_bed: target=75 temp=75.0 pwm=0.527  sysload=4.60 cputime=746.015 memavail=732280 print_time=9974.299 buffer_time=0.000 print_stall=2 extruder: target=205 temp=205.4 pwm=0.582

dmeagor, 2021-12-07:
Interesting, I just came here to post/research a similar issue (same timer too close).  I don't want to hijack your issue so I'll post my logs separately but when I saw the words "robin nano" I thought I'd better add a post.  

Also using the robin nano and RaspPi Zero (wired internally to the nano's TX/RX/5v/GND.)  Everything was stable until the the last firmware update. 

Mine seems to go wrong printing cogs exported by the latest superslicer using the accelerations supplied in the standard sapphire pro klipper config.

Are you also using the Pi Zero?  I was thinking of upgrading it but I wasn't sure whether the issue was the noisy robin nano which I was also half considering binning.

Sineos, 2021-12-07:
Generally speaking, a Pi Zero is very weak and does not leave much (none) headroom for any other tasks like Octoprint or other web front ends. A `Timer too close` is often an indication for a maxed out host.

```
stepcompress o=4 i=0 c=17 a=0: Invalid sequence
stepcompress o=4 i=0 c=17 a=0: Invalid sequence
stepcompress o=4 i=0 c=17 a=0: Invalid sequence
stepcompress o=4 i=0 c=17 a=0: Invalid sequence
Exception in flush_handler
```

looks fishy. Maybe @KevinOConnor should have a look.

@Chry113 please attach the complete, unmodified `klippy.log` with this error

Chry113, 2021-12-07:
I use a raspberry pi 3 with robin nano v1. 2. This is the first time this error has occurred. I have two same printers on the same hw. this where the error occurred I had installed all the updated software from scratch, from the OS, to kiauh etc. Made all the calibrations including adxl input shaper (there is an error in the call to matpol but the automatic calibration works perfectly). I attach the log file as soon as possible.

dmbutyugin, 2021-12-07:
Indeed without the full log it is difficult to tell if `stepcompress o=4 i=0 c=17 a=0: Invalid sequence` is the error that caused a print failure (which usually indicates a bug in the Klipper code), and 'Timer too close' is just a noise that happened afterwards, or if 'Timer too close' is really the initial error (which then indicates insufficient computing capacity of RPi or the MCU board for the requested step rate).

Chry113, 2021-12-07:
[klippy (2).log](https://github.com/Klipper3d/klipper/files/7672228/klippy.2.log)

I hope there is a solution. a little while ago I lowered the microsteps to 32 (I repeat I deactivated hold courrent and interpolation) but it returned the same error to the 4 layer ...