Klipper3d/klipper #5001: Transition to shutdown state: MCU 'mcu' shutdown: Timer too close - when [gcode_arcs] enabled

kaito83, 2021-12-07:
I recently enabled the arc feature in Klipper,  in Cura the arc settings:
G90 Influences extruder unchecked
Resolution: 0.05
Path tolerance: 0.05
Max arc radius: 1000000.0
Allow dynamic precision checked
FW compensation unchecked
Klipper 
[gcode_arcs]
resolution: 0.01

The print stops at the same point, what is the optimal settings or is there a bug in the Klipper?

[klippy.zip](https://github.com/Klipper3d/klipper/files/7666726/klippy.zip)

Sineos, 2021-12-07:
Using arcs in Klipper does not serve any real purpose:
- Arcs can help in Marlin to mitigate the serial bottleneck between Octoprint and the firmware by reducing the amount of data transferred
- Klipper completely eliminates this bottleneck with the virtual SD card feature
- There is no quality benefit associated with arcs whatsoever, on the contrary, in the Klipper world it rather reduces precision

If it gives you issues, just turn it off.

kaito83, 2021-12-07:
Thx, the fast response, I keep it disabled!

closed 2021-12-07 by kaito83