Klipper3d/klipper #5002: MCU Shutdown Timer Too Close, Robin Nano + RPI Zero.

blendfish, 2021-12-07:
Updated to latest klipper firmware using Kiauh as normal, no longer able to print for any length of time before the above error happens.

This is on a Robin nano with RPI Zero.

If I slow down the print from the standard settings to 1000 acceleration then it seems to work ok.  I'm happy to upgrade the Pi but I'm not sure that's the real issue as it's been working on for at least a year.

None of the recommendations given seem to be the case, I monitored the CPU and it was around 50%, voltage comes directly from the robin nano motherboard via the 5v line.  Data comes from a direct wire to the RX/TX line and there's no issue with the RPI SD Card.

[klippy.log](https://github.com/Klipper3d/klipper/files/7668587/klippy.log)
[moonraker.log](https://github.com/Klipper3d/klipper/files/7668589/moonraker.log)

blendfish, 2021-12-07:
Should have mentioned I'm using Fluidd as the frontend not octoprint.

blendfish, 2021-12-07:
I did another test using an external 5v official pi supply (data still connected via UART pins.)  

[image]

Klipper went over 200% (not sure how load can be over 100%?)  
It happened a few times, mostly caused a few pauses in the print but it eventually gave up.

I've ordered the new Pi Zero 2 which is apparently much more powerful, though I doubt I'm getting 200%+ speed on any current pi.

blendfish, 2021-12-07:
I discovered that going around the edges of these gears was the culprit.

[image]

Sineos, 2021-12-08:
Issue seems concluded. Please open a new one, if the problem persists with the new RPi

closed 2021-12-08 by Sineos