Klipper3d/klipper #5004: [Request] DWIN T5UID1 touchscreen support

LMS0815, 2021-12-07:
Kindly consider implementing T5UID1 touchscreen support into Klipper master branch.

[Desuuuu/klipper](https://github.com/Desuuuu/klipper)

Volanaro, 2022-01-20:
2nd this,

meesterwu, 2022-02-15:
4th this, thanks for all the great work!!!

ld3300, 2022-02-17:
Are there any interested developers that could submit this as a pull request?  I believe that the primary repo maintainers are pretty heavily involved in some other development tracks.

LaggAt, 2022-03-06:
I looked at the changes, which seems to be somewhat more than just the display code.

Just started an issue in the Desuuu repo, maybe we could get some help here:
https://github.com/Desuuuu/klipper/issues/113

Sineos, 2022-03-06:
Here @KevinOConnor would need to chime in: 
As far as I understand the current situation, all display topics are more or less on hold due to the need for having a solid code foundation first, i.e. to avoid patching display variant upon display variant with all the associated code duplication and maintenance burden.

Sineos, 2022-03-06:
That having said, it is preferable to continue such discussions on https://klipper.discourse.group

hakuyouhiro, 2022-03-26:
Are we covering this at https://klipper.discourse.group/c/features/9 ? i can't see the topic created

Sineos, 2022-03-27:
See https://klipper.discourse.group/t/dwin-t5uid1-touchscreen-support/2411

closed 2022-03-27 by Sineos

PeggyFree, 2022-04-23:
I support this request since the printers SOVOL SV02 and SOVOL SV04 also use this DWIN T5UID1 touchscreen

napter, 2022-08-22:
Is there any chance of this hpapeneing? It doesn't look like the discussion on the thread went anywhere. It would be great if there was support for the stock screen on the Ender 5 plus.

Rokin05, 2022-09-09:
+1
The Longer LK4 Pro printer also use this screen (and Desuuuu fw work fine on it)
(and surely Alfawise u30 pro who is the ~same printer)

blondsign, 2023-02-02:
That would be great

teichertw, 2023-02-22:
still waiting ...

stooged, 2023-03-08:
+1 for official support.

i have merged the current klipper repo to the dgus_reloaded fork for my own printers but i would love to see official support for the DWIN T5UID1 screens.

fryc88, 2023-03-15:
I would love to see official support as well. I'm using fork of Desuuu git to keep it up to date with main klipper branch but official support is always easier to handle.
Also screen is getting more popular, few types of same screen series is being used in more and more new printers so there will be more interested people I guess.

stooged, 2023-09-14:
I ended up just following this video and installed klipperscreen.
The dgus was cool but doing this mod was worth it.

https://m.youtube.com/watch?v=wt8o4NNgknY

fryc88, 2023-09-15:
Yeah, but he changed the screen, that's additional cost. Some people want to cut costs and didn't want to invest more.

stooged, 2023-09-15:
> Yeah, but he changed the screen, that's additional cost. Some people want to cut costs and didn't want to invest more.

Yeah i know, it cost me about $50 for the screen.
The way i looked at it was i could keep merging the klipper updates to the dgus repo which takes time and put up with the limited functionality of the dgus screen or i could spend the 50 and unlock the full klipperscreen ui and have official update support.

I was all for official dgus support but it looks like that is not going to be a thing :(

fryc88, 2023-09-15:
It's fairly easy to keep his branch updated, it can be merged with main branch pretty easy, even with automated merges weekly for exmaple ;) His files aren't involving anything major from Klipper, so it's doable.

PorcoMaster, 2023-10-27:
> It's fairly easy to keep his branch updated, it can be merged with main branch pretty easy, even with automated merges weekly for exmaple ;) His files aren't involving anything major from Klipper, so it's doable.

Hi, i don't know if i can ask this here, but you were able to update your klipper while using desuu firmware, i would love to test axis twist compensation, but desuu firmware is outdated from main branch, how did you update your firmware, is there any youtube or writen tutorial (preferred) 

thanks in advance, and if this is not the place, i will delete it.

fryc88, 2023-10-27:
I've gathered all details into one guide while I was playing with my Sovol SV06plus with this screen, a friend of mine made auto update for this branch, it still works as far as i'm aware.
https://github.com/fryc88/klipper-sv06plus-screen

fryc88, 2023-10-27:
I mainly mean branch. That gbkwiat is basically desuuus klipper branch but with auto merge weekly with main klipper branch so it's up to date. And if your screen is vertical in cr10 then it should be ok with these screen files too

PorcoMaster, 2023-10-27:
> I mainly mean branch. That gbkwiat is basically desuuus klipper branch but with auto merge weekly with main klipper branch so it's up to date. And if your screen is vertical in cr10 then it should be ok with these screen files too

Unfortunately it's horizontal screen, if it were vertical, I would for sure try it out, I at least learned about verbose -v and -g, did not make any difference just take out a warning in orange, so I reverted back. 

But I really appreciate your help, I just hope someday they will support T5UID1 on official repo.

fryc88, 2023-10-27:
It will never be added to official repo.
With orange not going away probably you missed some step or you need to give it more restart/refresh of updates to get rid of it, it won't go off on its own after adding parameters. 
With horizontal screen you can edit it or make your own firmware for it, however I'm pretty sure I've seen it somewhere too