Klipper3d/klipper #5005: Saved BED_MESH_PROFILE not visible in Heightmap

MalsakerLures, 2021-12-07:
I just updated to Klipper version: v0.10.0-172-g051133f8
Any bed_mesh_profiles saved using the command BED_MESH_PROFILE SAVE=<name> is stored on the config but are not visible or loadable under Heightmap.

When reviewing the config i can see the actual config has been written, but not visible under height map.

Attached are screenshots as well as klippy log
[klippy (6).log](https://github.com/Klipper3d/klipper/files/7671978/klippy.6.log)
[image]
[image]
g.

Sineos, 2021-12-08:
Your log contains:

```
Can't read autosave from config file - autosave state corrupted
```

and all the printer config information that is reflected in the log file do not contain any mesh information. I cannot really think of a reason, why this should happen, except manually tinkering with the `#*#` section or a corrupt SD card.
I would delete the entire `#*#` and repeat the process.

@KevinOConnor I wonder why Klipper does not shutdown on such an event

MalsakerLures, 2021-12-08:
Deleted the everything below  the save config parameters. Restarted. Ran a new bed mesh and tried to save the config.

Config saved and verified as i saw the entries in the .cfg, but still not able to see the saved profiles under heightmap.

MalsakerLures, 2021-12-08:
Ha, found the issue!

if the are sections other then the save_config the it dosent work. If i remove them, then it all works as expected.

 i had the following divider in my config previosly: #*#<——- Macro ——->

Sineos, 2021-12-30:
Issue seems concluded. If you need support, we will be happy to assist on https://klipper.discourse.group

closed 2021-12-30 by Sineos