Klipper3d/klipper #5007: Klipper stops responding when loads printer.cfg file

bulbersator, 2021-12-08:
Hi,
I have MKS E3D mainboard with RPI Zero 2W. 
The pair was working perfectly fine a couple days ago but I wanted to change to Fluidd. Now when I want to go back to the klipper with octoprint, I have an error that when the printer.cfg file is correct, the clipper stops responding and I have communication timeout in octoprint. I installed this a few days ago and I do everything exactly the same.
As long as the clipper reports an error in the printer.cfg file, there is contact with it, but when it reads it, the communication breaks off. Raspberry responds via SSH, but Octoprint reports no communication with the klipper. Please help.

When I installed fluidd, I had the same situation, but I thought it was a fluidd problem, and now I see the same in both cases.

I changed sd card, reinstalled everything, tried with RPI disconnected from everything and still the same.

[klippy.log](https://github.com/Klipper3d/klipper/files/7672364/klippy.log)

Sineos, 2021-12-08:
It seems your section

```
[mcu]
serial = /dev/ttyAMA0
```

is not correct. Please follow https://www.klipper3d.org/Installation.html#configuring-klipper how to obtain the correct connection path

bulbersator, 2021-12-08:
Hi, if you are referring to lack of the "reset" option, that's because I
was trying various versions to see if it changes anything. Even if the
config is not correct, there should be an error about no communication with
the printer, but in my case it's like the klipper service completely hangs
up so I don't think nthays the reason.

Sineos, 2021-12-08:
```
mcu 'mcu': Starting serial connect
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.
Klipper state: Not ready
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
```

bulbersator, 2021-12-08:
I'll look into that but I also tried USB connection instead of UART when I
set other serial option and the result was the same.
What I mean is that earlier even if I set wrong serial communication there
still should be answers from klippy service in terminal. When I boot the
RPI for example without printer.cfg file, it reports errors like it should,
when I upload the config file and restart the klipper, it hangs up and
octoprint loses connection to the virtual printer.
I need to remove the config file or make a misconfiguration on purpose and
reboot the RPI so the klipper service starts again.

bulbersator, 2021-12-12:
I reinstalled everything using Raspberry Pi OS Lite and KIAUH instead of FluiddPi and everything works. I don't know what was the cause.

closed 2021-12-12 by bulbersator