Klipper3d/klipper #5008: Incorrect parsing of M117, M118 commands when message starts with a digit

blalor, 2021-12-08:
`M117 ohai` and `M118 ohai` are properly handled.  `M117 0hai` and `M118 0hai` are not, with the `0` (I think actually `0h`) being parsed as a number.

The following is what appears in the Fluidd console:
```
06:17:57 $ M118 0hai
06:17:57 // Unknown command:"M118 0"
06:18:07 $ M117 0hai
06:18:07 // Unknown command:"M117 0"
06:18:18 $ M118 ohai
06:18:18 echo: ohai
```

`klippy.log` just says 
```
Unknown command:"M118 0"
Unknown command:"M117 0"
```

blalor, 2022-01-12:
Still present.

```
07:04:12 $ m115
07:04:12 // mcu build version:
         // mcu: v0.10.0-223-g4c8d24ae
07:04:12 // FIRMWARE_VERSION:v0.10.0-223-g4c8d24ae FIRMWARE_NAME:Klipper
07:04:23 $ M118 0hai
07:04:23 // Unknown command:"M118 0"
```

fleinze, 2022-01-18:
I experience the same issue. Seems it was introduced in PR #4727

The problem with the M117-Command is that any message starting with characters other than [A-Z_*/] will be ommitted.
`M117 5% complete` will display "complete" without "5%"

Sineos, 2022-01-18:
@pedrolamas maybe you can cross-check

pedrolamas, 2022-01-18:
Ok, found the issue and the good news (for me!) is that it is not related to that PR!

The problem is with [this regular expression](https://github.com/Klipper3d/klipper/blob/02d5f9754fc7f9e493c8bc5a6418e2a9ea9d7ae1/klippy/gcode.py#L175) that is used for command parsing.

This is ok:

```python
>>> args_r.split("M118 TEST")
['', 'M', '118 ', 'TEST', '']
```

From the above, the code will use index 1 and 2 and build "M118" as the command.

But if the first parameter starts with a number:

```python
>>> args_r.split("M118 0TEST")
['', 'M', '118 0', 'TEST', '']
```

The result of index 1 and 2 will be "M118 0" and that is obviously incorrect!

KevinOConnor, 2022-01-18:
Unfortunately, #4747 did introduce a regression.  (There is special handling of M117 in gcode.py to ensure the correct handler is called, but it doesn't fixup `gcode._command`.)

It should be fixed now (commit f97fd7c6).

-Kevin

pedrolamas, 2022-01-18:
@KevinOConnor seems it still fails for macro overrides:

```
[gcode_macro M117]
rename_existing: M117.1
gcode:
  M117.1 {rawparams}
  {action_respond_info(rawparams)}
```

If I have the above macro and then run `M117 0hai`:
```
// 0hai
// Unknown command:"M117.1 0"
```

fleinze, 2022-01-18:
Fixed my issue. Thanks!

pedrolamas, 2022-01-18:
I understand that, I just wanted to point out that this latest change will probably break anyone overriding the `M117` command (part of the reason why `raw_params` was added in the first place)

KevinOConnor, 2022-01-18:
> I understand that, I just wanted to point out that this latest change will probably break anyone overriding the M117 command (part of the reason why raw_params was added in the first place)

As I understand it, commit 7ef7bf60 introduced a regression (as M117 used to work with special characters, but did not after that commit).  I don't think commit f97fd7c6 introduces a regression (as M117.1 did not work with special characters before nor after that commit).

If I've missed something, let me know.
-Kevin

pedrolamas, 2022-01-18:
Sorry @KevinOConnor, you are correct: I should have tested the overrided `M117` with a "regular" use case without special characters - those are working just as before this change and no regression has been introduced.

blalor, 2022-01-18:
Thank you @KevinOConnor; `M117 0hai` does now work, but as @pedrolamas suggested, overriding it does not work, giving the error `Unknown command:"M117.1 0"`. 

My override:
```
[gcode_macro M117]
description: displays a message on the display and echos it to the console
rename_existing: M117.1
gcode:
    M117.1 { rawparams }

    {% if rawparams %}
        { action_respond_info(rawparams) }
    {% endif %}
```

blalor, 2022-01-18:
Yeah, the original intent of #4727 was to allow for overriding `M117`, which no longer works with @KevinOConnor's fix. What's the path forward?  @pedrolamas can you take a crack at that?

closed 2022-01-26 by github-actions[bot]